lvgl/lvgl #964: narrowing conversion  lv_color_hex(uint32_t c) function

InfiniteYuan1, 2019-03-20:
Error at compile time on v5.3.
Please help me porting on esp32.Thank you in advance.
```
lvgl/lv_hal/../lv_misc/lv_color.h: In function 'lv_color_t lv_color_hex(uint32_t)':
/lvgl/lv_hal/../lv_misc/lv_color.h:376:57: error: narrowing conversion of '((c & 255u) >> 3)' from 'uint32_t {aka unsigned int}' to 'uint16_t {aka short unsigned int}' inside { } [-Werror=narrowing]
 #    define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{b8 >> 3, g8 >> 2, r8 >> 3}})
```

embeddedt, 2019-03-20:
This looks like the same problem as #939 (but the exact fix there can't be applied to 5.3).

(cc @kisvegabor)

kisvegabor, 2019-03-20:
@InfiniteYuan1 
You can manually delete `lv_color_hex` and `lv_color_hex3` to make it work. They are not used in the library.

It will be fixed in `v6.0`.

closed 2019-03-23 by kisvegabor