lvgl/lvgl #967: Scrolling the contents of a tab left and right is nearly impossible

embeddedt, 2019-03-20:
Code sample (tested on the PC simulator with display width 480 and height 320):

```c
    lv_obj_t *tab = lv_tabview_create(lv_scr_act(), NULL);
    lv_obj_set_size(tab, LV_HOR_RES, LV_VER_RES);
    lv_obj_t *cont = lv_tabview_add_tab(tab, "Test");
    lv_tabview_add_tab(tab, "Test2");
    lv_obj_t *mbox = lv_mbox_create(cont, NULL);
    lv_obj_set_pos(mbox, 400, 300);
```

This one could be tough to fix. We need a way of distinguishing between the swiping on the tabview and the scrolling on the page.

embeddedt, 2019-03-20:
(cc @kisvegabor)

kisvegabor, 2019-03-21:
It's because horizontal scrolling of the page is considered as swipe left/right.
In other words, tabview is not designed to support horizontal scrolling of the pages.

However, meanwhile `lv_page_set_scroll_propagation` is added to propagate the drag to the parent if the page can't be scrolled further. Probably tabview can be updated to use that mechanism.

embeddedt, 2019-03-21:
I'll experiment with `lv_page_set_scroll_propagation` today.

Just so you know, I found this issue while testing `lv_test_theme_1`. Some of the widgets in that example are off the screen on a 480x272 screen like F746G-DISCO.

embeddedt, 2019-03-21:
I've started working on it on the `tabview_scroll` branch. The initial code is in https://github.com/littlevgl/lvgl/commit/55ff48fdd65a349c4dbdfaed0b7eef630dfa98da, but still needs more improvement.

Can I modify the scroll propagation API to have an option for only propagating horizontal and/or vertical scrolling?

embeddedt, 2019-03-21:
I decided to modify the API; if it's not suitable I can always revert it. Everything is in the `tabview_scroll` branch on both `lvgl` and `lv_examples`.

It's working a lot better now, but there's still a problem. In the PC simulator, if you drag right and then down in the top left corner of the tab content, there is a bug where the scrollable container inside the page scrolls too far. It's not easy to describe, but here's a screenshot to make it clearer:

[image]

Do you have any ideas? It's probably an issue with how I split the scroll propagation in `lv_page.c`, but I'm having trouble finding exactly what's wrong.

kisvegabor, 2019-03-22:
I've tested it and also seen some strange behavior. 

I don't see why it was required to change the interface to support hor/ver scroll propagation. I'm asking because maybe it was introduced an issue. (Not use though because the v6.0 updates were affected this feature a lot)

embeddedt, 2019-03-22:
@kisvegabor If I didn't change that, continuing to scroll down on the page would result in the tab content being shifted upwards (not desired behavior).

I'm going to look into it further now and see what I can do.

kisvegabor, 2019-03-22:
> If I didn't change that, continuing to scroll down on the page would result in the tab content being shifted upwards (not desired behavior).

I see. It's really an issue in this case.

`lv_test_tileview_1` which uses the scroll propagation very intensively seems working.

I've looked the tabviews signal function again and maybe found an easier way to achieve this. What if we check that the page is on the edge or not before setting `hor_drag` here:
https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_objx/lv_tabview.c#L737-L741

Independently from this possible bug of scroll propagation should be examined.

embeddedt, 2019-03-22:
@kisvegabor 

> why it was required to change the interface to support hor/ver scroll propagation. I'm asking because maybe it was introduced an issue.

I think you were right. I've been experimenting and it seems to be related to the fact that I'm trying to constrain the propagation to one axis, which is not how it was designed.

> I've looked the tabviews signal function again and maybe found an easier way to achieve this. What if we check that the page is on the edge or not before setting `hor_drag` here.

Sounds much simpler. I'll try that!

kisvegabor, 2019-03-24:
The pull request is merged so I close this issue.

closed 2019-03-24 by kisvegabor