lvgl/lvgl #968: Does this library support drawing in vector format?

sevencheng798, 2019-03-21:
Does this library support drawing in vector format? such as: SVG suffix format

manison, 2019-03-21:
Not right out of the box, but you could possibly integrate [SVG rasterizer](https://github.com/memononen/nanosvg) as [custom image decoder](https://blog.littlevgl.com/2018-10-05/png_converter).

kisvegabor, 2019-03-24:
We started a [Forum](https://forum.littlevgl.com/) to discuss questions there. GitHub issue should be rather about development related topics. See this [announcement](https://github.com/littlevgl/lvgl/issues/906).

If you have further questions about drawing in vector format please open a topic in the Forum and let's continue there.

closed 2019-03-24 by kisvegabor