lvgl/lvgl #969: Allow "snapshots"

beibean, 2019-03-21:
When working with a device which integrates a GUI, it's very common to take snapshots of what is currently being showed on screen basically for 2 reasons:

- Pass automatic tests based on screen.
- Analyze in detail image with, for example, GIMP.

So, my propossal is to have a simple function similar to this one:

``` C
/** Take snapshot of the screen zone.
* @param   ownBuffer   Buffer where the snapshot will be saved.
* @param   zone            Zone where the snapshot wants to be taken.
*/
lv_snapshot (uint8_t* ownBuffer, lv_coord_t zone)
```

Then, user of library is in charge of treating that ownBuffer. For example:

- Save it as binary file.
- Compare that binary file to a reference binary file (automatic test).
- Convert that binary file to image format using ImageMagick commands (analyze image).

What do you think about this idea? Is it already possible in any way?

PD: as said, this could be used for automatic tests, which is one of the big lacks of LittlevGL.

embeddedt, 2019-03-21:
@beibean I think it would be feasible to enable snapshots to be made of the canvas object. However, there is no method of reading back pixels from the display, so we can't take a screenshot of the whole display.

(cc @kisvegabor)

puzrin, 2019-03-22:
This does not looks like core LVGL feature - more close to driver implementation details.

embeddedt, 2019-03-22:
@puzrin The driver API doesn't require the ability to read back pixels from the display. Without that, taking a screenshot of the display itself would be more complex IMO.

The canvas is easier because it's input comes from a buffer anyways.

puzrin, 2019-03-22:
@embeddedt it's normal practice to make additional methods, specially for testing. I think driver-level access will be more easy to make & maintain, if we speak about testing.

kisvegabor, 2019-03-22:
@beibean 
I'd do it like this:
1. Call `lv_obj_invalidate(lv_scr_act())` to invalidate the whole screen
2. Tell your driver that now we are in "snapshot mode"
3. Call `lv_refr_now()` to redraw the screen
4. Your `disp_flush` function will get all parts of the rendered screen from top to bottom
5. Save these parts as you need
6. After `lv_refr_now()` disable "snapshot mode"

To make it easier I can image a special driver (e.g. `bmp_driver`) which saves the screen to a file instead of putting it to the screen.

I think that having and maintaining automatic tests would be too much burden. However, build tests with several configurations would be useful. We had a discussion about it here https://github.com/littlevgl/lvgl/issues/768

beibean, 2019-03-23:
@embeddedt 
>We use GitHub issues for development related discussions.
In the future, please use the forum to ask questions or suggest enhancements.

I thought enhancements could be posted here. From now on I will use the forum. Is it the *How to?* the right place to post enhancements proposals?

@kisvegabor 
> I'd do it like this: [...]

That's a good idea! Thank you, I will try that way

>I think that having and maintaining automatic tests would be too much burden. However, build tests with several configurations would be useful. We had a discussion about it here #768

You are right, tests require a big effort. Build tests may be good enough to ensure a new commit doesn't break the project. 

Automatic tests are thought also to check final screen output doesn't change with a new commit, or at least control those changes. Any way, I saw that you are using the library everyday and also others, so anyone can notice a change on final result and report it.

kisvegabor, 2019-03-24:
> I thought enhancements could be posted here. From now on I will use the forum. 

I'd prefer to discuss even enhancement ideas in the forum because it can turn out that the proposed enhancement can be achieved with the current functionalities too. In this case, it's not related to development.

> Is it the How to? the right place to post enhancements proposals?

It's better to describe what you want to achieve and tell what is your current approach or enhancement idea. If a new feature really should be/could be implemented we should continue on GitHub.

The forum is still very new and probably these "rules" still need to be polished. 

>> I'd do it like this: [...]

> That's a good idea! Thank you, I will try that way

I'm curiously waiting for your feedback. If it works the "bmp_driver" would be a reasonable update.

> Automatic tests are thought also to check final screen output doesn't change with a new commit, or at least control those changes. Any way, I saw that you are using the library everyday and also others, so anyone can notice a change on final result and report it.

I also think that it's enough in the current stage.

beibean, 2019-03-28:
I finally tested quickly easily. Taking into account within simulation vdb size is full screen, here is my flushBuffer function.

```C
/** Flush buffer.
 *
 * Flush buffer into a binary file.
 *
 * @note: data pixel should be formated to uint32_t RGBA. Imagemagick requirements.
 *
 * @param[in] pFileName   Output binary file name.
 *
 */
void DISP_IMPL_flushBuffer (const char_t* pFileName) {

  FILE* pFileOut;

  assert(DISP_IMPL_pCtx != NULL);

  pFileOut = fopen(pFileName, "w");

  if (pFileOut == NULL) {
    SIM_logMsg(("[Display] error: %s cannot be opened", pFileName));
    return;
  }

#ifdef LV_USE
  lv_vdb_t*   pVdb  =  lv_vdb_get();
  lv_color_t* pData = &pVdb->buf[0];
  uint32_t*   pBuff = DISP_IMPL_pCtx->pBuffer;

  lv_obj_invalidate(lv_scr_act());
  lv_refr_now();

  for (uint32_t i = 0; i < DISP_IMPL_pCtx->bufferBytesLen; i++) {
    pBuff[i] = DISP_IMPL_lvgl_formatPixel(pData[i]);                /* Convert RGB565 to RGBA8888 */
  }
#endif /* LV_USE */

  fwrite(DISP_IMPL_pCtx->pBuffer,
         sizeof(uint32_t),
         DISP_IMPL_pCtx->bufferBytesLen,
         pFileOut);

  fclose(pFileOut);
  SIM_logMsg(("[Display] data flushed to %s", pFileName));
}
```

Here is the Imagemagick command to convert binary file to bmp file, in case any of you want to use it:

```bash
convert -size 320X240 -depth 8 rgba:${BIN_FILE} ${BIN_FILE}.bmp
```

Here you can see one of my final test output images (converted from bmp to png for GitHub support):

[image]

> I'm curiously waiting for your feedback. If it works the "bmp_driver" would be a reasonable update.

Finally it was achieved from application side, without applying any change within LittlevGL. Is up to you to include:

- "bmp_driver" could work perfectly.
- Just information about How to do it. I have no problem to document it better, maybe in the blog?

Waiting for your decision 😉

kisvegabor, 2019-03-29:
Awesome!

The only issue I see is the screen sized VDB. With a smaller VDB if the screen is updated in chunks you need to:
- Add the "file writer" code to the display driver. (You can change `disp_flush` in run time)
- Use `lv_inv_area()` + `lv_refr_now` in loop for every chunk based on VDB size.

Anyway, as always, a blog post would be very welcome :smile:  

> I just realized that within the function code I attached there are some holder functions and contexts, but I hope you understand it.

Yes, it was clear! :smile:

beibean, 2019-03-29:
> The only issue I see is the screen sized VDB.

I just modified the code to permit smaller VDB but did not use lv_inv_area(). This is hoy I did it:

```C
void DISP_IMPL_flushBuffer (const char_t* pFileName) {

  FILE* pFileOut;

  assert(DISP_IMPL_pCtx != NULL);

  pFileOut = fopen(pFileName, "w");

  if (pFileOut == NULL) {
    SIM_logMsg(("[Display] error: %s cannot be opened", pFileName));
    return;
  }

#ifdef LV_USE

  bSnapshot = BOOL_TRUE;
  pBuff = DISP_IMPL_pCtx->pBuffer;
  lv_obj_invalidate(lv_scr_act());
  lv_refr_now();
  bSnapshot = BOOL_FALSE;
#endif /* LV_USE */

  fwrite(DISP_IMPL_pCtx->pBuffer,
         sizeof(uint32_t),
         DISP_IMPL_pCtx->bufferBytesLen,
         pFileOut);

  fclose(pFileOut);
  SIM_logMsg(("[Display] data flushed to %s", pFileName));
}

/** Function called from disp_flush before lv_flush_ready().
 *
 */
void APP_DISP_lvgl_flush (const lv_color_t* pData, const DISPLAY_BOUNDS_T* const pPxBounds)
{
  if (bSnapshot == BOOL_TRUE) {
    for (uint32_t i = 0; i < (pPxBounds->dimension.width * pPxBounds->dimension.height); i++) {
      *pBuff = DISP_IMPL_lvgl_formatPixel(pData[i]);
      pBuff++;
    }
  }
}

```

> Anyway, as always, a blog post would be very welcome.

I will do it yes, I don't know when though 😉.

kisvegabor, 2019-03-29:
Wouldn't be better to call `fwrite` in`APP_DISP_lvgl_flush`. This way the `pBuff` wouldn't be required.

beibean, 2019-03-29:
> Wouldn't be better to call fwrite inAPP_DISP_lvgl_flush? This way the pBuff wouldn't be required.

The problem here is that lvgl_data needs to be formatted, that's why pBuff is used to store ImageMagick fomatted pixels.

It may exist any Imagemagick solution to convert straight away the LittlevGL pixel format to .bmp, but I don't know it so far.

kisvegabor, 2019-03-29:
> I get you now. It's better to use fwrite everytime a pixel is formatted if you want to get rid of pBuff. I will explain it this way within the blog.

Exactly. You need only a buffer for one pixel and write it immediately. It will be slower but probably you have more time  than RAM to take a snapshot :)

beibean, 2019-03-29:
> It will be slower but probably you have more time than RAM to take a snapshot :).

Actually, RAM should not be a matter when working within *simulation*. Anyway I liked the idea, and test it. I did not notice difference, so I'll keep it that way 😉

kisvegabor, 2019-03-29:
Great! :smile:

embeddedt, 2019-04-04:
It looks like the main question was solved so I will close this issue. If there is more to be discussed feel free to reopen it.

closed 2019-04-04 by embeddedt