lvgl/lvgl #970: creat window prevent serial port receiving.

relliktei, 2019-03-22:
I am using 2 page in my project.
Page1: main.   
[image]

Page2:
[image]

I can Click button into second page calling lv_win_create()

I also use one serial port to receive data from remote device to update status in pages.
The realtime data can be shown correctly in page1.

But I notice one problem:
**When I go to Page2, data cannot update in time.**
[image]

I also dump data received by serial port as below:
[image]

**In dump data picture, the only one thing different is that I go to page2 by calling lv_win_create().**

It sounds like serial port receiving goes strange after I create new window.
I don't know why. How can I fix it?

relliktei, 2019-03-22:
Sorry, I solved it.
I use lv_task_create to create new task to update page2, then USART can receive data correctly.
I will close this issue.

closed 2019-03-22 by relliktei

embeddedt, 2019-03-22:
Glad you figured it out!
----------------------
We use GitHub issues for development related discussions.
In the future, please use the [forum](https://forum.littlevgl.com/) to ask questions.