lvgl/lvgl #972: BUG about "lv_page"？ (v5.3)

as564335sa, 2019-03-22:
Hi,
I have a puzzle. This is the code for  "lv_test_page_2.c",
```
lv_obj_t * page1 = lv_page_create(lv_scr_act(), NULL);
    lv_obj_set_size(page1, LV_DPI, LV_DPI * 2);
    lv_page_set_scrl_fit(page1, true, true);
    lv_page_set_scrl_layout(page1, LV_LAYOUT_COL_L);
    lv_page_set_sb_mode(page1, LV_SB_MODE_OFF);
    lv_page_set_rel_action(page1, page_rel_action);
    lv_page_set_style(page1, LV_PAGE_STYLE_BG, &bg);
    lv_page_set_style(page1, LV_PAGE_STYLE_SCRL, &scrl);
    lv_page_set_style(page1, LV_PAGE_STYLE_SB, &sb);

    lv_obj_t * label = lv_label_create(page1, NULL);
    lv_label_set_text(label, "First line of a text\n"
                      "Second line of a text\n"
                      "Third line of a text\n"
                      "Forth line of a text\n"
                      "Fifth line of a text\n");
```

When I debug this code，I found that the variable address of `page1` was not the same as the parent  variable  address of the `label`,
such as:  
`page1` : 0x96bc60,
`label->par` : 0x96bcf8
Then, i can't use `lv_obj_get_parent` to get the parent address of a obj.

BesitzeRuf, 2019-03-22:
Hi @as564335sa, can you show the code, by which you are reading the addresses of `page1  ` and `label->par` . Is it possible, that you are reading the addresses of pointers, not the value stored in these pointer?

as564335sa, 2019-03-22:
Hi, @BesitzeRuf 
I changed the code, like this:

```
lv_obj_t * page1 = lv_page_create(lv_scr_act(), NULL);
    lv_obj_set_size(page1, LV_DPI, LV_DPI * 2);
    lv_page_set_scrl_fit(page1, true, true);
    lv_page_set_scrl_layout(page1, LV_LAYOUT_COL_L);
    lv_page_set_sb_mode(page1, LV_SB_MODE_OFF);
    lv_page_set_rel_action(page1, page_rel_action);
    lv_page_set_style(page1, LV_PAGE_STYLE_BG, &bg);
    lv_page_set_style(page1, LV_PAGE_STYLE_SCRL, &scrl);
    lv_page_set_style(page1, LV_PAGE_STYLE_SB, &sb);

    printf("page1 : %08X\r\n",page1);
    lv_obj_t * label = lv_label_create(page1, NULL);
    lv_label_set_text(label, "First line of a text\n"
                      "Second line of a text\n"
                      "Third line of a text\n"
                      "Forth line of a text\n"
                      "Fifth line of a text\n");
    printf("parent of label : 0x%08X\r\n",lv_obj_get_parent(label));
    printf("parent of label 2: 0x%08X\r\n",lv_obj_get_parent(lv_obj_get_parent(label)));
```

and get this:
[image]
I don't understand the result.

embeddedt, 2019-03-22:
@as564335sa It's because the page contains a scrollable container object. When you create an object with the page as it's parent, it is automatically moved to the container.

The hierarchy is as follows: `page1 >>> page1_cont >>> label`

This is why the parent of the parent is the page.

as564335sa, 2019-03-23:
Thanks,i get it!

closed 2019-03-23 by as564335sa