lvgl/lvgl #974: [meta] lvgl code format and related tools.

C47D, 2019-03-22:
Follow up for #794 and #71 in regards of lvgl code format and tools that would help lvgl to ensure it.

In #71 we were discussing using clang-format (available [here](https://clang.llvm.org/)), the 'issue' i ind with it is it have lots of options for code formatting, so we will need to test the options we see appropriate.

kisvegabor, 2019-03-24:
Thank you for opening the issue!

So let's warm up this topic! The goal would be to achieve something described in the [styling guide](https://docs.littlevgl.com/#Coding-Style-Guide).

In addition (if possible) the behavior described [here](https://github.com/littlevgl/lvgl/issues/794#issuecomment-475497748) would be great.  Basically, that's my only problem with astlye. If we can find a solution for it we can change to clang.  

@C47D Do you have time and interest to make an initial clang format file?

C47D, 2019-03-25:
I do have the interest to do the clang-format file, I'm not sure about the second link being possible tho. Will upload the clang-format file later this week.

kisvegabor, 2019-03-26:
Awesome!

C47D, 2019-03-26:
So i am using [clang-format 8.0.0](http://releases.llvm.org/8.0.0/tools/clang/docs/ClangFormatStyleOptions.html#) for this, just unzip the attached zip file.

To test it you will need to install clang, to apply it to a file without applying changes:
> clang-format -style=file path_to_file

To apply changes:
> clang-format -style=file -i path_to_file

This clang-format file is the one i do usually use, i think the only option i changed was the pointer position, anyone can try it and improve it, it surely have a lot of non used options because it's based on the LLVM style guide.

[.clang-format.zip](https://github.com/littlevgl/lvgl/files/3010682/default.clang-format.zip)

kisvegabor, 2019-03-28:
@C47D Thank you!

I installed `clang`-format-7` because was 8 wasn't available in my package manager and got unsatisfied dependencies on manual install.

I renamed the style file to `a` to be sure I'm not mistyping the name and I got this error (I think it's not related to the version):
```
clang-format-7 -style=a lv_core/lv_obj.c
Invalid value for -style
```
However, it was working:
```
clang-format-7 -style=WebKit lv_core/lv_obj.c
```

Do you have any idea?

C47D, 2019-03-28:
Hi,

The -style option can only have `LLVM, Google, Chromium, Mozilla, WebKit or file` as values, the file should be named .clang-format.

kisvegabor, 2019-03-28:
Still the same error with `clang-format-7 -style=.clang-format ./lv_core/lv_obj.c`

C47D, 2019-03-28:
Try `clang-format-7 -style=file ./lv_core/lv_obj.c`

kisvegabor, 2019-03-28:
> Use -style=file to load

Uh, thank you. It"s working this way!  I should learn to read first... :sweat: 

I can immerse in it tomorrow.

C47D, 2019-03-28:
No problem, you can play with the options to get it as you want and choose the better option for the project.

kisvegabor, 2019-03-31:
I played with it and pushed the result into https://github.com/littlevgl/lvgl/tree/clang

Please have look at it and tell your remarks. From what I saw so far, I can say the result is beautiful.

At some places, care must be taken , e.g:
```c
/*Auto align consecutive `=`s */
int very_long_var_name = 1;
int a                  = 1;               
int b                  = 1;                   

/*New line should be added*/
int very_long_var_name = 1;

int a = 1;               
int b = 1;                  
```

C47D, 2019-03-31:
The alignment of consecutive assignments can be changed, let me find the option to do so.

kisvegabor, 2019-03-31:
It's fine to align consecutive assignments, just an extra new line need to be inserted to avoid ugly formattings at some places.

C47D, 2019-03-31:
Oh silly me, just read the comments on the code snippet, i used /* clang-format off */ with success at work, i don't know why it is not working for you. Do you think the clang-format file can be improved or something? there are a lot of unused options, but i didn't remove them.

embeddedt, 2019-03-31:
In general, I think it's great! I read through a few files and put my suggestions below.

>  I needed to exclude fonts because they are too large.

Which is fine; those files are generated by the converter so I don't think formatting is of extreme importance there.

----------------------

This is a bit weird:

https://github.com/littlevgl/lvgl/blob/9c60edd24b974e4af5e0a6eb8e5ce3e206505a48/src/lv_objx/lv_btn.c#L132-L138

But I don't think we can do much about it because I think the preprocessor statement causes `clang-format` to believe it's a separate group of consecutive assignments.

------------------------

It would be nice if `clang-format` could align these like it does for assignments. Not sure if that is possible.

https://github.com/littlevgl/lvgl/blob/9c60edd24b974e4af5e0a6eb8e5ce3e206505a48/src/lv_objx/lv_btn.c#L262-L266

------------------------

These are very strange. I think it is trying too hard to respect the maximum line width.

https://github.com/littlevgl/lvgl/blob/9c60edd24b974e4af5e0a6eb8e5ce3e206505a48/src/lv_objx/lv_btn.c#L433-L435

https://github.com/littlevgl/lvgl/blob/9c60edd24b974e4af5e0a6eb8e5ce3e206505a48/src/lv_draw/lv_draw_rect.c#L17-L24

-------------------------

One really important thing, in my opinion, is to check whether it behaves properly when it is run a second time over code it formatted. Other tools I have used in the past will change the line breaking every time they are run, which makes a nasty `git` history.

I will have a look at the options when I get a chance.

C47D, 2019-03-31:
> These are very strange. I think it is trying too hard to respect the maximum line width.

Yes, it does, to avoid that on the first case i tend to put the comment on top of the function so the line doesn't gets over 80 chars, you can change it with the option **ColumnLimit**, related option **ReflowComments**.

> One really important thing, in my opinion, is to check whether it behaves properly when it is run a second time over code it formatted. Other tools I have used in the past will change the line breaking every time they are run, which makes a nasty git history.

Agree, after using clang-format it's recommended to take a quick look at the results to see if it didn't messes the file.

> It would be nice if clang-format could align these like it does for assignments. Not sure if that is possible.

Maybe having the case, the assignment and break in different lines, i'm not sure about that. See **AllowShortCaseLabelsOnASingleLine**, **IndentCaseLabels** for switch case related options.

kisvegabor, 2019-04-01:
> But I don't think we can do much about it because I think the preprocessor statement causes clang-format to believe it's a separate group of consecutive assignments.

An empty line creates a new "block". See: https://github.com/littlevgl/lvgl/commit/4aae4e3ae3dfe235af2e59174d2e18176db16e64

> It would be nice if clang-format could align these like it does for assignments. Not sure if that is possible.

> Maybe having the case, the assignment and break in different lines, i'm not sure about that. See AllowShortCaseLabelsOnASingleLine, IndentCaseLabels for switch case related options.

It seems **IndentCaseLabels** aligns only the case label but not the first instruction after it.

> Yes, it does, to avoid that on the first case i tend to put the comment on top of the function so the line doesn't gets over 80 chars, you can change it with the option ColumnLimit, related option ReflowComments.

I played with it [here](https://zed0.co.uk/clang-format-configurator/) but wasn't able to fix it. I don't see other option than not placing long comments besides the code.

> One really important thing, in my opinion, is to check whether it behaves properly when it is run a second time over code it formatted. Other tools I have used in the past will change the line breaking every time they are run, which makes a nasty git history.

I run it again and didn't change anything.

---

So as I see care must be taken in two cases:
1. Don't add long in-line comments 
2. Add an empty line after initializations with a different width (usually logically different as well)

kisvegabor, 2019-04-04:
I run clang-format on dev-6.0 and pushed the result. 
I fixed some of the ugly formattings but if you find some while reading the source feel free to fix them and send PRs.

@C47D Thank you for putting together the clang-format file. It made the code more beautiful.

@embeddedt Thank you for the review of the result.

closed 2019-04-04 by kisvegabor

C47D, 2019-04-04:
Hi, no problem, i hope reviewing clang-format ugly formatting will become easier on later commits.
Thanks @kisvegabor , @embeddedt .