lvgl/lvgl #975: [meta] issue and pull request templates

C47D, 2019-03-22:
Hi,

Just found out there are [templates for new issues and pull requests](https://help.github.com/en/articles/about-issue-and-pull-request-templates), do you think this can be useful for this project?

[Create Pull request template](https://help.github.com/en/articles/creating-a-pull-request-template-for-your-repository).
[Create issue template](https://help.github.com/en/articles/creating-issue-templates-for-your-repository).

embeddedt, 2019-03-22:
@C47D Yes, I intend to add these so that people will know to direct questions and enhancement requests to the forum rather than here.

C47D, 2019-03-22:
@embeddedt Great, should we close this issue until the templates are added?

embeddedt, 2019-03-22:
@C47D I'm working on them right now.

embeddedt, 2019-03-22:
@kisvegabor Issue templates are added; let me know if you think something needs changing.

closed 2019-03-22 by embeddedt

C47D, 2019-03-22:
That was quick!, i don't know how often lvgl receive pull request, if so, do they need an special format?
It might be useful to add a template for them as well.

embeddedt, 2019-03-22:
@C47D I don't think templates are necessary for pull requests right now, because usually a pull request is only opened after a discussion on the feature (unless it's a bugfix).

In the future it might be something to look into though.

C47D, 2019-03-22:
I see, you are right, thanks for the quick reply!

kisvegabor, 2019-03-22:
Looks great! Thank you!

I've just renamed `bug_report.md` to `bug-report.md` for consistency.