lvgl/lvgl #979: lv_ddlist select wrong option on click

embeddedt, 2019-03-24:
When you click in either direction on the "Theme" or "Color" rollers, they usually move by two options instead of one. Not sure exactly what the problem is. This isn't hardware-specific.

embeddedt, 2019-03-24:
@kisvegabor I'll investigate `lv_label_get_letter_on` today.

kisvegabor, 2019-03-24:
Do you see the incorrect return value?

kisvegabor, 2019-03-24:
I found it. Stupid mistake...  See: https://github.com/littlevgl/lvgl/commit/770d1e08e1e5a94db4749da1b80b5cfe4e4153ac

Probably this line was deleted accidentally. Without it, the x coordinate could be very big and 
the `lv_label_get_letter_on()` found the next line.

Could you verify if it really works correctly now?

embeddedt, 2019-03-24:
@kisvegabor 

> Stupid mistake... See: 770d1e0

I know how it feels to discover something like that. :wink:

> Could you verify if it really works correctly now?

Works as expected now. Thanks for debugging it!

closed 2019-03-24 by embeddedt

kisvegabor, 2019-03-24:
> I know how it feels to discover something like that. wink

That's life :smile: