lvgl/lvgl #981: Use general integer type

kisvegabor, 2019-03-24:
In light of https://github.com/littlevgl/lvgl/issues/785#issuecomment-474791816 a general integer type should be used.

`uint16_t`, `int16_t`, `uint32_t`, `int32_t` should be replaced with `int` where the expected maximum size is about 1000. 

Function paramteres and local and static variables should be replaced too.

It would be great to start with the review of the obejct types in `lv_objx` folder.

Contribution is welcome here even in only some files.

embeddedt, 2019-03-24:
> where the expected maximum size is about 1000

Why can't it go all the way to 65000? I just want to understand the reason for this criteria.

kisvegabor, 2019-03-25:
I somewhere the expected maximum value is about 65000 then it might be too close to the limit and special care need to be taken. However if we are speaking about
- number of children
- number of points on a chart
- selected ddlist option 
- etc

we just need a "big enough number" because we will be far even from the 32K limit.

manison, 2019-03-25:
> In light of #785 (comment) a general integer type should be used.

How should this affect
* unsigned types (_lv_btnm.h_: `lv_btnm_ctrl_t`)
* enums (_lv_btnm.h_: `lv_btnm_style_t`)
* bitfields (_lv_btnm.h_: `lv_btnm_ext_t::recolor`/`one_toggle`)

How do we track progress of this? How do we prevent eventual conflicts? I'm eventually going to update btnm and mbox.

kisvegabor, 2019-03-25:
It should be applied mainly to function parameters and return values to make the API cleaner and more consistent. Local variables could be updated too.

So typedefs should not be affected by this update.

> How do we track progress of this?
I can add checkboxes to the first post.

C47D, 2019-03-25:
As far as I know int size is platform (architecture) dependant, if the maximum size expected is 1000 then uint16_t might be enough, or uint32_t if you expect the limit to grow.

uintX_t types are defined by the compiler in the stdint.h header file.

amirgon, 2019-03-25:
If you replace all instances of `uint16_t` with `int`, on 32-bit architectures you are going to waste RAM ( `int` will be defined there as 32-bit wide value, as opposed to `uint16_t` which is 16-bit wide).   

What is the motivation that justifies this waste of RAM on embedded architectures with limited RAM available?

kisvegabor, 2019-03-26:
> What is the motivation that justifies this waste of RAM on embedded architectures with limited RAM available?

The issue was that the integers even in the API weren't consistent. 

It affects mainly return values and function parameters so the amount of wasted RAM is negligible.
`int16_t` also could be used but `int` is clear.

kisvegabor, 2019-03-28:
To see the obstacles in practice I started to refactor some objects. In case of `lv_bar` there is:
```c
void lv_bar_set_value(lv_obj_t * bar, int16_t value);
```

and 
```c
typedef struct
{
    /*No inherited ext*/            /*Ext. of ancestor*/
    /*New data for this type */
    int16_t cur_value;              /*Current value of the bar*/
    int16_t min_value;              /*Minimum value of the bar*/
    int16_t max_value;              /*Maximum value of the bar*/
    uint8_t sym    :1;              /*Symmetric: means the center is around zero value*/
    lv_style_t *style_indic;        /*Style of the indicator*/
} lv_bar_ext_t;
```

Using `int` here really would introduce unnecessary extra RAM usage in `cur/min/max_value`.

Using `int16_t` instead of `int` should work in most of the cases but `int16_t` doesn't express that "it is the general integer type of LittlevGL".

My earlier suggestion about `typedef in16_t lv_int_t` would give control over the type of the integer values and would show that it the general integer type.

What do you think?

C47D, 2019-03-28:
I'm not an expert but i don't really see a benefit of using lv_int_t instead of uint16_t, take into account the amount of effort it will take reviewing all the codebase and changing the variable types, i rather use a general integer type of C and not lvgl, but this is just my opinion.

kisvegabor, 2019-03-29:
> I'm not an expert but i don't really see a benefit of using lv_int_t instead of uint16_t

The only benefit I see is a more consistent API. Now we have all `(u)int16/32_t` even when `int16_t` would be fine in most of the cases. If the majority is against `lv_int_t`, `int16_t` can work too.

> i rather use a general integer type of C and not lvgl, but this is just my opinion.

It would increase RAM usage as I described above.

canardos, 2019-04-01:
I am indifferent between using your own `typedef` referencing the `C99` fixed width integer types or using them directly, but I would be strongly opposed to using platform dependent types such as `int`.

This library targets embedded hardware, where memory resources are truly very limited and defined memory layout can be critical. Using a type such as `int`, which might be 2 or 4 (or potentially more) bytes to represent values up to ~1000 has no upside as far as I can see and quite a few drawbacks.

If consistency is the goal, then use the `C99` types consistently, and if ease of modification is the goal, then alias them, but please don't grow the memory requirements unless there is a tangible upside.

kisvegabor, 2019-04-02:
These arguments are correct. Probably a quick review on the API and the use of `int16_t` where possible would be enough for now.

Thank you for sharing your thoughts.

As there won't be a big refactoring I close this issue. If you find some inappropriate variable types feel free to send a PR to `dev-6.0`. I will also search for them and fix if I find any.

closed 2019-04-02 by kisvegabor