lvgl/lvgl #982: Can not draw the text using lv_draw_letter() outside the design function

szjdb, 2019-03-24:
Hi, kisvegabor ,
 I can not draw the text using lv_draw_letter() outside the design function and the lv_refr_get_disp_refreshing() return 0 in that case. I was wondering if we can draw the text or other basic segment  freely ?

Best Regards,

embeddedt, 2019-03-25:
(cc @kisvegabor)

kisvegabor, 2019-03-26:
The draw functions are for internal use. See: https://forum.littlevgl.com/viewtopic.php?f=12&t=57

We started a [Forum](https://forum.littlevgl.com/) to discuss questions there. GitHub issue should be rather about development related topics. See this [announcement](https://github.com/littlevgl/lvgl/issues/906).

If you have further questions about drawing please open a topic in the Forum and let's continue there.

closed 2019-03-26 by kisvegabor