lvgl/lvgl #985: Extend clickable area

GreyMS, 2019-03-27:
I didn't found appropriate section at the forum, so I'll try to write here.

I have asked about the way to extend clickable area of image button on the forum. It was recommended to create simple button and change images in action handlers.
It is solution, but I decided to do it in other way. I created own simple object, that includes invisible simple button, hacks signal function and passes most of the signals to its children.

Now I can create "clickable_area", add any button to it as a child to container and this button will receive all signals of its parent's clickable_area object, like "pressed", "released" and so on. I tested it in my project with little image buttons and it works like a charm.

I can create pull request to dev branch with this new object, but does some one needs such feature? Or may be someone have suggestions why such approach is not good or there are any improvements?

embeddedt, 2019-03-27:
(cc @kisvegabor)

embeddedt, 2019-03-27:
How did you handle styles for `clickable_area`? Does it look like a larger button?

GreyMS, 2019-03-27:
I kept style_set and style_get for this function just like in imgbtn, but my object doesn't have any visual representation. It is invisible container. Styles were kept to be applied to object's children if needed.

embeddedt, 2019-03-27:
@GreyMS This seems like it would be inconsistent to a user. If there is no visible border on the container, how would they know what image they are clicking? Could you send a screenshot so I can get a better idea of how this looks?

GreyMS, 2019-03-27:
[image]
This is my little image button. It is about 40x40 pixels and it is hard to user press it with finger. So it is really handful to extend clickable area to 100x100 pixels

embeddedt, 2019-03-27:
@GreyMS That is a good idea for that specific use case, because there aren't any other widgets within a close distance.

Anyways let's wait and see what @kisvegabor thinks about it.

GreyMS, 2019-03-27:
[image]
With my new `clickable_area` user can press at any point at the red square (it can be customized by `obj_set_size`) and button will react as if user pressed it precisely.
Yes, it can be useful only for lonely widgets, but it is the way to decompose widget visible size and clickable area, which leads to more freedom in design of visible elements.
Another way is to create image with wide transparent border, but it leads to increase of memory usage for such image.
And creating of action handlers forces people to decline usage of `imgbtn` in some cases and to create analogue by them self.

kisvegabor, 2019-03-28:
I think a cleaner solution would be to add an `extra_click_area` attribute to `lv_obj_t` itself.  This attribute should be taken into account here: https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_core/lv_indev.c#L979

The user should be to enable this feature in `lv_conf.h` to reduce the memory usage of the objects.

What do you think?

GreyMS, 2019-04-03:
I tried and it works!
I patched [https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_core/lv_indev.c#L979](url) as 
```
#if USE_LV_EXTENDED_CLICK_AREA
	if(lv_area_is_point_on(&obj->ext_coords, &proc->act_point)) {
#else
    if(lv_area_is_point_on(&obj->coords, &proc->act_point)) {
#endif
```
and added two members to `lv_obj_t`:
```
#if USE_LV_EXTENDED_CLICK_AREA == 1
	lv_area_t ext_paddings;
	lv_area_t ext_coords;
#endif
```
and new function:
```
#if USE_LV_EXTENDED_CLICK_AREA == 1
void lv_obj_set_ext_paddinds(lv_obj_t * obj, lv_coord_t w, lv_coord_t h);

void lv_obj_set_ext_paddinds(lv_obj_t * obj, lv_coord_t w, lv_coord_t h)
{
    obj->ext_paddings.x1 = w;
    obj->ext_paddings.x2 = w;
    obj->ext_paddings.y1 = h;
    obj->ext_paddings.y2 = h;

	obj->ext_coords.x1 = obj->ext_paddings.x1 > obj->coords.x1 ? 0 : obj->coords.x1 - obj->ext_paddings.x1;
    obj->ext_coords.x2 = obj->coords.x2 + obj->ext_paddings.x2;
	obj->ext_coords.y1 = obj->ext_paddings.y1 > obj->coords.y1 ? 0 : obj->coords.y1 - obj->ext_paddings.y1;
    obj->ext_coords.y2 = obj->coords.y2 + obj->ext_paddings.y2;
}
#endif
```
Also I patched all places in `lv_obj.c` to synchronize ext_coords with object's coords.
Should I merge it into dev-6.0 or it can be done in a better way?

embeddedt, 2019-04-03:
@GreyMS Please submit a pull request. It makes it easier to review the changes.

GreyMS, 2019-04-03:
@embeddedt I've created pull request with my proposal.

embeddedt, 2019-04-03:
@GreyMS Great! Let's continue the discussion there.

closed 2019-04-03 by embeddedt