lvgl/lvgl #987: Text Area: LV_CURSOR_UNDERLINE blink doesn't work

upbeat27, 2019-03-27:
**Describe the bug**

For Text Area object: blink animation for LV_CURSOR_UNDERLINE does not work correctly.
Other cursor types animate correctly.
The cursor will either not blink at all, or it will change size instead of blinking.

**To Reproduce**

Create a Text Area with lv_cursor_type_t set to LV_CURSOR_UNDERLINE

**Expected behavior**

The cursor should blink.

**Additional context**

It used to work as of 92747ec37a19d73402c050e6e4b514254a97e764, I think the culprit may be 7ac12cd07100e7adfcd54a6bc8d876f675101d45

embeddedt, 2019-03-27:
(cc @kisvegabor)

kisvegabor, 2019-03-28:
It's working for me on the latest `master`:
```c
    lv_obj_t * ta = lv_ta_create(lv_scr_act(), NULL);
    lv_ta_set_cursor_type(ta, LV_CURSOR_UNDERLINE);
```
[image]

upbeat27, 2019-03-29:
Hmmm... it definitely is not working for me. Maybe it is a combination of my theme and font, but I couldn't figure out what the cause is. Tried changing lots of things, nothing seemed to have an effect.
[theme_and_font.zip](https://github.com/littlevgl/lvgl/files/3023587/theme_and_font.zip)

Some gifs (only difference being which lvgl commit is used):

Current Master:
[image]

When it used to work, on commit 92747ec37a19d73402c050e6e4b514254a97e764:
[image]

kisvegabor, 2019-04-01:
Still blinking for me.
[image]

Can you send a minimal PC simulator project?

embeddedt, 2019-04-01:
@upbeat27 As well as sending a project, check whether animations are enabled and whether you are calling `lv_tick_inc` at the appropriate times. Those shouldn't be a problem, but I ask just in case it's related.

upbeat27, 2019-04-01:
So... I couldn't get it to do it with the simulator, but I knew it was happening!
So I had to dig a little deeper.
The issue is caused by having a round area callback.
The callback doesn't even have to do anything, just setting a callback causes the problem. For example:
```c
static void round_area_cb(lv_area_t* a)
{
// Don't even do anything...
}
// In init:
lv_refr_set_round_cb(round_area_cb);
```

Rather than send the entire project, just add that to your existing test.

kisvegabor, 2019-04-02:
Thank you, I really found a bug if  `round_cb` is enabled. Pushed the fix into master.

Now the cursor is blinking but I couldn't test it with real hardware where you are rounding to 8. Please verify if it does not break the "normal" rounding behavior.

upbeat27, 2019-04-02:
@kisvegabor this fixed it for me, blink works fine now. Normal rounding behavior seems to be working on my target with OLED requiring x values to be on a 4 pixel boundary. In my case the rounding is rather simple though. I feel like there could potentially be a problem if someone needs to round up y2, for example (I don't know if any display would require such a thing).

closed 2019-04-02 by upbeat27

kisvegabor, 2019-04-03:
> this fixed it for me, blink works fine now. 

Great!

>  I feel like there could potentially be a problem if someone needs to round up y2, for example (I don't know if any display would require such a thing).

Rouning y to 8 is a common scenario for monochrome displays where 1 byte represents 8 pixels in a  column.