lvgl/lvgl #989: Reduce Glyph Storing Size

beibean, 2019-03-28:
Hi,

I just saw how the Font Glyphs/Symbols are stored and I can see that there is loads of Flash memory it's been wasted.

This is because for every Glyph/Symbol we store every pixel within its height. No matters if it is useful information or not:

[image]

When I have been working with glyph storing, **baseline** concept has been taken into account. Where baseline stands for the difference between the _height 0_ and the first row with any non-zero pixel starting to count from bottom.

I think LittlevGL should include this concept and get rid of loads of _0x00 useless bytes_. So, for the two characters shown above, they would be stored like this:

[image]

If this new concept is included several changes should be done:

- Tools to get fonts.
- Update font files.
- Add new baseline parameter to structure lv_font_glyph_dsc_t.
- Writing glyphs/symbols functions.

Let me know what you think about this idea!

puzrin, 2019-03-28:
> > As far as i understand, you mean ascent. I'd suggest to dig fontkit's renderer, because real fonts keep much more info. I don't beleive single accent will solve all your "problems".
> 
> What other problems do we have?

Since started experimenting with `lv_font_conv`, i can now give you one of examples.

See image at https://graphicdesign.stackexchange.com/a/8964/72488

Symbols like `Ć`, `(`, `)`, often can exceed ascent (box top) or descent (box bottom). For example, you may have font 16 pixels heigh, but couple of glyphs 17-18 pixels height.

In this case font header must contain additional info - maxHeight and minHeight, to check draw area.

PS. Encoding larger glyphs should not be difficult - allow glyphs optionally have metadata with [xOffset, yOffset, customHeight].

beibean, 2019-04-02:
@puzrin Do you still need some calculations about RLE savings, selecting the best algorithm for each glyph? I think I could have them by the end of this week.

puzrin, 2019-04-02:
@beibean not necessary, only if you wish

beibean, 2019-04-02:
@puzrin In that case I won't do the calculations. If you need it at any point, don't hesitate to ask for it 😉

puzrin, 2019-04-02:
https://github.com/littlevgl/lv_font_conv/ has working reader/dumper in `dev`

1. Reader PROPERLY extract glyphs. Font size XXX gives the same result as in browser. If multiple fonts merged, those are aligned by baseline.
2. Images are "floating" - can exceed expected `fontHeight`*`advanceWidth`.
3. Dumper draws glyphs and show overflows with red background.

Now anyone can quickly analyse source fonts and start experimenting with store/compression.

puzrin, 2019-04-10:
Is this issue still needed?

- General approaches are in https://github.com/littlevgl/lvgl/issues/995
- Technical details about compressions summarized in https://github.com/littlevgl/lv_font_conv/issues/8

kisvegabor, 2019-04-14:
As an additional reference: New Font API: #1006 

As "font cropping" will be part of the new fonts I close this issue.

Thank you very much the contribution and the very interesting conversation here.

closed 2019-04-14 by kisvegabor