lvgl/lvgl #990: New font format requirements

puzrin, 2019-03-28:
**Goals**

Migrate to improved font format, with all required features support + better compression + extendability.

Here is preliminary list of missed font features. Even if features like "combined symbols" are not supported right now, they should not be restricted by font.

**Missed features:**

- Some glyphs CAN exceed ascent/descent/left. This can NOT fit to current "fixed height" format (scale down all is bad idea). Examples, in `Roboto Regular`:
  - exceeds left: `j`
  - exceeds top: `(`, `)`
  - zero width - combining symbols
- No way to add compression and extend with new algorythms
- Font format version check, to show errors on lvgl upgrade if external custom font should be gerenerated.
- No "baseline" info (ascent).
- Split to sparse/continuous fonts is not effective. Should split to chunks instead (continuous chunk / sparse chunk)
  - note, with vary-able glyph data length continuous ranges become less useful. Probably can drop.
- Should substitute missed chars with 0xFFFE char or some black box
- kerning info
- more TBD...

**Uncertain:**

- Probably (?), text format is not very useful. Binary is more "straight" and allows track font size with ease.
- Some glyphs can be duplicated in different languages. Don't know if savings worth efforts.

**Reject**

- Hinting? We rely on FontKit features.
- Sub-pixel smoothing - does not worth efforts been spent, just use High DPI displays instead :)
- Color emoticons :)

---

**Possible solutions**

- Add missed font header fields
  - format version
  - maxBBox info to track exceeds of ascent/descent
  - accent
  - bit's length for some dimentions (width/height/...)
  - compression algorythm ID (+ optional metadata/length)
- Use chunks for glyph ranges, instead of trying make all font sparse or continuous
- Switch to bit's stream for glyphs store. Effective compression is not possible without.
- Allow customize local glyph dimensions (add optional `[xOffset, yOffset, w, h]`). This happens rare, but required.
- (?) Try describe glyph data via "Bounding Box"
  - Need compare benefits for combinations [ BBox in each glyph, compression ] vs [ Optional BBox, compression ]
- Keep font as binary blob (like image), avoid C structs. Price to pay - font headers will move to RAM. Should not be a problem.

TODO:

- Recheck TTF & BDF specs for missed features
- Consider some kind of simplified binary index in chunk metadata, for faster glyphs scan. Full path could be: Font -> Chunk of 256-1024 glyphs -> 4-8 Subranges -> scan glyph

---

Let me know about missed/unclear/disagree items.

**Note:** this is generic issues about "features", not format spec. Spec needs separate issue or may be separates repo.

@kisvegabor this require someone to rewrite lvgl core appropriately. I can do only convertor's part. What do you think?

kisvegabor, 2019-03-29:
> Some glyphs CAN exceed ascent/descent/left. This can NOT fit to current "fixed height" format (scale down all is bad idea). Examples, in Roboto Regular:

I've checked it. `j` really has on x offset.
[image]
  
As I see we need at 4 bytes to describe a glyph with bounding box:
- glyph width (cropped)
- glyph height (cropped)
- glyph x offset (0 in most of the cases)
- glyph y offset (the gap between the top of the font and to the row. Can be negative if has an accent.)

With `ext_size` (an extra area where to object can draw) we can handle it on LittlevGL's side. 

It would mean some extra RAM usage: 4 bytes/ glyph to 8 byte/glyph. With 1000 glyphs 4 kB (still acceptable). 

> No way to add compression and extend with new algorithms

It's possible via the `get_bitmap` and `get_width` callback functions. If we will officially support some kind of RLE besides `sparse` and `continues`, `rle` callbacks should be added to the library.

> Font format version check, to show errors on lvgl upgrade if external custom font should be generated.

You can use `LV_VERSION_CHECK(5,3,0)` to raise an error on incompatibility. But you need to know the current version of LittlevGL. How is it possible?

> Split to sparse/continuous fonts is not effective. Should split to chunks instead (continuous chunk / sparse chunk)

There are chunks with `unicode_first` and `unicode_last`. Every chunk can have its own callbacks.

> note, with vary-able glyph data length continuous ranges become less useful. Probably can drop.

Continuous means the glyph come after each other without gaps. `sparse` was introduced to support CJK when only a few characters are used from big range. It other (Latin-like?) languages `continues` should be used because it's faster and needs less memory (less metadata to find a glyph).

> kerning info

It can be added later in a backward compatible manner with a `kerning_table` field in `lv_font_t`. I suggest leaving it for now.

> Probably (?), text format is not very useful. Binary is more "straight" and allows track font size with ease.

I'd keep the readability of the fonts.  `sizeof(lv_font_dejavu_20_glyph_bitmap)` will tell the size of the bitmap array's size. 

> Some glyphs can be duplicated in different languages. Don't know if savings worth efforts.

It should be handled by the font converter.

puzrin, 2019-03-29:
> As I see we need at 4 bytes to describe a glyph with bounding box:

Not necessary:

- You can set data length in font headers. For example 3 bits for 8x8 monospace font.
- You did not counted savings, produced by bounding box
- You can use bounding box only when necessary

Anyway - that's all small technical details. Key point is - we need

- Additional properties in font header to show mar heal top/bottom
- Dynamic glyph size in some cases, with negative offsets.

If you see no problem to process this in LVGL, i can care about spec & binaries.

> You can use LV_VERSION_CHECK(5,3,0) to raise an error on incompatibility. But you need to know the current version of LittlevGL. How is it possible?

I quess, we need reverse dependencies.

Somewhere in font (blob packed to C file, with some cosmetic guards):

```c
#include "lvgl.h"

LV_IS_FONT_VERSION_SUPPORTED(1.0)
```

> Continuous means the glyph come after each other without gaps. sparse was introduced to support CJK when only a few characters are used from big range. It other (Latin-like?) languages continues should be used because it's faster and needs less memory (less metadata to find a glyph).

Continuous means possibility to use "multiply" for indexed access. That means, each item must have the same size. If glyphs are compressed, each size will be different.

In theory, this may work for very small  1bpp fonts, where compression is difficult. Also, may be some savings if keep descrption tables separate of glyph bitmaps. Need to experiment.

> It can be added later in a backward compatible manner with a kerning_table field in lv_font_t. I suggest leaving it for now.

It can be *implemented* later, but should be verified in current format spec.

> I'd keep the readability of the fonts.

Situation is similar to lv_i18n. You can try to keep human readability (for unknown reasons), or just drop it and make convenient tool first.

Current convertor has `--format dump` to show all info in human readable form.

> `sizeof(lv_font_dejavu_20_glyph_bitmap)` will tell the size of the bitmap array's size.

I talked about all font size, not separate glyph.

Also, i can not compile C code just because i need to know binary size in font convertor to select proper compression :). That means, i will have to create binary writer anyway. Just to know size of all font.

And one more note - binary format will protect you from mistakes, when data is represented as executable code (function pointers and #ifdef-s inside).

> It should be handled by the font converter.

1. Only if profit is significant
2. Requires more complicated tables structs.

I did not estimated size saving with this feature.

---

@kisvegabor let me know general impression:

- Did you find suggestions reasonable?
- Do you have significant disagreement with any?
- Are you ok to implement all (most?) except kerning, in LVGL core?

If no principal objections, i will polish details some time, and switch to format spec writing.

kisvegabor, 2019-03-29:
**Bounding box**
We should handle this in a generic way. Having fonts with or without bounding box would make things more complicated. So having the bounding box concept with the above described 4 parameters is fine for me. There is only one extra requirement: LittlevGL needs to know what is the largest "out of the bounding box size" because the label's draw area needs to be extended with that value.  It can be an `uint8_t extra_area` variable in `lv_font_t`.

**Variable glyph size**
> Continuous means possibility to use "multiply" for indexed access. That means, each item must have the same size. If glyphs are compressed, each size will be different.

If you mean it like this then I agree we don't need to deal with access by "multiplication". `lv_font_glyph_dsc_t` tells where the bitmap starts in the bitmap array.

**Version checking**
> I quess, we need reverse dependencies.
Somewhere in font (blob packed to C file, with some cosmetic guards):

Actually, I'm not sure it is required at all. The fonts should be compatible amongst minor versions and we can expect to re-generate them when a new major version comes. It can be stated in the release note.

**Kerning**
> It can be implemented later, but should be verified in current format spec.

It would require the rework of internal text processing functions to modify the letter space with the kerning values. But seems doable without affecting the public API.

**Bin/Text**
It's much easier to extend the "text" version. For example, adding the kerning table will just mean a new member in `lv_font_t` which is initialized to `NULL` if not set explicitly in an older font. 

Actually, what would binary representation mean? Is it one "big" `uint8_t xy_font[] = {};` with the bitmap and glyph descriptions? Now we have 2 arrays: one for bitmaps and one for glyph info. To know the size of the bitmap you just need to count the bytes written into it. The size of the glyph info array is not relevant as it is not affected by the compression algorithm. 

---
> Did you find suggestions reasonable?
Do you have significant disagreement with any?
Are you ok to implement all (most?) except kerning, in LVGL core?

I categorized the questions above. In summy:
- Bounding box: seems clear
- Variable glyph size: use `glyph_index` from `lv_font_glyph_dsc_t` to tell where a glyph starts
- Kerning: leave now, can be added later.
- Version checking: not required IMO
- Bin/text: need to be discussed but if "bin" is required mainly for the development of the font converter then we should use the text version finally because it's easier to read (better then not readable) easier to extend and already works :)    

So they look fine for me.

> Are you ok to implement all (most?) except kerning, in LVGL core?

The main change from LittlevGL side is the new boxing model. I can add it.

puzrin, 2019-03-29:
Ok, let's drop version check for now, and continue to discuss binary format, that need consensus.

> but if "bin" is required mainly for the development of the font converter

1. Current text format is "write only". After been saved, only C compiler can read it back.
2. I see from another side - convertor may need binary format as "normal" + something "annotated" for strange reasons. There is nothing wrong to wish interoperable format, not pinned with nails to specific language.

As programmer, if i develop binary data, i expect to see it's "natural form" in the end - as binary file, or converted to one "big" `uint8_t xy_font[] = {};`. This is required to verify correctness of write/read.

I'd say, until you can not touch binary with hands you can not guarantee anything about format :)

> then we should use the text version finally because it's easier to read (better then not readable)

Why do you read autogenerated files? Also note, current file looks nice only because current font format is primitive.

> easier to extend and already works :)

Yesteday we discussed how to RIP couple of "easy to extend things" - multibpp & function pointers. You could not do both that way, if been based initially on binary format :).

kisvegabor, 2019-03-29:
> Current text format is "write only". After been saved, only C compiler can read it back.

I think it is not a problem. It was not a goal to make the result work with other languages than C (or not even outside of LittlevGL).
If you want to make the converter more versatile and add binary output too, of course, I have no problem with that, it's your tool. What I don't want is to use the binary format in LittelvGL without any benefit to LittlevGL itself. 

> Why do you read autogenerated files? Also note, current file looks nice only because current font format is primitive.

After the proposed updates (e.g. bounding box) they will be almost similarly easy to read. Making the font file human readable:
- can help with debugging
- contributors will see how it really works/looks like
- 

> Yesteday we discussed how to RIP couple of "easy to extend things" - multibpp & function pointers.

Drop function pointers? Have I missed something? I definitely don't want to drop them. (Dropping multi bpp font files is clear). 

> You could not do both that way, if been based initially on binary format :).
 
I don't understand. Do you tell that binary is less flexible?

--- 

Can you list the advantages of binary (expect the language independence)?

Right now I see two advantages of text version:
- readable: I think it's better to have and can help in some cases
- more flexible: as the font is compiled every time you change things without breaking backward compatibility

@embeddedt What's your opinion about it?

embeddedt, 2019-03-29:
Personally, I would prefer using text format. As @kisvegabor said, it's more readable and flexible to minor changes within LittlevGL itself.

But for the JS converter, using text format is problematic because there is no way to convert it to binary. (Technically, on a PC, you could run `gcc` and `objcopy` in a script, but that wouldn't be available on the web).

@puzrin If the converter needs to be able to read back the generated font to decide things like compression, etc., couldn't the size of the finished font be calculated in advance rather than needing to actually generate a file? That should allow it to produce a text format at the end. It seems a bit strange to me when the converter has to read it's own output to decide how to produce the final version.

puzrin, 2019-03-29:
> I think it is not a problem. It was not a goal to make the result work with other languages than C (or not even outside of LittlevGL).
> If you want to make the converter more versatile and add binary [...]

Imagine, that at certain level of quality requirements, code without tests is not be accepted at all. How could you suggest to verify that created font file is correct?

> Drop function pointers? Have I missed something? I definitely don't want to drop them. (Dropping multi bpp font files is clear).

https://github.com/littlevgl/lv_font_conv/issues/1#issuecomment-475825959

php convertor inject function names. That a mix of data vs code. Result is not serializable.

> I don't understand. Do you tell that binary is less flexible?

They don't allow you mix data with code: use hardcoded fn-pointers and `#ifdef`-s inside. Is it less flexible, is that does not allow use bad practice :) ?

---

My comments to arguments about

> * can help with debugging

- new font dumper could be much more declarative
- when you do unit tests for critical code, there are usually no much need to debug.

> * contributors will see how it really works/looks like

I don.t understand. What should they see? There will be doc with spec about font format and so on.

> * more flexible: as the font is compiled every time you change things without breaking backward compatibility

Just do good data format spec once, without reinventing all from zero. And it will be stable for many years. We dont keep each image as C structs with format details. Why should we do this with font?

embeddedt, 2019-03-29:
@puzrin 

>We dont keep each image as C structs with format details.

I think we do (excerpt from [img_bubble_pattern.c](https://github.com/littlevgl/lv_examples/blob/dev-6.0/lv_apps/demo/img_bubble_pattern.c)):
```c
const LV_ATTRIBUTE_MEM_ALIGN uint8_t img_bubble_pattern_map[] = {
#if LV_COLOR_DEPTH == 1 || LV_COLOR_DEPTH == 8
/* pixel data */
#endif
#if LV_COLOR_DEPTH == 16 && LV_COLOR_16_SWAP == 0
/* pixel data */
#endif
#if LV_COLOR_DEPTH == 32
/* pixel data */
#endif
};
lv_img_dsc_t img_bubble_pattern = {
  .header.always_zero = 0,
  .header.w = 234,
  .header.h = 480,
  .data_size = 112320 * LV_COLOR_SIZE / 8,
  .header.cf = LV_IMG_CF_TRUE_COLOR,
  .data = img_bubble_pattern_map,
};
```

puzrin, 2019-03-29:
@embeddedt 

Your sample is just a trivial raw bitmap. Imagine PNG or JPG with all internal records.

Regarding to your question, why writer should read font. I don't know alternatives to test generated data and proove correctness :).

Binary size calculation without full writer has significant disadvantages:

- complexity is comparable with full writer
- difficult to sync after format change (or you should write all those tests as for full writer)

kisvegabor, 2019-03-29:
> Imagine, that at certain level of quality requirements, code without tests is not be accepted at all. How could you suggest to verify that created font file is correct?

Two things to notice:
1. To be honest I don't think it's a top priority thing right now. Test-driven development is fine but it would require a lot more work and even significant rework of the library. Until there is no stable core development team with  5-10 developers we can't apply this approach because it would mean a huge burden to that few people who maintain the library right now. So it'd be great but it's not a concern right now.
2. I don't think "accepting font files" is a real-life scenario. LittlevGL has its built-in fonts. We don't accept new fonts into the core library. 

>> Drop function pointers? Have I missed something? I definitely don't want to drop them. (Dropping multi bpp font files is clear).

> littlevgl/lv_font_conv#1 (comment)

Still can't see where we were talking about dropping the function pointers. Quote it, please.

> php convertor inject function names. That a mix of data vs code. Result is not serializable.

It's true, it's not serializable. I think it's an other point to "text" version. Having function pointers enables cool features, like using FreeType with LittelvGL: https://forum.littlevgl.com/viewtopic.php?f=12&t=38

> Your sample is just a trivial raw bitmap. Imagine PNG or JPG with all internal records.

However, raw bitmaps cover the needs of 99.9% of LittlevGL users so it's the most relevant use case. `lv_img_dsc_t` is also a general, temporary format used even when a png and jpg image is decompressed. It describes all the info required by the library to render the image (width, height color format, etc. )

In the case of both images and fonts, having `struct`s has the advantage of manipulability. 
Some examples for fonts:
- Having callbacks makes possible to place the bitmap into an external flash and load from there on demand. 
- As mentioned, callback makes possible to integrate FreeType to real-time render ttf fonts.
- Now fonts are concatenated to create a "font-set" (simple linked list). This way an ascii font can be extended. E.g. the same sized built-in fonts are concatenated so if you set `lv_font_dejavu_20` for a label, it can reach `latin`, `cyrillic` and `symbol` fonts too. 
- I turned a normal font to monospace by copying the `lv_font_t` of an existing 20px font and manually set `monospace=12`. It made the font to be rendered with fix width. It was required to display measurement data because it was ugly when the width was changing. This way don't need to add a new typeface and a new font which would require some memory.

embeddedt, 2019-03-29:
Taking into consideration everything that has been discussed, I think that the text version is a better solution.

puzrin, 2019-03-29:
> Two things to notice:
>
> 1. To be honest I don't think it's a top priority thing right now. Test-driven development is fine but it would require a lot more work and even significant rework of the [...]

I don't mean you "must use TDD" in LVGL. I explain how/why i split LVGL into parts:

- Because i have limited time budged and can not rewrite code endlessly when new bugs reported. I have to guarantee some level of quality, even if appropriate part of LVGL on another end is not ready.
- Because i process data in different language and need some level of transparency.

> Still can't see where we were talking about dropping the function pointers. Quote it, please.

https://github.com/littlevgl/lv_font_conv/issues/1#issuecomment-476151205 see from this post, last paragraph. And below + new issue about unicode.

Probably we misunderstood each other, but my question was about why you hardcode function pointers into const data instead of put flag there. Later, you dropped non-UTF8 support and this question disappeared. I thought, those pointers can be dropped as unused.

> In the case of both images and fonts, having structs has the advantage of manipulability.
> Some examples for fonts:

That may happen due language barrier or wrong terms used. Let me try to explain again what i mean.

I'm not against having function hooks in general. I'm against hardcoding those into flash (const) data. But nobody prohibits to create all appropriate objects in RAM when you initialize font:

```c
ram_font_struct = load_font(binary_data)
```

Such approach does not takes a lot of ram, but helps to make accurate isolation between data and flexible features (code/functions).

puzrin, 2019-03-29:
@embeddedt 

> Taking into consideration everything that has been discussed, I think that the text version is a better solution.

I respect your opinion. In this case i could suggest a different thing - i could implement only legacy textual "lvgl5" format in new convertor. And leave binary things + new features for internal experiments. 

Then result could be like this:

- `--format dump` - (now in dev branch) for debug & development, analyze glyphs/fonts
- `--format lvgl5` - legacy textual, C, without "multi-bpp" and new features
- `--format bin1` - binary, to develop new features and compressors.

Convertor will stay self-sufficient & tested. Result (lvgl5 format) will be useable. When someone decide to add something like `--format bin1_lvgl` or core devs decide use binary data in LVGL - we could move forward with new features.

kisvegabor, 2019-03-30:
> Because i have limited time budged and can not rewrite code endlessly when new bugs reported. I have to guarantee some level of quality, even if appropriate part of LVGL on another end is not ready.

It's a fair thing.  

> Because i process data in different language and need some level of transparency.

I understand that but it can't result in losing valuable features in LittlevGL. (E.g. function pointers)

> Probably we misunderstood each other, but my question was about why you hardcode function pointers into const data instead of put flag there. Later, you dropped non-UTF8 support and this question disappeared. I thought, those pointers can be dropped as unused.

Sorry for the confusion, I didn't mean to suggest dropping function pointers. They were introduced to increase the flexibility of the fonts (as you can see from examples above)

> I'm not against having function hooks in general. I'm against hardcoding those into flash (const) data. But nobody prohibits to create all appropriate objects in RAM when you initialize font:

Actually, they are not hardcoded in to flash. See: https://github.com/littlevgl/lvgl/blob/master/lv_fonts/lv_font_dejavu_20.c#L9157

They can't be the flash becasue then `next_page` can't be written to concatenate the fonts. On the other hand, the user can create his custom `lv_font_t` variable, initialize it with ranges, function pointer etc (e.g. to support FreeType) and use it in a style.

> I respect your opinion. In this case i could suggest a different thing - i could implement only legacy textual "lvgl5" format in new convertor. And leave binary things + new features for internal experiments.

We are planning to release `v6.0` this summer so `v5.x` will be outdated in a few months. In `v6.0` we discussed to use a bounding box model and it won't be compatible with `v5.x`. So `--format lvgl6` seems more reasonable. Is it good for you to create the converter for v6.0 (with the discussed modification like the bounding box model, no multi-bpp etc)?

puzrin, 2019-03-30:
> I understand that but it can't result in losing valuable features in LittlevGL. (E.g. function pointers)

Consider all my questions as "how to change format without feature loose". For example, even when we discussed drop multi-bpp from font file, we found how to keep it as feature in LVGL.

> Is it good for you to create the converter for v6.0 (with the discussed modification like the bounding box model, no multi-bpp etc)?

- Right now i don't like to make `--format lvgl6` text writer, but ok to do other things (including binary writer).
- I can do `--format lvgl5` to provide immediate working result, if you wish to stop use php convertor as soon as possible.

Note, we will move forward anyway, independent on my personal preferences. There are lot of work prior "v6 text writer" can become a "problem"

- fix rasterizer/CLI bugs
- font format spec develop (for new features)
- investigate compression details

First iteration, with CLI and `--format dump` should be ready next week. I have no estimate about other things. Initially i did not planned to do those, only to create sandbox to experiment with compressions.

kisvegabor, 2019-03-31:
> Consider all my questions as "how to change format without feature loose". For example, even when we discussed drop multi-bpp from font file, we found how to keep it as feature in LVGL.

Clear, however dropping function pointers would cause feature loss.   
Your prosed `ram_font_struct = load_font(binary_data)` solution if LittlevGL preloads all fonts during initialization but from library's and the user's point of view it's an extra step. I don't want to start the whole argument again. It seems to support multiple `--format`s is the common denominator. 

> I can do --format lvgl5 to provide immediate working result, if you wish to stop use php convertor as soon as possible.

The current version of the converter downscales to fonts to make the accent fint into the given hegiht.  I'm sure it worth the effort to do it and drop it a littel bit later.

Just to be sure: what are the planned features for the font converter? IMO the are three tihnks which are required to make it usabe in v6.0:
1. Have at least the same functionality than now (but with cropped glyphs and improved boxing model)
2. Have a CLI tool
3. Make it work online (running in the client's browser)

Compression and other extra features are optional. 

Is it good for you?

puzrin, 2019-03-31:
> Compression and other extra features are optional.

That does not change too much for me. We have task with breaking changes + need spec update. Each change is very expensive, so doing in small steps is not rationale. IMO the only possible split is:

- Develop spec with full features set
- Postpone implementation of compression

Mocking compression algorythms should not change too much.

kisvegabor, 2019-04-01:
Fine, I agree!

puzrin, 2019-04-04:
Convertor (dev branch) now can read kerning info and collect it to internal structs (can be written later to any format). See example from 16px Roboto dump: 

```json
    {
      "code": 70,
      "advanceWidth": 8.84375,
      "bbox": {
        "x": 1,
        "y": 0,
        "width": 8,
        "height": 12
      },
      "kerning": {
        "44": -1.828125,
        "46": -1.828125,
        "65": -1.328125,
        "74": -2.0625,
        "84": 0.15625,
        "97": -0.265625,
        "99": -0.1640625,
        "100": -0.1640625,
        "101": -0.1640625,
        "103": -0.1640625,
        "111": -0.1640625,
        "113": -0.1640625,
        "114": -0.203125,
        "117": -0.171875,
        "118": -0.1875,
        "121": -0.1875
      }
    }
```

Seems there are additional requirements:

1. Kerning value is fractional
2. Probably advanceWidth (glyph width) should be fractional too for proper offset calculation.

We can NOT round those values before SUM(kern, advanceWidth).

At first glance, Fixed Point with 4 bits after dot would be enougth. Without FP resulting quality can have artefacts.

puzrin, 2019-04-04:
One more update. Switched renderer from `FontKit` to `opentype.js`, and enabled hinting. Result is MUCH better.

How to play with:

```
git clone https://github.com/littlevgl/lv_cont_conv.git
cd lv_cont_conv
npm install
./lv_font_conv.js --size 14 -o 1111 --bpp 2 --format dump --font ./node_modules/roboto-fontface/fonts/roboto/Roboto-Regular.woff --range 0x20-0x7f
```

puzrin, 2019-04-05:
@kisvegabor After experimenting with dumper and anylizing image quality, my personal opinion is:

- Display quality (smoothness) does NOT depend on font size, and depends on bpp only (+ DPI). I mean, if you display large fonts, you can not reduce bpp, because pixelation becomes visible.
- For low DPI (normal displays), `--bpp 2` is not enougth, `--bpp4` is ok.

**Suggestion**

I'd suggest to add support for `--bpp3`:

- Not a problem for new format (because internal data is bit stream anyway)
- Can be exposed as "bpp4" in internal API for convenience
- May save ~20% of data (not sure)

Saving should be tested, because 3bpp data may be less compressible than 4bpp, + other tables overhead. Anyway, support of `--bpp 3` should cost nothing from technical point of view.

In general, i'd suggest to make accent on 2 formats:

1. `--bpp 1` for bitonal displays
2. `--bpp 3` for "High quality Material Design interfaces"

What do you think?

kisvegabor, 2019-04-05:
I agree quality depends on bpp and DPI. 

I've chosen 1/2/4/8 the ease rendering. With bpp = 3 pixels will cross byte borders which makes rendering more difficult. To decide if it is would be required to see texts rendered with 2, 3, 4 bpp next to each other. 

Anyway, my view is that with cropping and some kind of compression we already make a big step. bpp = 3 can be added later too if we are no happy with the compression.

puzrin, 2019-04-05:
> Anyway, my view is that with cropping and some kind of compression we already make a big step. bpp = 3 can be added later too if we are not happy with the size.

IMHO this is very cheap, because bpp = 3 can be returned as bpp = 4 with minimal "decomressor" change. We just need to return bpp value with bitmap data, instead of taking const from font header.

From decompressor's point of view, there is no difference - take 2, 3 or 4 bits from stream.

kisvegabor, 2019-04-08:
If it's hidden with the compression then it fone!

puzrin, 2019-04-08:
I guess, this will be hidden by replacement of current getGlyphBitmap(). New signature should return something like struct with { uint8_t[], bbox, bpp }.

kisvegabor, 2019-04-08:
I see. Let's discuss the API spec in a new issue (with [other things](https://github.com/littlevgl/lvgl/issues/995#issuecomment-480854896)). Can you add an initial suggestion?

puzrin, 2019-04-08:
Ok. I'll reserve first post for you, filling it with imaginated sample.

kisvegabor, 2019-04-10:
@puzrin 
Is there anything else to discuss about the requirements?

puzrin, 2019-04-10:
I'd say those should be verified by font spec discussion:

- did we implemented everything needed?
- did we found something missed in process of spec design?

But having issue open is not critical.

closed 2019-04-10 by puzrin

kisvegabor, 2019-04-11:
Ok, we sill if we have missed something.