lvgl/lvgl #991: Help wanted - VPS configuration

kisvegabor, 2019-03-29:
I have some good news. As @seyyah suggested, when we were talking about the Forum, I registered to  DigitalOcean's Open source Sponsorship program and LittlevGL got $250 support. 

It's enough to run one powerful or two moderate servers for a year. Next year we can ask for support again.

All this is great but I've almost no experience with the "server side world". I'm thinking about security issues, best practices, available tools, etc.

One thing we can use the server(s) is to run Discourse (a modern forum) and see how it works. If it's good and has the features we need we replace the current PhpBB forum with that.

So I'm looking for somebody (or more people) who has some experience in this field and interested in playing with the VPS.

seyyah, 2019-03-29:
I've experienced with VPS, Ruby-on-Rails. And i would like to help/support as much as possible.

We can use $20/mo (12 x $20 = $240) virtual machine if you ask.

[image]

According to the hardware requirements [1],this machine sufficient for Discourse.

- modern single core CPU, dual core recommended
- 1 GB RAM minimum (with swap)
- 64 bit Linux compatible with Docker
- 10 GB disk space minimum

If we want to migrate from phpBB to Discourse, we should follow below guides

- installation with "Discourse One-Click Application" [2]

- import scripts for phpBB [3]

- tutorial for phpBB [4]

## :clipboard: Reference
1. https://github.com/discourse/discourse/blob/master/docs/INSTALL.md#hardware-requirements
2. https://www.digitalocean.com/docs/marketplace/discourse/
3. https://github.com/discourse/discourse/blob/master/script/import_scripts/phpbb3.rb
4. https://meta.discourse.org/t/importing-from-phpbb3/30810

puzrin, 2019-03-29:
I'd suggest to start with 5$/month account. Forum's content is very small and will not be big for a long time.

> As @seyyah suggested, when we were talking about the Forum, I registered to DigitalOcean's Open source Sponsorship program

Were can i read details about? Could not find anything in google.

kisvegabor, 2019-03-30:
@seyyah I'd consider the $10 or the $15 option. As @puzrin pointed out LittlevGL's forum is still small so I think a smaller server will be fine to get started. $5 option is too small because it not fulfills the basic requirements of Discourse. (I started to install Discourse on $5 server and got a warning because of the few RAM)

In addition, it would be nice to save some money to run an other VPS later for any reason.

> Where can i read details about? Could not find anything in google.

[DigitalOcean Open Source Partnerships - One Pager.pdf](https://github.com/littlevgl/lvgl/files/3025190/DigitalOcean.Open.Source.Partnerships.-.One.Pager.1.pdf)

embeddedt, 2019-05-02:
@kisvegabor Does Discourse allow direct transfer of issues from GitHub? That would be convenient and would allow us to avoid restating "Please use the forum" all the time.

seyyah, 2019-05-03:
> @kisvegabor Does Discourse allow direct transfer of issues from GitHub?

@embeddedt it's possible.

> Making it easy to migrate issues between GitHub repos and Discourse [1]

[image]

## :clipboard: Reference
1. https://www.discourse.org/github
2. https://meta.discourse.org/t/introducing-github-issues-to-discourse/46671

kisvegabor, 2019-05-03:
@embeddedt 

It looks like a useful feature!

@seyyah 

Are you still interested in the VPS configuration?

seyyah, 2019-05-07:
I'm interested but nowadays I'm very busy. I need some time. I think i can start after the 15th of the this month.

kisvegabor, 2019-05-08:
@seyyah No problem, take your time!

seyyah, 2019-05-23:
Can we start to installation? Can you add member your digitalocean account/team [1, 2]?

My account: nurettin.senyer@gmail.com, nurettin.senyer

## :clipboard: Reference
1. https://www.digitalocean.com/products/teams/
2. https://www.digitalocean.com/docs/accounts/teams/create/

kisvegabor, 2019-05-24:
@seyyah I added you the team. Does it seem correct?

seyyah, 2019-05-24:
@kisvegabor I created a droplet using one-click app at DigitalOcean with 2 GB RAM (10$/mo).

IP Address: 139.59.153.8

According to [3], can you configure your DNS as subdomain (etc. discourse.littlevgl.com)?

In addition to can you configure email settings [2]?

## :clipboard: Reference
1. https://github.com/discourse/discourse/blob/master/docs/INSTALL-cloud.md
2. https://github.com/discourse/discourse/blob/master/docs/INSTALL-cloud.md#email
3. https://github.com/discourse/discourse/blob/master/docs/INSTALL-cloud.md#domain-name

kisvegabor, 2019-05-25:
@seyyah Great, thank you very much!

> According to [3], can you configure your DNS as subdomain (etc. discourse.littlevgl.com)?
In addition to can you configure email settings [2]?

I can do it on Monday.

seyyah, 2019-05-25:
@kisvegabor 

I have edited the configs according to my own data for testing purposes.
When the threee informations comes from you, I will update.

Test domain: http://forum.sade.app

And i send a invation @kisvegabor

seyyah, 2019-05-25:
> > @kisvegabor Does Discourse allow direct transfer of issues from GitHub?
> 
> @embeddedt it's possible.
> 
> > Making it easy to migrate issues between GitHub repos and Discourse [1]
> 
> [image]
> 
> ##  Reference
> 1. https://www.discourse.org/github
> 2. https://meta.discourse.org/t/introducing-github-issues-to-discourse/46671

I test and it's successful. 

Making it easy to migrate issues between GitHub repos and Discourse:
- Copy the issue to the Discourse destination
- Add references between the two systems
- Close the original issue

seyyah, 2019-05-25:
@embeddedt I also send an invitation your email.

embeddedt, 2019-05-25:
@seyyah I am getting a 521 error: the website is apparently offline. I'll try it again in a few minutes.

seyyah, 2019-05-25:
I'm testing different plugins so i have to rebuild discourse app and server may be return 521 error.

seyyah, 2019-05-25:
I tried some plugins and very excited

Plugins:
- Code review: https://meta.discourse.org/t/discourse-code-review/103142
- Assign: https://github.com/discourse/discourse-assign.git
- Solved: https://github.com/discourse/discourse-solved.git
- voting: https://meta.discourse.org/t/discourse-voting/40121?u=joebuhlig
- push notification: https://github.com/discourse/discourse-push-notifications.git

## :clipboard: Reference
1. https://github.com/discourse

seyyah, 2019-05-25:
> @seyyah I am getting a 521 error: the website is apparently offline. I'll try it again in a few minutes.

can you try again @embeddedt ?

embeddedt, 2019-05-25:
@seyyah Worked this time.. I've made an account.

seyyah, 2019-05-25:
I give an administrator role but you have to verify from your email.

kisvegabor, 2019-05-26:
@seyyah 

I moved the forum to http://forumtest.littlevgl.com

Still working on the SMPT.

kisvegabor, 2019-05-26:
@seyyah I can't login with GitHub:
[image]

Do you know why does it happen?

seyyah, 2019-05-26:
şunu yazdı:

> @seyyah <https://github.com/seyyah>
>
> I moved the forum to http://forumtest.littlevgl.com
>
> Still working on the SMPT.
app.yml and dns configs have to update.

You cannot login with Github before we update configs.

seyyah, 2019-05-26:
şunu yazdı:

> @seyyah <https://github.com/seyyah> I can't login with GitHub:
> [image: image]
> <https://user-images.githubusercontent.com/7599318/58379185-c948d800-7fa0-11e9-9f60-f932b1abc9ab.png>
>
> Do you know why does it happen?
As far as I know

1. We update github oauth apps config
2. Discourse app.yml will be edited
3. Rebuild app

embeddedt, 2019-05-26:
@kisvegabor I created a new OAuth configuration and updated Discourse accordingly. Here's the link (only works for members of the organization): https://github.com/organizations/littlevgl/settings/applications/1071191

@seyyah might still need to change some other setting.

seyyah, 2019-05-26:
> @kisvegabor I created a new OAuth configuration and updated Discourse accordingly. Here's the link (only works for members of the organization): https://github.com/organizations/littlevgl/settings/applications/1071191

@embeddedt I delete my OAuth config. I can login with Github.

I checked the `app.yml` file (`/var/discourse/containers/app.yml`) and there is not problem.

@kisvegabor 

1. we may change SMTP config.
2. If we doesn't use cloudflare like service, we can set up via [1].

## :clipboard: Reference
1. https://meta.discourse.org/t/how-to-set-up-ssl-in-discourse/56577

seyyah, 2019-05-26:
I change the notification email address from `noreply@forum.sade.app` to `noreply@littlevgl.com`

Current SMTP config will be changed,

```yml
DISCOURSE_SMTP_ADDRESS: smtp.eu.mailgun.org
DISCOURSE_SMTP_PORT: 587
DISCOURSE_SMTP_USER_NAME: postmaster@mg.sade.app
DISCOURSE_SMTP_PASSWORD: "Secr3t"
```

Due to this SMTP, sending email is `LittlevGL noreply@littlevgl.com mg.sade.app üzerinden`

seyyah, 2019-05-26:
@kisvegabor OR @embeddedt can you set "code review plugin settings" [1, 2]. It maybe useful for us.

## :clipboard: Reference
1. https://meta.discourse.org/t/discourse-code-review/103142
2. http://forumtest.littlevgl.com/admin/site_settings/category/plugins?filter=code_review

embeddedt, 2019-05-26:
@seyyah I'm not sure if we need this feature... it seems like a duplicate of what we already have on GitHub.

Our vision is to have questions, feature requests, etc. on Discourse but keep development discussions on GitHub.

kisvegabor, 2019-05-26:
@embeddedt 
>@kisvegabor I created a new OAuth configuration and updated Discourse accordingly. Here's the link (only works for members of the organization): 

Thank you. It worked fine. But how do you mean "only works for members of the organization". I was able to log in with an other GitHub account too (name: kisvkriszti) which is not part of this organization.

I also regsitered a user via email and it worked fine and she got the activation email. @seyyah can we keep the mailgun SMTP?

embeddedt, 2019-05-26:
@kisvegabor I meant that only members of the @littlevgl organization can adjust the settings on the application.

kisvegabor, 2019-05-26:
@embeddedt 
I see, thank you for the clarification.

seyyah, 2019-05-26:
> @seyyah can we keep the mailgun SMTP?

As long as we can go Mailgun free plan, no problem.

Github sponsorship waiting list should be used [1].

kisvegabor, 2019-05-27:
> As long as we can go Mailgun free plan, no problem.

Actually, my host gives 100 mails/day for free and 10,000 mail/month for $40/month. 
Mailgun gives 10,000 mail/month for free. If it's not enough we can send the email via DigitalOcean: https://www.digitalocean.com/community/questions/sendmail-limit-for-my-droplet

> Github sponsorship waiting list should be used

I've already applied a few days ago. I'm waiting to see how it will work.

kisvegabor, 2019-05-28:
@seyyah 
I love this forum. It's really much much better PhpBB.
So thank you very much for the setup!

The only big question is the migration of the current forum. According to [this post](https://meta.discourse.org/t/importing-from-phpbb3/30810) migration from PhpBB 3.2 is not supported right now. 
Do you have any idea what to do now?

seyyah, 2019-05-29:
@kisvegabor According to [1], 

> phpBB 3.2 uses a XML based format for storing posts which is completely different from previous versions.

So there is _not_ an easy way to get the import script working again for version 3.2.

But in the Roadmap, "Add support for phpBB 3.2" is planned.

## :clipboard: Reference
1. https://meta.discourse.org/t/importing-from-phpbb3/30810/316

kisvegabor, 2019-05-29:
> But in the Roadmap, "Add support for phpBB 3.2" is planned.

They wrote [this](https://meta.discourse.org/t/importing-from-phpbb3/30810/445) on 11 March. I wouldn't count on the PhpBB 3.2 support in the near future.

It's quite awkward but I can't imagine a better solution than moving the current forum to `forumold.littlevgl.com`, close all topics and direct the users to the new forum.

seyyah, 2019-05-29:
> > But in the Roadmap, "Add support for phpBB 3.2" is planned.
> They wrote [this](https://meta.discourse.org/t/importing-from-phpbb3/30810/445) on 11 March. I wouldn't count on the PhpBB 3.2 support in the near future.

I agree with you.

> It's quite awkward but I can't imagine a better solution than moving the current forum to `forumold.littlevgl.com`, close all topics and direct the users to the new forum.

Right now, the best way is to switch directly to Discourse without migration. Yes it's quite awkward but there is not feaseble way.

kisvegabor, 2019-05-29:
Okay. Still need to create some categories and announcements on the new forum. Is there anything else to do?

seyyah, 2019-05-29:
As far as I know there's no need for more.

embeddedt, 2019-05-29:
@kisvegabor @seyyah I would like to suggest that, if possible, we set a date for when we will officially switch to the new forum, and post a banner on the old forum that it will be moved to `forumold.littlevgl.com` on that date. That way, users won't be surprised when they go to `forum.littlevgl.com` and find a completely new site with no data.

seyyah, 2019-05-29:
I like this idea @embeddedt

kisvegabor, 2019-05-30:
@embeddedt 
I also like the idea. I can add the banner soon. I predict that we can release v6.0 at end of June. What about using the release date as a deadline for the forum too?

amirgon, 2019-05-30:
> Right now, the best way is to switch directly to Discourse without migration. Yes it's quite awkward but there is not feaseble way.

So you are planning to throw away all existing discussions??  
There are ~1500 posts already in the forum. These posts could be useful for people even after a conversation is resolved, since people tend to hit the same problems again and again.  

I would like to raise an unpopular question - **Is Discourse really that better than phpBB?**
Yes, Discourse looks modern and phpBB looks older in its design, but does Discourse really have enough "killer features" that are missing from phpBB? Does Discourse provide that much better user-experience?  
At start, I thought Discourse was much better, but now after using the existing forum with all the plugins for markdown, code highlighting etc, I'm not sure about this any more (but you are welcome to disagree :wink:).  

There are other forums which had the same discussion - "should we migrate from phpBB to Discourse?", for example, see [this thread](https://forums.linuxmint.com/viewtopic.php?t=231767&start=20)  on the Mint forums.  
In their case, even under the assumption that all content would be migrated, not all people were happy with this change. Some liked it very much, others didn't like it at all. Eventually, on the Mint forum case, the migration did not happen although they are discussing this since 2016 till today...

kisvegabor, 2019-05-30:
> So you are planning to throw away all existing discussions??
There are ~1500 posts already in the forum. These posts could be useful for people even after a conversation is resolved, since people tend to hit the same problems again and again.

As @embeddedt suggested: "... it will be moved to `forumold.littlevgl.com` on that date." So not thrown away. It's really not optimal, but people still can find it on Google...

> I would like to raise an unpopular question - Is Discourse really that better than phpBB?

I was the most surprised at how much it is better. I wouldn't believe (and actually didn't believe while we were talking about it earlier) if I hadn't seen it my own eyes. It's really an other level of user experience. PhpBB just looks and feels like something done by developers for developers. Discourse feel something made for "people".  It's only my personal opinion though. I needed to hack the CSS of PhpBB a lot but I wasn't able to make it perfect. Now the UI is fragile and inconsistent.  

In my opinion, (but as we can see from the other discussion there are very different views as well) Discourse is more future proof and it's worth to change. However, the more we wait the more "harmful" it will be.

@amirgon, have you played on the new forum? If yes, what was your impression? Haven't you "felt" the difference I mentioned?

@embeddedt @seyyah (and anybody who reads this :) )
Are you also convinced about it worth to change or you also think the current Forum is fine enough?

embeddedt, 2019-05-30:
> Are you also convinced about it worth to change or you also think the current Forum is fine enough?

I think it's worthwhile to change it. The Discourse user interface feels more like an extension of GitHub, whereas phpBB looks "ancient". :laughing:

However, I understand @amirgon's point: making older discussions available is important.

I think we could provide a banner at the top of Discourse saying that older discussions are available at http://forumold.littlevgl.com.

kisvegabor, 2019-05-31:
Here is the banner:
[image]

I'm a notorious mistyper so, please verify it. :smile: 

I've found an auto-responder plugin for PhpBB which might also help to direct people to the new forum if a new topic has opened. And there is a mass email function too to notify all users about the change.

embeddedt, 2019-05-31:
@kisvegabor Perfect!

kisvegabor, 2019-05-31:
I created the categories: http://forumtest.littlevgl.com/categories

@seyyah The forum runs on `http`. If it's not too difficult could you configure it as `https`?

seyyah, 2019-05-31:
> @seyyah The forum runs on `http`. If it's not too difficult could you configure it as `https`?

@kisvegabor I' using the cloudflare service for this purpose [1].

## :clipboard: Reference
1. https://blog.cloudflare.com/how-to-make-your-site-https-only/

seyyah, 2019-05-31:
> 2\. If we doesn't use cloudflare like service, we can set up via [1].

Alternative method is [1].

## :clipboard: Reference
1. https://meta.discourse.org/t/how-to-set-up-ssl-in-discourse/56577

kisvegabor, 2019-06-02:
@seyyah 
Thank you for the links. Do you have time / are you interested in adding the https support?

kisvegabor, 2019-06-04:
@seyyah 

[This tutorial](https://meta.discourse.org/t/setting-up-https-support-with-lets-encrypt/40709) says:

> Note: ./discourse-setup will enable Let’s Encrypt. And as of March 2017, you can run it again, and press return a few times and enter your email address ; the script will include the required templates and insert your email address as required. Unless you are an expert sysadmin and know a reason not to do that, you should run discourse-setup rather than read any further. (If you installed Discourse a long time ago, you might still have to edit app.yml by hand.)

So it seems it should work by default. However for https://forumtest.littlevgl.com I got `ERR_CONNECTION_REFUSED`.

I could quickly do it if you pointed the right direction. It's the last thing I'd like to do before directing people to the new forum.

kisvegabor, 2019-06-05:
Finally, I was able to add the https support so everything is ready to run the new forum. I added the banner and the autoresponder too to the old forum.

I saw that there are great themes and theme plugins which could be tried but it's a different thing.

Thank you very much, especially for @seyyah, for helping to get started with the Discourse forum and VPS. I hope will have a great time there! :smile:

closed 2019-06-05 by kisvegabor

kisvegabor, 2019-06-05:
@embeddedt GitHub Login is broken again... Somehow I can't make it work. Can you have a look at it?

seyyah, 2019-06-05:
@kisvegabor

I'm on vacation till June 10th. If there is no emergency, when the holiday
is over, I might be interested.

şunu yazdı:

embeddedt, 2019-06-05:
@seyyah He edited his post to say it is now working, but thanks!

kisvegabor, 2019-06-06:
@seyyah All seems to work now. Have a nice holiday!