lvgl/lvgl #992: I can't receive confirmation mail.

3038922, 2019-04-01:
Sorry, I can't receive confirmation mail.
my username 3038922
my email 31612534@qq.com
[image]

I want to make a terminal . Like this. I want to ask what module should be used to achieve?
hiden==true
[image]
hiden==false
[image]

embeddedt, 2019-04-01:
@kisvegabor would have to look into the account problem.

As for the terminal question, I don't quite understand what you mean? Are you trying to replicate that user interface on LittlevGL?

3038922, 2019-04-01:
I just want to implement a drop-down animation that can open a terminal.

Now I'm using the button to hide and display the terminal. But he doesn't look cool...

embeddedt, 2019-04-01:
@3038922 You should build your own animation with `lv_anim_t`. You can move the object's Y position so that it slides off or on to the screen.

If you just want a basic animation you can try `LV_ANIM_IN` or `LV_ANIM_OUT`.

closed 2019-04-01 by 3038922

3038922, 2019-04-01:
> @3038922 Please, try the "Resend activation mail" link on the login page and check your spam folder too.
I've tried more than ten times. I couldn't get a confirmation email last month. It's the same today.
Can you change it for me in the database?

kisvegabor, 2019-04-01:
I've activated you manually.