lvgl/lvgl #993: NOT suport multiple spinbox like roller

crazyuav, 2019-04-02:
1.   I create three spinbox,and add them to group.

`
	/*create three spinbox*/
    app.current.spinbox = lv_spinbox_create(lv_scr_act(), NULL);
    app.voltage.spinbox = lv_spinbox_create(lv_scr_act(), app.current.spinbox);
    app.power.spinbox = lv_spinbox_create(lv_scr_act() , app.current.spinbox);
    ...
    ...
    ...
     /*add to group*/
    lv_group_t * g = lv_group_create();
    lv_group_add_obj(g, app.current.spinbox); 
    lv_group_add_obj(g, app.voltage.spinbox); 
    lv_group_add_obj(g, app.power.spinbox); 
    lv_group_set_editing(g, false);
    lv_indev_set_group(indev_encoder, g);
    lv_indev_set_group(indev_keypad, g);
`

2. modify spinbox function**lv_spinbox_signal**

`    else if(sign == LV_SIGNAL_RELEASED) {
        lv_group_t * g = lv_obj_get_group(spinbox);
        bool editing = lv_group_get_editing(g);
        if(editing) lv_group_set_editing(g, false);     /*In edit mode go to navigate mode if an option is selected*/
        
#if 0    
        lv_indev_t * indev = lv_indev_get_act();
        if(lv_indev_get_type(indev) == LV_INDEV_TYPE_ENCODER) {
            if(lv_group_get_editing(lv_obj_get_group(spinbox))) {
                if(ext->step > 1) {
                    lv_spinbox_step_next(spinbox);
                } else {
                    /*Restart from the MSB*/
                    ext->step = 1;
                    uint32_t i;
                    for(i = 0; i < ext->digit_count; i++) {
                        int32_t new_step = ext->step * 10;
                        if(new_step >= ext->range_max) break;
                        ext->step = new_step;
                    }
                    lv_spinbox_step_previous(spinbox);
                }
            }
        }
#endif        
    }`

3.then press encoder
step1: press LV_GROUP_KEY_ENTER  ->edit mode
step2: press LV_GROUP_KEY_ENTER  ->navigate mode
step3: turn right/turn left three times -> generate a bug enter HardFault_Handler
because  the function **focus_next_core** call the **lv_obj_get_hidden**,then generate HardFault_Handler

embeddedt, 2019-04-02:
(cc @kisvegabor)

crazyuav, 2019-04-03:
It works，Thanks!

on branch dev-6.0, modified  **lv_group_style_mod_func_t** callback function ,and add a new parameter **group**,but I don't synchronize with callback

closed 2019-04-03 by crazyuav