lvgl/lvgl #995: New font format spec

puzrin, 2019-04-03:
This issue is to stabilize new font format at binary level. It accommodates features, discussed in #990.

Comments and suggestions welcome.

---

## Font header

Size (bytes) | Description
-----|-------------------------------------
2 | Record size (for quick skip)
2 | Height
2 | Ascent
2 | BBoxMax X
2 | BBoxMax Y
2 | BBoxMax Width
2 | BBoxMax Height
1 | Glyph BBox x/y/w/h bits length
4 | Min Unicode charcode
4 | Max Unicode charcode
1 | Compression alg ID (0 - raw bits)
2 | total chunks
11 | chunk1 header
11 | chunk2 header
...|...

Chunk header:

Size (bytes) | Description
-----|-------------------------------------
1 | Chunk type (continuous, sparse BMP, sparse full)
4 | Min Unicode charcode
4 | Max Unicode charcode
2 | Entries count

## Chunk data

Internal offsets are 2-bytes, so chunk length should not exceed 64K.

TBD: kerning info?

### Continuous chunk

For compact sequences representation.

Size (bytes) | Description
-----|-------------------------------------
2 | Record size (for quick skip)
2 | Glyph1 data offset (entry 1)
2 | Glyph2 data offset
...|...
XX | glyphs bitmap data

CAN have "missed" glyphs: offset = -1.

### Sparse chunks, BPM range (0...65535)

Size (bytes) | Description
-----|-------------------------------------
2 | Record size (for quick skip)
2 | Glyph1 code
2 | Glyph1 data offset
...|...

### Sparse chunks, full range (0...0x10FFFF)

Size (bytes) | Description
-----|-------------------------------------
2 | Record size (for quick skip)
4 | Glyph1 code
2 | Glyph1 data offset
...|...

## Glyph data

Stream of bits.

Note, bounding box is NOT related to typographic area (height * advanceWidth),
and can be shifted anyhow. Real image can overflow typographic space.

Size (bits) | Description
-----|-------------------------------------
NN | Typographic width (length in font header)
NN | BBox X (length in font header)
NN | BBox Y (see length in font header
NN | BBox Width (see length in font header)
NN | BBox Height (see length in font header)
?? | Compressed bitmap

TBD:

- BBox dimentions length is usually 4-5 bits.
- Should we use separate dimentions for overflowed value
  (those are 1-2 bits larger)?

puzrin, 2019-04-03:
Initial idea of keeping chunk headers into font head was possibility of binary search. But this does not seems very useful

- because number of chunks will be small
- cache can be implemented in runtime

I tend move chunk headers back into chunks, to have this info more "condensed"

puzrin, 2019-04-03:
Moved chunk heads to chunk bodies.

https://github.com/littlevgl/lv_font_conv/blob/font_spec/font_spec.md - here is my "sandbox" of updates, is anyone interested.

puzrin, 2019-04-04:
After investigating kerning info, i found data to become too complicated. Tend to pack it more close to OpenType tables (without chains).

1. Glyph data should have internal continuous enumeration (not pinned directly to char codes)
2. Features of "continuous"/"sparse" compact store is related to charcodes only, not to glyph bitmaps. In this case it has equivalent in OpenType spec (subtables).

**Pros:**

- More friendly to next extensions
- More easy to understand/verify
- Reduce "inventing new standard" attempt => less design errors
- (?) able to deduplicate idedtic glyphs from different scriptings (unicode blocks)

**Cons**

- May need some state store in RAM (pointers to active tables and so on) for effective work

puzrin, 2019-04-04:
Updated first message with new format, based on OpenType principles.

History: https://github.com/littlevgl/lv_font_conv/commits/font_spec

@kisvegabor, @embeddedt, @beibean, I think this format can be useable. Need to discuss, check missed features and stabilize.

kisvegabor, 2019-04-05:
@puzrin This format will be used internally, and writes can convert it to the required format, right?

puzrin, 2019-04-05:
IMHO it would be nice to use it in LVGL directly.

Anyway, when we design complex data, we should think in "bytes" domain, not in specific language. That's more clear, more easy to apply external patterns (from OpenType in our case), to quick-check features availability (like binary search) and so on. I've completely rewritten spec 3 times, this would not be so easy if it was in C.

Probably, this data can be expressed in C, but for me this would be a waste of time. If someone provide me correct example of such translation - i probably could try to create such writer. But i would strongly recommend to start with redesign of public API (font expression in C is internal thing).

We are still in the middle of of new fonts data/API design. It's too early to speak about C writer. I tried to explain all process here https://github.com/littlevgl/lv_font_conv/issues/7#issuecomment-480119718

puzrin, 2019-04-05:
Under new API i mean something like:

- font->render(string)
- font->getGlyphId(charCode)
- font->getBitmapInfo(id)
- font->getKerning(left_id, right_id)

That's only an example, only to demonstrate that existing API should be reworked, to reflect new data/features. Don't consider as real methods.

kisvegabor, 2019-04-05:
It sounds reasonable to update the font API this way, however, as I stated in other posts it can be part of an n other release.

puzrin, 2019-04-05:
@kisvegabor i don't agree. It does not worth to plan jump over chasm with two jumps :). Data format change is ass pain every time. API rework is MUCH more simple, than change data twice.

To be constructive, i suggest to create new issue about font API rework, collect requirements and design function signatures. This does not involve coding, allows verify almost everything and do estimates about cost. If we find cost not acceptable - we can start search plan B. But new format is very modular and should not be difficult to switch.

puzrin, 2019-04-05:
For example: i you find decompressor implementation difficut, we can support "raw bits" in 6.0 release, and add compressor in 6.1. This will not affect existing spec and API.

puzrin, 2019-04-06:
Updated TBD section with questions. Need to discuss.

kisvegabor, 2019-04-08:
> To be constructive, i suggest to create new issue about font API rework, collect requirements and design function signatures. This does not involve coding, allows verify almost everything and do [...]

Sounds good. Please open a new issue for the new font API and we will see if it's possible to add it in `v6.0`

kisvegabor, 2019-04-14:
In light of the progress in the API topic, we need to clarify what is the exact purpose of this data representation format. 

Anyway, to make some progress here I answered the general questions of the TBD section: 

>  Coordinate system. Where to place (0, 0) ? Baseline? Ascent? Descent?

 I vote for the ascent. It's the most natural while drawing. We should have "baseline" attribute too (e.g. `baseline == 12`) to enable positioning the label with its baseline.

> Should we keep advanceWidth in separate table? (need measure size difference)
current location seems ok for quick lookup.

I see no disadvantage of storing `advanceWidth` in different tables. When people need a monospace font they usually have a resource constraint environment. So saving memory, in this case, is important.

> Need samples for CJK subsets, to understand need of spatial subtables (still can
use lot of holed Subtable Format 6 with binary search for fast lookup).
> Do we need something special for proper CJK render?

As you already know currently we have `continuous` and `sparse` fonts. `Sparse` is tested with CJK by a lot of people and they had no problem with that. So I think we don't need anything else than supporting sparse fonts.

> Should we care about data align? What is the size cost of it?

Yes, sections/tables should be aligned to 4 bytes boundary.

puzrin, 2019-04-14:
> In light of the progress in the API topic, we need to clarify what is the exact purpose of this data representation format.

That's a part of "standard" design process:

[ collect use cases ] => [ design data flow ] => [ design api/code ]

1. This data reflects required features (https://github.com/littlevgl/lvgl/issues/990).
2. This format helps to verify logical correctness.
3. It's used to design writer (font convertor) & reader (lvgl api).

> I vote for the ascent. It's the most natural while drawing. We should have "baseline" attribute too (e.g. baseline == 12) to enable positioning the label with its baseline.

Just to clarify, you suggest set (0, 0) to upper left corner? And glyph will be drawn "right and down" (except case with negative shifts)?

> I see no disadvantage of storing advanceWidth in different tables.

I'd say, i could not find reason to create additional table. OpenType's one is for more complicated features, not used in our case.

> When people need a monospace font they usually have a resource constraint environment. So saving memory, in this case, is important.

That's not related to storage location. Just set number of bits in `advanceWidth` to 0 + add default value field to font header.

> As you already know currently we have continuous and sparse fonts. Sparse is tested with CJK by a lot of people and they had no problem with that. So I think we don't need anything else than supporting sparse fonts.

Can i see examples from real world? I need to understand subsets of used charcodes

That's required to understand do we need additional subtable format, or we can use multiple of existing subtables.

> Yes, sections/tables should be aligned to 4 bytes boundary.

Ok, i will align everything according to addressed data width

The only question is with `glyf` table. It would be nice to aligh each "glyph bitstream start" to 4 bytes, but this can add notable overhead. Need to compare losses for real fonts.

kisvegabor, 2019-04-14:
> That's a part of "standard" design process:
[ collect use cases ] => [ design data flow ] => [ design api/code ]
- This data reflects required features (#990).
- This format helps to verify logical correctness.
- It's used to design writer (font convertor) & reader (lvgl api).

I agree it is useful the verify logical correctness but not sure it's required to specify the exact format (bits, sizes, alignment). However, if you think it will help later, I believe. :)

> Just to clarify, you suggest set (0, 0) to upper left corner? And glyph will be drawn "right and down" (except case with negative shifts)?

Yes.

AdvanceWidth table:
> That's not related to storage location. Just set number of bits in advanceWidth to 0 + add default value field to font header.

Ok, if it works here. Probably in `lv_font_t` it really should be a different table but don't mix LittlevGL specific things here.

> Can i see examples from real world? I need to understand subsets of used charcodes

I can't link any exact resources. I just saw here that people sent screen shoots with CJK characters to ask something. 

>That's required to understand do we need additional subtable format, or we can use multiple of existing subtables.

I'm not sure subtables really could work in this case because characters can be really sparse. My experience is CJK symbols, because of mean words, can't be grouped to logical subsets. Have a look at [this](https://en.wikipedia.org/wiki/CJK_Unified_Ideographs_(Unicode_block)). The symbols are just divided into 4 equal parts.

puzrin, 2019-04-14:
> I agree it is useful the verify logical correctness but not sure it's required to specify the exact format (bits, sizes, alignment). However, if you think it will help later, I believe. :)

If people can understand something wrong - they will do it :). It's much more cheap to write exact spec, than do multiple code rewrites and try to find difference between JS writer and C reader :).

> Ok, if it works here. Probably in `lv_font_t` it really should be a different table but don't mix LittlevGL specific things here.

If you speak about `head` table - that's for internal implementations and NOT `lv_font_t`. I guess, it can partially go to extension of `lv_font_t`, specific for LVGL font engine.

Note, font spec is not about particular implementation. It's about data to make features possible. `head` store assorted properties, because those are required.

> I can't link any exact resources. I just saw here that people sent screen shoots with CJK characters to ask something.
>
>
> I'm not sure subtables really could work in this case because characters can [...]

Ordered tuples with delta-coded index should be ok:

```c
subtable_format_sparsed[] = {
  { uint16_t (CJK1 - range_start), uint16_t glyph_id1 }
  { uint16_t (CJK2 - range_start), uint16_t glyph_id2 }
  ...
}
```

- 4 bytes per entry (every subtable covers up to 65336 wide range).
- Ready for binary search.

The only reason why i did not added this yet - there are no direct analogues in OpenType spec, and i can't understand why.

https://docs.microsoft.com/en-us/typography/opentype/spec/cmap there format2 & format4 but those break my brain :).

puzrin, 2019-04-14:
Refreshed spec according to comments. Key points:

- Updated header info (drop bboxMax, leave Y min/max only)
- Improve cmap subtables descriptions
- Add sparsed subtable format
- Aligned everything possible
- Rearranded kerning table to simplify aligned access

Diff: https://github.com/littlevgl/lv_font_conv/commit/43c6f315cd61542ba142bf6a1cd84c97b93add58

kisvegabor, 2019-04-14:
> Ordered tuples with delta-coded index should be ok:

Something like that could work.

> There are Format2 & Format4 but those break my brain :).

It really looks complicated. I don't think we need to follow OpenFont so strictly. 

Glyph's bitmap alignment:
As they will be stored and used byte-based I don't think alignment is required there. Now it's not aligned and there were no problems with that. A problem could occur if an `uint32_t` would be read from a not 4 aligned address. But for bytes, it should be fine.

puzrin, 2019-04-14:
> Glyph's bitmap alignment:
> As they will be stored and used byte-based I don't think alignment is required there. Now it's not aligned and there were no problems with that. A problem could occur if e.g. an `uint32_t` would be read from a not 4 aligned address. But for bytes, it should be fine.

Content of glyph data is bit stream. Normal operation usually is:

- prefetch some bytes
- peak requested number of bits, until new prefetch needed

With aligned data we could prefetch 4 bytes at once. I guess, there should not be big difference, but byte access looks not nice.

> 
> The updated spec look good to me. Do we need this issue opened?

Yes, i'd like to keep it open for 1-2 weeks.

- Too few feedback in general.
- We did not collected feedback from API issue.
- Need to decide where to land spec document.

kisvegabor, 2019-04-15:
> With aligned data we could prefetch 4 bytes at once. I guess, there should not be big difference, but byte access looks not nice.

I see no advantage of accessing it as 32 bit data. You can consider the glyphs data as an `uint8_t` array:
```c
const uint8_t * p = glyph_bitmap[glyph_start];
p[x] ... ;

``` 

> Yes, i'd like to keep it open for 1-2 weeks.

Ok

puzrin, 2019-04-15:
See updated TBD. Should we expose recommended glyph buffer size into header? This looks too "low-level-ish".

- Rigth values is `MAX(all_glyphs)`
- Each glyph's value is `wordAlign(bitmap_width) * bitmap_height` (`bitmap_width` depends on `bpp`)

This will require to scan `loca` / `glyf` tables.

I can't explain why i don't like to hardcode allocator hint into font.

kisvegabor, 2019-04-17:
I think storing this information is useful and I see no problem with that. It's better if do it instead of letting it to the user.

puzrin, 2019-04-20:
Updated spec. Found some minor issues after started with bin writer.

- missed bpp info :)
- BBox X/Y are signed, BBox W/H are unsigned. Splited bits length to separate filelds.
- advanceWidth MAY be in fixed point format (if kerning info used, last 4 bits should be fractional). Added format field + appropriate notes. For "ugly" OLED fonts this may be integer.

[Diff](https://github.com/littlevgl/lv_font_conv/commit/f376eca20328d61d7a61db5625c725c5b6e811d7)

puzrin, 2019-04-21:
https://iamvdo.me/en/blog/css-font-metrics-line-height-and-vertical-align - this is very useful article to understand font align inside of line.

I'm not sure we need features like font mixtures and different aligns, but still worth to read.

kisvegabor, 2019-04-22:
I'll check it!

puzrin, 2019-04-23:
Quick summary about "proper" default font positioning inside line

```
$ env DEBUG=* ./lv_font_conv.js --font ./node_modules/roboto-fontface/fonts/roboto/Roboto-Regular.woff -r 0x20-0x7f --bpp 3 --size 24 --format dump -o 1111
```

24px dump will give this params:

```json
{
  "ascent": 22,
  "descent": -6,
  "size": 24,
}
```

I will use terms from [article](https://iamvdo.me/en/blog/css-font-metrics-line-height-and-vertical-align).

- EM Box (1em) is 24px.
- Content box [ascent + descent] is 28px (bigger). Note, some glyphs still can be larger.

Normal line height is usually 1.2-1.5em => 29-36px. By default, EM Box (and Content Area) are centered inside of line.

Content Area (28 height) offset from line top will be 0.5-3.5px.

Currently we agreed to count glyph bitmap (x, y) offset relative to content area (top, left).

kisvegabor, 2019-04-23:
I read the article. Just a few things to be sure we understand things in the same way:

1. `"size": 24` doesn't really matter while drawing or calculating things, right?
2. No character will be larger than 28 px? So any text will fit into a 28 px heigh box?
3. `"ascent": 22` means the baseline is 22 px down from the top of the 28 px line box?  
4. Now `style->text.line_space` means the gap between two lines. 
It should be updated to `style->text.line_height`. 
The default value should be e.g. `font->size * 1.2`.

puzrin, 2019-04-23:
>  1. `"size": 24` doesn't really matter while drawing or calculating things, right?

Yes. It's intendend to implement design with ease. So, when designer says "This screenshot uses Roboto Regular 16px" - you will get the same look in LVGL, without need to scale, shift and so on.

> 2. No character will be larger than 28 px? So any text will fit into a 28 px heigh box?

There are a lot of confusion about this. If you open FontForge, there THREE types of ascent/descent.

- General asc/desc - used to draw top/bottom lines when you open glyph
- Win asc/desc - (?) for Windows, includes space to line top, descent is positive (inverse)
- HHead asc/desc - (?) for Mac, similar to Win asc/desc, negative descent. May be different, but not significantly.

We need some time to understand, how to properly scale/center font.

> 3. `"ascent": 22` means the baseline is 22 px down from the top of the 28 px line box?

To be more precise - fonts have no top. They have baseline and lot of metrics, counted from baseline.

> 4. Now `style->text.line_space` means the gap between two lines.
>    It should be updated to `style->text.line_height`.
>    The default value should be e.g. `font->size * 1.2`

- Yes, it's much better to operate with line-height
- Usually, line-height should be set according with used font-size, to get "round" result. That also depends on design.

kisvegabor, 2019-04-23:
> There are a lot of confusion about this. If you open FontForge, there THREE types of ascent/descent.

> To be more precise - fonts have no top. They have baseline and lot of metrics, counted from baseline.
```
> {
>   "ascent": 22,
>   "descent": -6,
>   "size": 24,
> }
```

It was for the whole font, not for a single glyph, wasn't it? If yes, based on the article, it should mean the
`content area = 22 - (-6) `.

puzrin, 2019-04-23:
> It was for the whole font, not for a single glyph, wasn't it? If yes, based on the article, it should mean the
> `content area = 22 - (-6) `.

Yes.

But we expected this to be "general ascent/descent", and js libraries seems to return HHead asc/desc. Difference is significant:

- General asc: 1536
- General desc: 512
- EM: 2048
- OS/2 Typo Ascent: 1536
- OS/2 Typo Descent: -512
- OS/2 Typo Line Gap: 102
- OS/2 Win Ascent: 1946
- OS/2 Win Descent: 512
- OS/2 HHead Ascent: 1900
- OS/2 HHead Descent: -500
- OS/2 HHead Line Gap: 0

We need couple of days to understand what hell happens. And which metrics use to scale down.

puzrin, 2019-04-23:
Collapsed all previous commits to hide garbage.

Also, did some changes to reflect latest thought and ideas:

- Added "typographic" ascent/descent/linegap
- Changed subtable headers format:
  - use "range" instead of "last code", to save 2 bytes
  - use saved bytes to store Glyph ID offset. That will allow use `Format 0` "everywhere", and mark `Format 6` as "reserved" (tend to drop, but not ready to do it now)
  - Updated fields order to have longest (4 bytes) first.

[Diff](https://github.com/littlevgl/lv_font_conv/commit/4acad613a5a7e2d920f7f7cd00721b273ac46ce6)

**Why "typoXX" metrics**

Current ascent/descent reflect MAX possible top/bottom + include line gap in ascent. So, align inside line may change, if you add letters like `Ȃ` and others. Typographic ascent is more close to "upper line of capital letters" and not depends on artefacts.

I don't mean we must use typoXX metrics, but prefer to reserve 6 bytes of flash and avoid format change if we need experimenting.

Also, i don't know what to do if we merge 2 fonts with different metrics. Now we just scale down each font according to their `EM Units` and use metrics from first one. Should be enough for us.

To count intersection with framebuffer region - use `minY` and `maxY` - those are always recalculated for all glyphs.

Convertor script is not yet updated, but will reflect changes soon.

puzrin, 2019-04-24:
Updated convertor. Now seems ok.

Since different ascents can be used, it seems more reasonable to define bitmap's bbox `y` offset from baseline - it's more "stable" and more "common" (as in vector fonts).

Any principal objections?

puzrin, 2019-04-26:
Updated convertor, added tests and fixed some bugs. Now binary outpot should be correct.

Current coordinates of glyph's bitmap bounding box are:

- (x, y) defines (bottom, left) corner.
- BBox (0, 0, ...) is at (left side, baseline)
- Axis directions are (right, up)

Bitmap's content is from (top, left) to (right, down), as usual.

kisvegabor, 2019-04-26:
> Since different ascents can be used, it seems more reasonable to define bitmap's bbox y offset from baseline - it's more "stable" and more "common" (as in vector fonts).

I agree.

> (x, y) defines (bottom, left) corner.

Why the bottom corner? Using the right - down axis direction is more common. Is there any reason to do it differently in this case?

puzrin, 2019-04-26:
> Why the bottom corner? Using the right - down axis direction is more common. Is there any reason to do it differently in this case?

2 reasons:

- More close to OpenType fonts (and font editors). I don't like to reinvent wheels and new standards without need.
- Not depend on prefered ascent type (you can switch from "typography ascent" to "hhead ascent", without need to update coordinates).

puzrin, 2019-04-26:
Image, to avoid confusion:

[image]

kisvegabor, 2019-04-27:
However, it's the same for letters with "special lower part", right? For example `Ų` or `Ç`.

puzrin, 2019-04-27:
> However, it's the same for letters with "special lower part", right? For example `Ų` or `Ç`.

`y = 0` is always at baseline. `g`, `j`, `Ų`, `Ç` will have negative `y`

kisvegabor, 2019-04-27:
I mean it seems to me the problem still persists just now with "HHead Descent" and "Typo Descent".

puzrin, 2019-04-27:
I'd say, this coordinates may seems strange to one who did not worked with fonts. But that's not fatal problem and not technical difficulty.

- Usually, you already have layout and know exact linebox where to fit line of text.
- Then you can use `typoAscent` + `typoDescent` to estimate "visual height" of text, align this inside line box and count baseline position.

kisvegabor, 2019-04-27:
I need to know the `content-area` to set the label's size accordingly. It should include all accents because the letters can be drawn only on the label's area. I can modify this line's height with the `line_height` from the style to get the `line_box`.   

In summary, I need to know 3 things:
1. the absolute maximum height of a line (virtual_area ?). Probably HHead ascent + HHead descent
2. the baseline relative to the maximum height
3. Glyph's BBox y offset. It can represent the top or bottom edge of the box, doesn't really matter.

puzrin, 2019-04-27:
Hm... i don't see question in your post. What is the problem with?

kisvegabor, 2019-04-27:
Oh, sorry. :) The question is: does it "work" this way? Or in other words: Is it possible to and do you agree to create the listed data for the built-in font format?

puzrin, 2019-04-27:
> Is it possible to and do you agree to create the listed data for the built-in font format?

Not sure i understand you again :). See first post, font header table section. All this data seems already exist, and generated in `--format bin`.

https://github.com/littlevgl/lv_font_conv/blob/50c8664844085b0e7bd4ce585f6a269ee23a6a90/lib/font/table_head.js#L53-L62

Dit i missed something?

kisvegabor, 2019-04-28:
> Not sure i understand you again :). See first post, font header table section. All this data seems already exist, and generated in --format bin.

I wanted to be 100% sure that we will have an "ultimate line-height" because it's the most important from LittelvGL's point of view. But then it seems fine. Thank you.

Next week I'll make some experiments with kerning too. When it's ready, I'll be able to provide a template file for LittlevGL built-in fonts. You mentioned that you are not interested in creating an lvgl6 text writer. Is this still true?

puzrin, 2019-04-28:
> You mentioned that you are not interested in creating an lvgl6 text writer. Is this still true?

The most boring part is inventing good text format. I'm not interested in spending time for this. But if someone provides ready format, i could try to make JS part, if that helps.

Under "ready format" i mean "no magical numbers" (except array sizes). For example, i have no ideas how to define data offsets in `glyf` & `cmap` tables.

kisvegabor, 2019-04-29:
It would b great if you created the JS part. I will make the ready text format. Is it good if we will have a small example font? Like (this)[https://github.com/littlevgl/lvgl/blob/font/src/lv_fonts/lv_font_dejavu_20.c].

puzrin, 2019-04-29:
> Is it good if we will have a small example font?

Yes. In case of "text format", example font with couple of dummy glyphs should be enougth.

kisvegabor, 2019-04-29:
Awesome!

puzrin, 2019-04-30:
Need to decide where to land font format spec:

- `lvgl` repo
- `lv_font_conv` repo

`lv_font_conv` would be more logical if script will be used independent, with other libs (to keep project consistent). In other cases `lvgl` may be preferable.

I'll be ok with any choice, if i have write access to spec next 1-2 months, for polish.

kisvegabor, 2019-05-01:
> lv_font_conv would be more logical if script will be used independent, with other libs (to keep project consistent). In other cases lvgl may be preferable.

I agree. Please upload it there.

puzrin, 2019-05-01:
Landed to https://github.com/littlevgl/lv_font_conv/blob/master/doc/font_spec.md

Seems all done, but i'd suggest keep issue open until API finished. To discuss data change & clarification requests (i hope this will not happen but can not guarantee 100%).

puzrin, 2019-05-04:
**Kerning table size**

Usually ~50% of glyphs has kerning. Size of kerning table in [Format 0](https://docs.microsoft.com/en-us/typography/opentype/spec/kern#format-0) has quadratic dependency on glyphs count.

If we dump ALL Roboto glyphs (edge case), size will be huge:

```
$ env DEBUG=* ./lv_font_conv.js --font ./node_modules/roboto-fontface/fonts/roboto/Roboto-Regular.woff -r 0x20-0x10FFFF --size 16 --bpp 3 --format bin -o test.bin
  font last_id: 894 +0ms
  font minY: -4 +2ms
  font maxY: 18 +1ms
  font glyphIdFofmat: 1 +0ms
  font kerningFormat: 0 +25ms
  font advanceWidthFormat: 1 +0ms
  font xy_bits: 5 +3ms
  font wh_bits: 5 +1ms
  font advanceWidthBits: 10 +4ms
  font monospaced: false +0ms
  font indexToLocFormat: 0 +32ms
  font.table.head table size = 44 +0ms
  font.table.cmap table size = 1412 +0ms
  font.table.loca table size = 1800 +0ms
  font.table.glyph table size = 32484 +0ms
  font.table.kern 447 kerned glyphs of 893, 413 max list, 52248 total pairs +0ms
  font.table.kern table size = 261252 +1ms
  font font size: 296992 +161ms
```

Simple optimization (with 2-dimentional array) will reduce size to ~160K only.

As far as i understood, real kernig is defined not for glyphs but for classes (groups of glyphs). For example for `AO`, `AC` & `AQ` chars `O`,`C`,`Q` belongs to the same class.

- See [Format 2](https://docs.microsoft.com/en-us/typography/opentype/spec/kern#format-2).
- Also see [Format 3](https://developer.apple.com/fonts/TrueType-Reference-Manual/RM06/Chap6kern.html), it seems more easy to understand, and more compact

So, we have to store:

1. Mapping chars to Left Class ID (893 bytes), ~ 40 classes
2. Mapping chars to Right Class ID (893 bytes), ~ 40 classes
3. Table leftClassesCount * rightClassesCount * kern_value_size, ~1600 or 3200 bytes.

Big difference!

Of cause, this may work only if original font uses definition via table. But all modern fonts should have it (because that's more convenient for font authors).

I did not investigated details, may be for ascii Format 0 will be more compact.

---

Second point is "micro optimization" for kerning value size. 8 bits is not enougth for huge fonts (> 50px). Currently spec switches to 16-bits values.

Alternative may be introduce property `kerningMultiplier`. Actually, 7-bits  (+sign) resolution is enougth, if used with proper scale. Requires trivial FP multiplication (hardware int mul support should be available in all lvgl targets).

Just an idea.

This also open ability to create "font packs" with different font sizes and shared kerning table. BUT, at first glance, after add `Kerning subtable Format 3` support, saving will be not significant.

---

Next 1-2 weeks will try to invent "table restorer from glyph pairs" algorythm.

puzrin, 2019-05-04:
Checked `cmap` subtables for full Roboto dump:

`12 subtable(s): 9 "format 0", 3 "sparse"`

That means, in worst case 4 binary search iterations required to find subtable. Nice. Should not affect performance.

puzrin, 2019-05-05:
https://github.com/littlevgl/lv_font_conv/commit/e4e0a8504d2922379bc5cd1f7c4705cbb613f493

Updated spec & code to use `kerningScale` FP12.4. That will simplify upcoming kerning subformats changes.

---

Also pushed to temporary `kern` branch calculation of glyph "classes" (grouping) stat, required for subtables format2/format3 support.

0x20-0x7f:

```
$ env DEBUG=* ./lv_font_conv.js --font ./node_modules/roboto-fontface/fonts/roboto/Roboto-Regular.woff -r 0x20-0x7F --size 16 --bpp 3 --format bin -o test.bin
  ...
  font.table.glyph table size = 2836 +0ms
  font.table.kern 51 kerned glyphs of 95, 40 max list, 434 total pairs +0ms
  font.table.kern unique right classes: 40 +8ms
  font.table.kern unique left classes: 35 +2ms
  font.table.kern table size = 1316 +0ms
  font font size: 4524 +25ms
```

0x20-0x10FFFF:

```
$ env DEBUG=* ./lv_font_conv.js --font ./node_modules/roboto-fontface/fonts/roboto/Roboto-Regular.woff -r 0x20-0x10FFFF --size 16 --bpp 3 --format bin -o test.bin
  ...
  font.table.glyph table size = 32484 +0ms
  font.table.kern 447 kerned glyphs of 893, 413 max list, 52248 total pairs +0ms
  font.table.kern unique right classes: 83 +116ms
  font.table.kern unique left classes: 103 +124ms
  font.table.kern table size = 261252 +1ms
  font font size: 296992 +412ms

```

In `Format 3` kerning will be:

- ascii - 96\*2 + 40\*35 => 1592 bytes
- full -> 893\*2 + 83\*103 => 10335 bytes

Note, for unknown reasons Roboto stores data in 2 subtables (GPOS, pairs + format2). And it's format2 table dimension is ~40*40 (we have twice more). Need some checks what happens, but even preliminary results looks nice.

puzrin, 2019-05-05:
https://github.com/littlevgl/lv_font_conv/commit/70784b624d4dd965a2f16661ce3ccce2ab0c6532

Upcoming update of `Format 3` for kerning table.

kisvegabor, 2019-05-06:
I've also seen that usually using classes are used to save memory. It seems more effective but we need to ensure that the converter works with **any font**. At worst case, it should skip the kern table if it's format is not supported. 

**Width byte size**
I'm not sure 2 bytes are really required. Probably 2 bits for the fractional part would be enough too:
- 1 bit sign
- 5 bit integer part
- 2 bit fractional part

This we will have -31.75 .. 31.75. It should be enough for 100 px fonts too.

puzrin, 2019-05-06:
> It seems more effective but we need to ensure that the converter works with any font.

It will select most optimal format of possible. I don't understand why you decided in can not work with some fonts.

> At worst case, it should skip the kern table if it's format is not supported.

Sounds like "magic". Convertor works as it can, and write least possible size. It's user's responsibility to take modern font.

> Width byte size
> I'm not sure 2 bytes are really required. Probably 2 bits for the fractional part would be enough too:

Are you about width or kerning? Kerning is 0.1 in many cases of 16px font. I don't like to restrict resolution with 2 bits fractional. And i don't like to restrict max possible size with 200px (some chars will have kerning > 31)

kisvegabor, 2019-05-07:
> Sounds like "magic". Convertor works as it can, and write least possible size. It's user's responsibility to take modern font.

I absolutely don't agree. We need to add as few restrictions as possible. If the user wants to use a font for any reason we need to do our best to make possible the use of that fonts.

> Are you about width or kerning? Kerning is 0.1 in many cases of 16px font. I don't like to restrict resolution with 2 bits fractional. And i don't like to restrict max possible size with 200px (some chars will have kerning > 31)

Limiting the font size around 200 px and 0.25 px resolution is reasonable compared to the cost of an extra byte. With proper rounding the error will be <= 0.125 px which is very acceptable.

puzrin, 2019-05-07:
> I absolutely don't agree. We need to add as few restrictions as possible. If the user wants to use a font for any reason we need to do our best to make possible the use of that fonts.

Seems i missed the point of discussion. Initial (source) font  font are correct, and author draw it with appropriate features (with of without kerning, depending). Changing author intents is not good. Convertor keep those features and write file of minimal size. What's wrong?

> Limiting the font size around 200 px and 0.25 px resolution is reasonable compared to the cost of an extra byte. With proper rounding the error will be <= 0.125 px which is very acceptable.

IMO, you discuss problem that not exists.

https://github.com/littlevgl/lv_font_conv/commit/e4e0a8504d2922379bc5cd1f7c4705cbb613f493

Current spec has scale property, 8-bit kerning values and no restriction about font size.

[9 more replies]

closed 2019-05-09 by kisvegabor