lvgl/lvgl #1001: text is hown mirrord via /dev/fb0

btrhad, 2019-04-04:
# All enhancement requests or questions should be directed to the Forum.

We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

btrhad, 2019-04-04:
I am using a framebuffer as display, and followed the littlevgl/framebuffer tutorial.
The text "Hello World" however is shown upside down and mirrored on the display
If I cat a bmp image to /dev/fb0 it is displayed correctly. Is there any configuration/setting that I need to use to let littlevgl know how the display configuration/orientation is?

embeddedt, 2019-04-04:
On Linux you change the rotation with `/sys/class/graphics/fb0/rotate`. You should be able to use that. Valid values range from 0 to 3.

closed 2019-04-04 by embeddedt