lvgl/lvgl #997: [6.0] LV_EVENT_CLICKED should not trigger when scrolling a list

embeddedt, 2019-04-04:
I have an `lv_list` object on the screen. Every list button is assigned an event handler that performs an action when `LV_EVENT_CLICKED` is fired.

The problem I'm seeing is that if I start scrolling the list, when I let go, the button that I touched to begin scrolling triggers `LV_EVENT_CLICKED`.

(cc @kisvegabor)

kisvegabor, 2019-04-04:
I've fixed and tested it. IS it working there too?

embeddedt, 2019-04-04:
@kisvegabor No, it's still not working for me. Both `LV_EVENT_CLICKED` and `LV_EVENT_LONG_PRESSED` are still being triggered when the list is scrolled.

embeddedt, 2019-04-04:
@kisvegabor I just had a thought. Is it because `LV_EVENT_CLICKED` and related events are sent from `indev_proc_release` and the related functions in `lv_indev.c`?

kisvegabor, 2019-04-04:
I fixed it (again) in the last commit. It was a general issue not related only to list. 
SHould work now :smile:

embeddedt, 2019-04-04:
@kisvegabor Much better now. The only remaining issue is if you try to scroll past the edge, it still triggers whatever button you touched. I think that's because the list isn't moving.

kisvegabor, 2019-04-05:
> The only remaining issue is if you try to scroll past the edge, it still triggers whatever button you touched. I think that's because the list isn't moving.

It is the intended behavior because the events are not called if the page object was dragged. And the "drag flag" is set only if the object was really moved.

embeddedt, 2019-04-05:
@kisvegabor I found a fix that seems to solve the issue. In `lv_indev.c` I removed the requirement to have the object moving to set the "drag flag". This doesn't appear to have broken anything. It seems to be working fine from my tests.

I will add the fix to the `obj_drag_dir` branch in #1003.. can you test it as well?

kisvegabor, 2019-04-08:
> I found a fix that seems to solve the issue. In lv_indev.c I removed the requirement to have the object moving to set the "drag flag". This doesn't appear to have broken anything. It seems to be working fine from my tests.

I have a vague memory that I did it this way intentionally. But it seems fine to me now. We will see in the practice :)

Just `pop_inv_buf` part should remain in the code as I described in the PR's comment.

embeddedt, 2019-04-09:
Fixed by #1003.

closed 2019-04-09 by embeddedt

patricklaf, 2019-11-04:
I am not sure that the problem is solved.
I use last version of lv_arduino 2.0.3, with STM32 Arduino core. I have a list populated with filenames. When I click on the list to scroll it, often if not allways, LV_EVENT_CLICKED is triggered and the file is selected.

embeddedt, 2019-11-04:
@patricklaf Send a simple code sample which can reproduce the issue and I'll look into it.

patricklaf, 2019-11-06:
@embeddedt thanks for your comment, this is the code.
This is the function to create the UI:
```cpp
void lvgl_setup(void) {

	// Toolbar zone
	{
		lvgl_toolbar = lv_btnm_create(lv_scr_act(), NULL);
		lv_obj_set_event_cb(lvgl_toolbar, lvgl_button_action);
		lv_obj_set_size(lvgl_toolbar, LV_HOR_RES, 40);
		lv_obj_align(lvgl_toolbar, NULL, LV_ALIGN_IN_TOP_MID, 0, 0);
	}

	// Files zone
	{
		// Create a scroll bar style
		static lv_style_t style_sb;
		lv_style_copy(&style_sb, &lv_style_pretty_color);
		style_sb.body.padding.right = 5;
		style_sb.body.padding.inner = 10;

		// Create styles for the buttons
		static lv_style_t style_btn_rel;
		static lv_style_t style_btn_pr;
		lv_style_copy(&style_btn_rel, &lv_style_btn_rel);
		style_btn_rel.body.main_color = LV_COLOR_MAKE(0x30, 0x30, 0x30);
		style_btn_rel.body.grad_color = LV_COLOR_BLACK;
		style_btn_rel.body.border.color = LV_COLOR_SILVER;
		style_btn_rel.body.border.width = 1;
		style_btn_rel.body.border.opa = LV_OPA_50;
		style_btn_rel.body.radius = 0;

		lv_style_copy(&style_btn_pr, &style_btn_rel);
		style_btn_pr.body.main_color = LV_COLOR_MAKE(0x55, 0x96, 0xd8);
		style_btn_pr.body.grad_color = LV_COLOR_MAKE(0x37, 0x62, 0x90);
		style_btn_pr.text.color = LV_COLOR_MAKE(0xbb, 0xd5, 0xf1);

		// Create the list
		lvgl_files = lv_list_create(lv_scr_act(), NULL);
		lv_obj_set_size(lvgl_files, tft.width(), tft.height() - 40);
		lv_obj_align(lvgl_files, lvgl_toolbar, LV_ALIGN_OUT_BOTTOM_MID, 0, 0);

		// Apply styles
		lv_list_set_sb_mode(lvgl_files, LV_SB_MODE_AUTO);
		lv_list_set_style(lvgl_files, LV_LIST_STYLE_BG, &lv_style_transp_tight);
		lv_list_set_style(lvgl_files, LV_LIST_STYLE_SCRL, &lv_style_transp_tight);
		lv_page_set_style(lvgl_files, LV_LIST_STYLE_SB, &style_sb);
		lv_list_set_style(lvgl_files, LV_LIST_STYLE_BTN_REL, &style_btn_rel);
		lv_list_set_style(lvgl_files, LV_LIST_STYLE_BTN_PR, &style_btn_pr);
	}

	// Play zone
	{
		// Create the container
		lvgl_play = lv_cont_create(lv_scr_act(), NULL);
		lv_obj_set_style(lvgl_play, &lv_style_pretty);
		lv_obj_set_size(lvgl_play, tft.width(), tft.height() - 40);
		lv_obj_align(lvgl_play, lvgl_toolbar, LV_ALIGN_OUT_BOTTOM_MID, 0, 0);
		lv_obj_set_hidden(lvgl_play, true);

		lvgl_filename = lv_label_create(lvgl_play, NULL);
		lv_label_set_long_mode(lvgl_filename, LV_LABEL_LONG_SROLL);
		lv_obj_set_width(lvgl_filename, tft.width() - 20);
		lv_obj_align(lvgl_filename, NULL, LV_ALIGN_IN_TOP_LEFT, 10, 10);

		lvgl_size = lv_label_create(lvgl_play, NULL);
		lv_label_set_long_mode(lvgl_size, LV_LABEL_LONG_SROLL);
		lv_obj_set_width(lvgl_size, tft.width() - 20);
		lv_obj_align(lvgl_size, lvgl_filename, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 10);

		lvgl_motor = lv_cb_create(lvgl_play, NULL);
		lv_cb_set_text(lvgl_motor, "Moteur");
		lv_obj_align(lvgl_motor, lvgl_size, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 10);

		// Create a bar
		lvgl_bar = lv_bar_create(lvgl_play, NULL);
		lv_obj_set_size(lvgl_bar, tft.width() - 20, 30);
		lv_obj_align(lvgl_bar, NULL, LV_ALIGN_IN_BOTTOM_MID, 0, -10);
	}
}
````
The function to populate the list:
```cpp
void lvgl_files_load(void) {
	char buffer[FILENAME_SIZE];
	Stack<String*> folders;
	Stack<String*> files;

	// Clean the list
	lv_list_clean(lvgl_files);
	// Build the list
	fat.vwd()->rewind();
	fat.vwd()->getName(buffer, FILENAME_SIZE);
	// Loop files
	while (file.openNext(fat.vwd(), O_RDONLY)) {
		file.getName(buffer, FILENAME_SIZE);
		if (!file.isHidden()) {
			if (file.isDir()) {
				folders.push(new String(buffer));
			} else {
				files.push(new String(buffer));
			}
		}
		file.close();
	}
	// Sort, folders then files, alphabetical order
	folders.sort(compare);
	files.sort(compare);
	// Load
	for (unsigned int index = 0; index < folders.count(); ++index) {
		lv_obj_t *button = lv_list_add_btn(lvgl_files, LV_SYMBOL_DIRECTORY, folders[index]->c_str());
		lv_obj_set_event_cb(button, lvgl_folder_action);
	}
	for (unsigned int index = 0; index < files.count(); ++index) {
		lv_obj_t *button = lv_list_add_btn(lvgl_files, LV_SYMBOL_FILE, files[index]->c_str());
		lv_obj_set_event_cb(button, lvgl_file_action);
	}
}
```
And the two callbacks for the buttons:
```cpp
void lvgl_folder_action(lv_obj_t *button, lv_event_t event) {
	if (event == LV_EVENT_CLICKED) {
		worker.filename = (char *)lv_list_get_btn_text(button);

		if (file.open(worker.filename, O_RDONLY)) {
			worker.state = state_folder;
			file.close();
		}
	}
}

void lvgl_file_action(lv_obj_t *button, lv_event_t event) {
	if (event == LV_EVENT_CLICKED) {
		worker.filename = (char *)lv_list_get_btn_text(button);

		if (file.open(worker.filename, O_RDONLY)) {
			worker.state = state_file;
		}
	}
}
```
And finally, the input driver function:
```cpp
bool lvgl_input_read(lv_indev_drv_t * drv, lv_indev_data_t *data) {
	UNUSED(drv);
	static lv_point_t last = { 0, 0 };

	TSPoint point = ts.getPoint();

	pinMode(XM, OUTPUT);
	pinMode(YP, OUTPUT);

	if ((point.z > 200) && (point.z < 1000)) {
		data->point.x = map(point.x, TS_LEFT, TS_RT, 0, tft.width());;
		data->point.y = map(point.y, TS_TOP, TS_BOT, 0, tft.height());
		data->state = LV_INDEV_STATE_PR;
		last = data->point;
	} else {
		data->point = last;
		data->state = LV_INDEV_STATE_REL;
	}
    return false;
}
```
With this code, when I try to scroll the list, it begins to scroll a little and then the file or folder action is triggered unexpectedly.

embeddedt, 2019-11-06:
@patricklaf Can you simplify your application code to just the necessary code to reproduce the issue (i.e. just create a list with some static elements)? There are quite a few APIs you've used (`fat.vwd()`) that I don't have access to.

patricklaf, 2019-11-06:
This is the modified code, as requested:
```cpp
#define SIZE 10

void lvgl_files_load(void) {
	const char *folders[SIZE] = {
				"folder00", "folder01", "folder02", "folder03",	"folder04",
				"folder05", "folder06", "folder07", "folder08",	"folder09"
		};
	const char *files[SIZE] = {
				"file00", "file01", "file02", "file03",	"file04",
				"file05", "file06", "file07", "file08",	"file09"
		};

	// Clean the list
	lv_list_clean(lvgl_files);
	// Load
	for (unsigned int index = 0; index < SIZE; ++index) {
		lv_obj_t *button = lv_list_add_btn(lvgl_files, LV_SYMBOL_DIRECTORY, folders[index]);
		lv_obj_set_event_cb(button, lvgl_folder_action);
	}
	for (unsigned int index = 0; index < SIZE; ++index) {
		lv_obj_t *button = lv_list_add_btn(lvgl_files, LV_SYMBOL_FILE, files[index]);
		lv_obj_set_event_cb(button, lvgl_file_action);
	}
}
```

kisvegabor, 2019-11-08:
I've tested your code with this `event_cb` and it worked well for me:
```c
void event_cb(lv_obj_t * obj, lv_event_t e)
{
    if(e == LV_EVENT_CLICKED) printf("Clicked\n");
}

```

patricklaf, 2019-11-08:
@kisvegabor  that is what I feared.
What can cause my behaviour? LV_INDEV_STATE_REL returned to early? Buttons, in list and matrix, are working well. The only problem is this erratic scroll.

kisvegabor, 2019-11-10:
Could you print the coordinates and state (released/pressed) before `return false;`?

patricklaf, 2019-11-13:
@kisvegabor you give me the solution: sometimes when keeping button press, the touchscreen returns pressure as 0 so a release state is generated. I saw that with prints in `lvgl_input_read`.
I increased `NUMSAMPLES` in the [TouchScreen](https://github.com/adafruit/Adafruit_TouchScreen) library. It's not perfect but it helps.
So it's definitely not a problem with LittlevGL.
@kisvegabor, @embeddedt thanks for your help.

kisvegabor, 2019-11-15:
> you give me the solution

Glad to hear that! :slightly_smiling_face:

commit aec671dc (2019-04-04): lv_list: don't sent CLICK event if dragged. fix #997
PR #1003 (2019-04-09): Enhance drag logic
This also incorporates #1002 since it was needed for testing.

This pull request introduces the ability to constrain object drag along a specific axis. This is useful for pages that should only scroll in one direction.

The latest commit also completely fixes #997, as the list object will now consider scrolling past the edge as drag.
commit 2794b786 (2019-04-15): Merge pull request #1 from littlevgl/dev-6.0

lv_list: don't sent CLICK event if dragged. fix #997