lvgl/lvgl #999: Allowing styles to be const

canardos, 2019-04-04:
On most MCUs, `const` objects can be stored in flash rather than occuping precious RAM. A typical LVGL application has numerous `lv_style_t` objects, which are rather large and do not change during program execution, and are thus a prime candidates to be made `const`. Storing these styles in RAM seems unecessary, but as the code stands, it's currently not possible to make them `const`.

For an application recently, I made my styles `const` and modified all of the functions that accept lv_style_t types (e.g. `lv_xxx_set_style` etc.) to be `const`, but after the program crashed I quickly discovered that in (at least) one area of the code, the style in use is temporarily modified.

Specifically in `lv_page_design`, the page background style is modified for a few specific drawing operations and then reverted:

```
/*Draw without border*/
lv_coord_t border_width_tmp =  style->body.border.width;
style->body.border.width = 0;
lv_draw_rect(&page->coords, mask, style, lv_obj_get_opa_scale(page));
style->body.border.width = border_width_tmp;
...
/*Draw only a border*/
lv_style_t * style = lv_page_get_style(page, LV_PAGE_STYLE_BG);
lv_coord_t shadow_width_tmp =  style->body.shadow.width;
uint8_t empty_tmp =  style->body.empty;
style->body.shadow.width = 0;
style->body.empty = 1;
lv_draw_rect(&page->coords, mask, style, lv_obj_get_opa_scale(page));
style->body.shadow.width = shadow_width_tmp;
style->body.empty = empty_tmp;
``` 

As far as I can tell, this function is the only one I noted that modifies the styles, but there may be others.

I'm wondering what it would take to avoid this and thus allow all styles to be immutable. This could free all of the RAM used by styles and would also have the added benefit of not needing to deal with guarding this if multi-threading support is added at some point.

Temporarily copying the style would work, but seems like overkill. Modifying the style before calling `lv_draw_rect` doesn't really feel like the right solution in the first place, but I'm not familiar enough with the painting workflow to really understand why it's required and the possible alternatives.

With my current changes, I can make all styles except `LV_PAGE_STYLE_BG` `const`, but it's a bit of a hack. @kisvegabor, any thoughts?

embeddedt, 2019-04-04:
> Temporarily copying the style would work, but seems like overkill.

A better solution, in my opinion, would be to regenerate the extra modified styles every time `lv_page_set_style` is called and keep them in the object's `ext_attr` area. This would avoid the need for a temporary copy.

As far as I can tell the cost of the extra two `lv_style_t` objects per page would be offset significantly by the removal of all other default `lv_style_t` objects from RAM.

canardos, 2019-04-04:
> As far as I can tell the cost of the extra two lv_style_t objects per page would be offset significantly by the removal of all other default lv_style_t objects from RAM.

You could get even away with a single mutable copy of the `LV_PAGE_STYLE_BG` style and modify it in the same way as it's done currently. It's just one or two instructions during the call, which is probably preferable to the extra copy.

As you said, the RAM savings here are all of the built in styles, plus any user created ones, minus one style for each live page object. It's quite significant.

embeddedt, 2019-04-04:
@canardos One issue I see is that the way themes are initialized would need to be redesigned. Here's an example:

https://github.com/littlevgl/lvgl/blob/master/lv_themes/lv_theme_default.c

This kind of inheritance wouldn't work with constant styles.

canardos, 2019-04-04:
You're right that the initialization code would need to be changed for the built in styles to take advantage (i.e. get them out of RAM), but it wouldn't be necessary to make those changes initially if it's an issue.

As a first step, the function parameters could be made `const` and the changes discussed above made to the page component. That would at least allow (but not require) custom or other styles to be `const` and avoid them taking up RAM. As a second step, the built-in styles/themes could be migrated. The `const` function parameter additions are good practice regardless.

With respect to the built-in styles/themes, most of my code is C++, where I am able to copy, modify, and assign existing styles to new `static const` vars at compile time, but to my knowledge this is not possible in C. Initializing static objects in C require constant expressions and existing `const` objects wont work.

I'm not a C expert, but I would think C99 designated initializers in combination with defines for shared values is probably the cleanest and most maintainable method. It would result in more verbose source code for the style definitions, but smaller compiled code (and quicker startup) compared with the current code, since they would now just be static objects in the .text/.rodata section, requiring no copying/initialization on startup.

embeddedt, 2019-04-04:
@canardos Sounds like a good plan. Would you like to make a pull request that changes the parameters to enforce `const`?

kisvegabor, 2019-04-05:
@canardos It's a good idea to make styles and `lv_xxx_set_style` parameters const.

In case of `lv_page`, I see no problem with copying the style into a local variable. I've already done this with several objects.

I agree to deal with themes later once this basic step is done. 

So please send a PR! The whole source code was changed in `dev-6.0` because code formatter was run so don't forget to pull the lates version.

canardos, 2019-04-06:
Will do.

With regard to copying to a local variable, did you mean making a copy on the stack in the  `lv_page_design` function, or a permanent copy that remains with the object in its attributes?

kisvegabor, 2019-04-08:
> Will do.

Great!

> With regard to copying to a local variable, did you mean making a copy on the stack in the lv_page_design function, or a permanent copy that remains with the object in its attributes?

A copy on the stack is enough in this case. The computational overhead is smaller then the memory overhead would be.

canardos, 2019-04-08:
> A copy on the stack is enough in this case. The computational overhead is smaller then the memory overhead would be.

I agree. Will try to get it submitted this week.

kisvegabor, 2019-04-11:
The PR is merged so I close this issue.

Thank you very much!

closed 2019-04-11 by kisvegabor