lvgl/lvgl #1004: Find TTF replacements for current embedded fonts

puzrin, 2019-04-05:
With upcoming new font format and tools it worth to find replacements for current embedded fonts and revisit build process.

Goals:

- Instead of providing C sources only with unknown generation option, there should be clear process of generation (via `Makefile.js` or anything else suitable for core devs)
- We should know exact font source and licence.
- Decide if users really need terminal-like fonts or not.

Additional needs:

- Describe exactly which fonts should be built-in and why (should we provide bitonal 8x8 terminal fonts and so on).
- Suggest how to quickly add pixel-perfect fonts (may be, collection of links or separate repo to keep core clean).
- Check licenses and attribution requirements.

Plan B.

In worst case i could add BDF support to convertor for monospace fonts of fixed size, but it really worth move build process for TTF/WOFF only.

---

If we tend use "modern proportional fonts" then Roboto is perfect choice. But, if we care about bitonal OLED displays, then terminal fixed size fonts required. In this case i'd expect to collect here links to such fonts in TTF/WOFF format.

Example:

- http://terminus-font.sourceforge.net/ (BDF only)
- Google search "Terminus TTF font" (did not tested in different sizes)

nacyzhouw, 2019-04-08:
How about using FreeType?

nacyzhouw, 2019-04-10:
@kisvegabor 
Do you mean the FreeType would take large memory in most of the embedded sysems? I have successfully tryed using FreeType in VS simulator to  dynamically and directly generate the bitmap from TTF font. Yes, if transplantating in the embedded sysems, there may be more work need to be done.

And do you have any suggestion that a method instead of FreeType, which using less memory but can dynamically and directly generate the bitmap from TTF font?

puzrin, 2019-04-10:
@nacyzhouw this issue was created NOT about freetype, but about font sources and prefered out of box packaging.

kisvegabor, 2019-04-10:
@nacyzhouw I found this on GitHub (not tried) https://github.com/mcufont/mcufont
If you have further questions about this topic, please continue on the [Forum](https://forum.littlevgl.com/)

@puzrin 
I don't insist on Dejavu. We can compare it with Robot later.

puzrin, 2019-04-10:
@kisvegabor to you have some vision of target deveces for LVGL? Do we need to care about bitonal OLEDs? That's the only case when pixel-perfect monospace fonts required. From my point of view, bitonal OLEDs are outdated ~ as AVR.

kisvegabor, 2019-04-11:
Besides OLEDs:
1. simple monochrome displays are still used because they are very cheap.
2. eInk displays also have 1bpp color depth and they are widely used 

So we should have a built-in pixel perfect font (1 size could be enough) to make possible to quickly do some experiment on these displays too.

puzrin, 2019-05-02:
not stale

kisvegabor, 2019-05-09:
Let's check how Robot looks like when the font converter is ready.

Terminus TTF also looks good.

---

Let's continue in #1057

closed 2019-05-09 by kisvegabor

puzrin, 2019-05-09:
@olikraus posted interesting links in https://github.com/olikraus/u8g2/issues/890#issuecomment-491020003

- http://www.fontspace.com/total-fontgeek-dtf-ltd/glasstown-nbp - pixel perfect font in TTF format
- https://github.com/olikraus/u8g2/wiki/fntgrpnbp - list of other fonts, i guess, all available as TTF on the same site.

Seems use only ttf/woff as source is possible, and we can send bdf to hell :)

olikraus, 2019-05-09:
There are actually much more fonts than this:

https://github.com/olikraus/u8g2/wiki/fntgrp

or here:

https://github.com/olikraus/u8g2/wiki/fntlistall

embeddedt, 2019-05-09:
> Seems use only ttf/woff as source is possible, and we can finally stop using bdf :)

Why? If people are already using BDF in their projects (or they find a BDF font they want to use), then we shouldn't be restricting them to only using TTF, especially since BDF support is already present.

puzrin, 2019-05-09:
BDF is not supported in php script. And, to be honest, i don't like spend my own time to add this support. Only if alternatives are not available at all.

I don't like abstract "universality". Prefer unification instead. If all can be done via TTF - it's better focus our limited resources on TTF first.

kisvegabor, 2019-05-10:
BDF is really supported by an [other tool](https://github.com/littlevgl/lv_utils/blob/master/bdf_font_converter.py).

> f people are already using BDF in their projects (or they find a BDF font they want to use), then we shouldn't be restricting them to only using TTF

I agree, it really can happen. However, I think in this case the bdf converter should be updated to support the new format. For simple fonts (like ASCII monochrome), the new format is not so complicated than for "complex fonts" (like Roboto).

puzrin, 2019-05-10:
I'd suggest to check initial reasons first. LVGL has tails of some "strange" (from my point of view) things, with better alternatives available. Sometime it worth to cut those, to simplify maintenance and be able apply efforts to more important features.

I guess, BDF convertor was done because proper TTF (pixel-perfect) sources were not found in time.

embeddedt, 2019-05-10:
> I guess, BDF convertor was done because proper TTF (pixel-perfect) sources were not found in time.

Nevertheless, it exists now, so people might be using it. I don't think it's right to remove features like that, especially if people may be relying on them for their projects.

@puzrin If we really want to drop BDF support, then we should provide a converter to convert from BDF to pixel-perfect TTF fonts. Is that possible?

puzrin, 2019-05-10:
> Nevertheless, it exists now, so people might be using it. I don't think it's right to remove features like that, especially if people may be relying on them for their projects.

In ideal world, it would be nice to support everything and more forever. But who will pay for it :)? All things, been done "free" happens because authors has personal interest.

If switch from BDF to TTF not acceptable, people may stay at previous lvgl version.

> @puzrin If we really want to drop BDF support, then we should provide a converter to convert from BDF to pixel-perfect TTF fonts. Is that possible?

There are some convertors, but i don't know how those work - never used. They may embed bitmaps instead of build vector outlines of pixels. Try to google "bdf to ttf" and dig sources.

I don't consider situation as BDF drop. New convertor is completely different thing, with "everything done right". It was more simple to do everything from scratch, instead of try fix old one. And i don't like idea to keep compatibility with all possible ancient kludges, instead of do things "right and simple".

But... all sources are available, anyone can try to improve, if he has time for it.

embeddedt, 2019-05-10:
@puzrin I did some quick research:

* https://github.com/koron/bdf2ttf - embeds the bitmaps like you mentioned. It seems like an acceptable compromise for backwards compatibility. Does the TTF converter support embedded bitmaps?
* https://github.com/Tblue/mkttf - creates the outlines of the pixels. One issue I did read was that depending on the font the quality might be lower.

puzrin, 2019-05-10:
> Does the TTF converter support embedded bitmaps?

No. It uses vectors. Embedded bitmaps is a very strange extention, actually not used anywhere.

kisvegabor, 2019-05-10:
Let's ask @BrianPugh the author of the bdf converter. 
Brian, can you mention some use cases when BDF can't replace by pixel perfect TTF fonts?

BrianPugh, 2019-05-13:
I'm not opposed to dropping BDF support if TTF offers comparable alternative. Pixel-perfect is indeed very important when working with low resolution lcd/oled screens. I haven't been keeping super up to date on the font discussion, I'll have to do some reading.

kisvegabor, 2019-05-14:
Thank you @BrianPugh.