lvgl/lvgl #1007: How to use task mutex?

ripe909, 2019-04-08:
I just updated my lvgl codebase from quite an old version.  Now I see that my task handler always returns before doing anything because it fails to obtain a mutex and my screen never updates.

Is there an example somewhere of how to use this mutex correctly?  All of the example code looks the same, and I don't see any mutex handling.  It looks like it obtains the mutex on the first run, then hits lv_task_run == false and returns, then every subsequent call returns immediately because the mutex is now true.

`LV_ATTRIBUTE_TASK_HANDLER void lv_task_handler(void)
{
    LV_LOG_TRACE("lv_task_handler started");

    /*Avoid concurrent running of the task handler*/
    static bool task_handler_mutex = false;
    if(task_handler_mutex) return;
    task_handler_mutex = true;

    static uint32_t idle_period_start = 0;
    static uint32_t handler_start = 0;
    static uint32_t busy_time = 0;

    if(lv_task_run == false) return;`

embeddedt, 2019-04-09:
@ripe909 It looks like an oversight; that is the only place where the function prematurely returns.

If you change the last line to:

```c
if(lv_task_run == false) { task_handler_mutex = false; return; }
```
it should work.

@kisvegabor I wonder why neither of us have encountered this problem yet? Is there always a task to run?

kisvegabor, 2019-04-09:
True, the mutex should be released at `if(lv_task_run == false) ...`.

 `lv_task_run` is set to `true` in `lv_task_init()` which is called in `lv_init()`. So you can experiment the issue if you call `lv_task_handler` before `lv_init`. Is this the case?

embeddedt, 2019-04-09:
> The mutex should be released at `if(lv_task_run == false) ....`

Fixed in https://github.com/littlevgl/lvgl/commit/e8b081b015c48ff5ad5e3d69ba7538e959640c19.

ripe909, 2019-04-09:
> True, the mutex should be released at `if(lv_task_run == false) ...`.
> 
> `lv_task_run` is set to `true` in `lv_task_init()` which is called in `lv_init()`. So you can experiment the issue if you call `lv_task_handler` before `lv_init`. Is this the case?

Yes, in this case lv_init() is called before lv_task_handler, but I will need to attach the debugger to confirm that lv_task_run is set as expected.

ripe909, 2019-04-11:
OK, I tested the task init and the handler execution.  It looks like my situation is that I am calling the task_handler() from a FreeRTOS task, which is blocked on a semaphore given by an interrupt.  During startup, the interrupt is giving the semaphore before the lv_init() actually happens in a different FreeRTOS task, so the lv_task_handler() ends up stuck with the task_handler_mutex = true from the first execution, even though lv_task_run ends up enabled eventually.

embeddedt, 2019-04-12:
Good to hear! It would be best to avoid calling `lv_task_handler` too early though.

closed 2019-04-12 by embeddedt