lvgl/lvgl #1008: [6.0] vars may be used uninitialized

amirgon, 2019-04-08:
```
|| .../lv_draw_line.c: In function ‘lv_draw_line’:
/.../lv_draw_line.c|585 col 13| error: ‘aa_shift2’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
||              lv_draw_aa_ver_seg(prev_p.x + pattern[width_safe - 1].x + aa_shift2,
||              ^
.../lv_draw_line.c|440 col 16| note: ‘aa_shift2’ was declared here
||      lv_coord_t aa_shift2;
||                 ^
.../lv_draw_line.c|583 col 13| error: ‘aa_shift1’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
||              lv_draw_aa_ver_seg(prev_p.x + pattern[0].x - aa_shift1, prev_p.y + pattern[0].y,
||              ^
.../lv_draw_line.c|439 col 16| note: ‘aa_shift1’ was declared here
||      lv_coord_t aa_shift1;
||                 ^
|
```

When building lvgl on linux with these gcc options (removed include paths and defines, for clarity):
```
 -Wall -Werror -Wpointer-arith -Wuninitialized -std=gnu99 -Os -fdata-sections -ffunction-sections  -c -MD -o 
```

embeddedt, 2019-04-09:
Hmm. I build it with very strict warnings usually so I'm surprised I haven't run into this yet. I will fix it tomorrow.

kisvegabor, 2019-04-09:
It was introduced when I added display level anti-aliasing on/off.

They will be initialized if anti-aliasing is enabled else not used. So it's fine to add `0` init. value to both.

I found other minor warnings too with the latest version.

```
/home/kisvegabor/projects/littlevgl/eclipse-workspace/pc_simulator_sdl_eclipse/lvgl/src/lv_core/lv_obj.c: In function ‘lv_init’:
/home/kisvegabor/projects/littlevgl/eclipse-workspace/pc_simulator_sdl_eclipse/lvgl/src/lv_core/lv_obj.c:102:5: warning: "LV_INDEV_READ_PERIOD" is not defined, evaluates to 0 [-Wundef]
 #if LV_INDEV_READ_PERIOD != 0
```
`#if LV_INDEV_READ_PERIOD != 0` should be removed.

```
/home/kisvegabor/projects/littlevgl/eclipse-workspace/pc_simulator_sdl_eclipse/lvgl/src/lv_objx/lv_roller.c: In function ‘inf_normalize’:
/home/kisvegabor/projects/littlevgl/eclipse-workspace/pc_simulator_sdl_eclipse/lvgl/src/lv_objx/lv_roller.c:670:20: warning: declaration of ‘roller_scrl’ shadows a parameter [-Wshadow]
         lv_obj_t * roller_scrl   = lv_page_get_scrl(roller);
                    ^~~~~~~~~~~
/home/kisvegabor/projects/littlevgl/eclipse-workspace/pc_simulator_sdl_eclipse/lvgl/src/lv_objx/lv_roller.c:656:34: note: shadowed declaration is here
 static void inf_normalize(void * roller_scrl)
                                  ^~~~~~~~~~~
/home/kisvegabor/projects/littlevgl/eclipse-workspace/pc_simulator_sdl_eclipse/lvgl/src/lv_objx/lv_roller.c:671:27: warning: declaration of ‘ext’ shadows a previous local [-Wshadow]
         lv_roller_ext_t * ext    = lv_obj_get_ext_attr(roller);
                           ^~~
/home/kisvegabor/projects/littlevgl/eclipse-workspace/pc_simulator_sdl_eclipse/lvgl/src/lv_objx/lv_roller.c:659:23: note: shadowed declaration is here
     lv_roller_ext_t * ext = lv_obj_get_ext_attr(roller);
```

@embeddedt can you fix these issues too?

embeddedt, 2019-04-09:
@amirgon @kisvegabor I will fix all of them today.

embeddedt, 2019-04-09:
Should have been fixed by https://github.com/littlevgl/lvgl/commit/ea7c82f47827a297b5d4a235454c67139154c46f.

amirgon, 2019-04-09:
> Should have been fixed by [ea7c82f](https://github.com/littlevgl/lvgl/commit/ea7c82f47827a297b5d4a235454c67139154c46f).

Thanks @embeddedt!

closed 2019-04-09 by amirgon