lvgl/lvgl #1010: Replacing the signal function on buttons in a list breaks lv_list_get_next_btn

fovea1959, 2019-04-09:
If buttons are created in a list, and the signal function for the buttons are replaced, then lv_list_get_next_btn will not find the buttons with the overridden signal function.

lv_list_get_next_btn uses the address of the signal function as a sanity check to see if the object is really a button. Might need a different way to check?

fovea1959, 2019-04-09:
It is not a good practice to be adding something besides a button to a list, but in this case, I *am* adding a button; it just has a custom signal function (which calls the default signal function and does some additional processing).

embeddedt, 2019-04-09:
@fovea1959 Your use of the button is the correct practice; I was saying that I don't see a need to have the check in there at all, since there is no need to accommodate the wrong practice.

kisvegabor, 2019-04-10:
In [lv_list_btn_single_selected](https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_list.c#L955) it can cause issues if there is a non-button object on the list because it will try to set the button's state.

I can imagine 2 cases when users want to add custom objects to the list:
1. Separator line
2. A title for a group of buttons

I think the type should be checked to be at least a button. As @embeddedt suggested it can be done with `lv_obj_get_type`.

What do you think?

embeddedt, 2019-04-10:
> In lv_list_btn_single_selected it can cause issues if there is a non-button object on the list because it will try to set the button's state.

I hadn't read that function. In that case we definitely need a check of some sort.

embeddedt, 2019-04-10:
I created a PR for `dev-6.0` here: #1014 

Once it's merged I can cherry-pick the patch to `master` as well.

kisvegabor, 2019-04-11:
@embeddedt I merged the PR, thank you. I needed to add fix after merge so please merge that too into master.

embeddedt, 2019-04-11:
@kisvegabor Both have been cherry-picked to master.

@fovea1959 With the latest `master` branch does `lv_list_get_next_btn` now work as expected?

kisvegabor, 2019-04-11:
@embeddedt  Great, thank you!

fovea1959, 2019-04-16:
on latest master, lv_list_get_next_btn indeed works as expected.

embeddedt, 2019-04-16:
@fovea1959 Great, then I'll close this issue. Thanks for the report!

PR #1014 (2019-04-11): lv_list: check actual object type instead of relying on signal handler
Fixes #1010.
closed 2019-04-16 by embeddedt