lvgl/lvgl #1017: Should LittlevGL support FreeType?

kisvegabor, 2019-04-12:
While redesigning the fonts with @puzrin we have taken into account to support [FreeType](https://www.freetype.org/).

However, FreeType is a large software while LittlevGL is designed for embedded devices. 

So once you have a powerful hardware to use FreeType probably you will use a more complex GUI library as well. 

Can you list use cases when LittlevGL and FreeType can be used together?

puzrin, 2019-04-12:
I'd say, "imaginated FreeType" is good checkpoint to validate architecture and borrow ideas, but i would not expect real implementations in near future.

IMO features set, proposed in new font spec should solve all quality questions for several years ahead.

embeddedt, 2019-04-12:
I agree with @puzrin. The current rendering quality of LittlevGL fonts is outstanding (and I expect it to improve further with the new engine) so there is no need to use FreeType just for better quality.

The only viable case I can see for it is where someone wants vector fonts that can be resized to any size at runtime.

kisvegabor, 2019-04-12:
> The only viable case I can see for it is where someone wants vector fonts that can be resized to any size at runtime.

In this case e.g. [mcufont](https://github.com/mcufont/mcufont) can be used. But it should result in the same format as the result of the converter tool. Or it can generate the individual glyphs run time on demand as well.

Rikikooo, 2019-04-17:
Reposting from [this](https://forum.littlevgl.com/viewtopic.php?f=12&t=38) thread.

FreeType might be large, but is also pretty fast. I'm not sure how the LittlevGL engine will compare, but if executable size isn't a problem for someone, I could see them wanting to trade memory for performance.
In my case FT makes my binary approximately 500kbs bigger, and actually I was loosely looking for replacements (specifically [stb_truetype](https://github.com/nothings/stb/blob/master/stb_truetype.h), as I think mcufont only support its own format of rle-compressed fonts).

I haven't really looked into 6.0 features, but rendering fonts directly with LittlevGL seems very exciting!
Just at a glance, the metrics seem to be similar to FreeType (bbox, etc.), it shouldn't be too hard to convert from the FreeType values.
If LittlevGL exposes the bitmap buffer, it should simply be a matter of copying the data rendered by FT to LVGL (at the expense of needing two buffers), which is what I am currently doing in my project. If you're interested in this, you can check my code [here](https://pastebin.com/mpepMA9e) (especially the LvglFont class). The overhead should be really minimal, it's a single memcpy after having calculated the offset to reposition the bitmap to.

In conclusion I don't think lvgl should support FT out of the box, but keeping the API flexible enough to allow its use (as is the case as of today) seems like a good idea.

kisvegabor, 2019-04-18:
Thank you @Rikikooo. It seems we are on the right track then!

puzrin, 2019-04-19:
Probably, it would be interesting to collect info about LVGL "market" to better understand user needs.

Questions:

---

Project:

- Target customers (hobbysts/busness/...)
- single unit or series? If series - amount (< 10 ok not)

Hardware:

- SOC type, RAM/Flash size
- Who manufactures (provided / standard dev board / self-made via easyeda and so on)
- If hardware is powerful, why selected LVGL instead of more "common" GUIs.

Interface:

- Short project description
- Number of layouts, animations...

---

Examples of my answers:

Project:

- Target: hobbysts with some soldering skills
- Amount: big (many hobbysts).

Hardware:

- ARM with minimal external components. Usually stm32 M0-M4 in TSOP or LQFP48/64.
- Self-made, via EasyEDA/LCSC
- Selected LVGL because more powerful GUIs not available. Embedded alternatives are less featured (and seems not actively developped).

Interface:

- Current: micro reflow solderer https://easyeda.com/reflow/reflow-micro-table
- Usually, main screen + settings menu. But should look cool like Material Design.
- i18n (depends on available momory)

---

@kisvegabor may be worth to create survey issue?

kisvegabor, 2019-04-20:
It would be really interesting but I don't think it's required for font converter. I think we know what we need to know. For me, the conclusion so far is the "font level" interface is enough.

FYI, I have a minimal survey on all sites of littlevgl.com: https://littlevgl.com/get-started

kisvegabor, 2019-05-09:
Let continue this topic in #1057 if there are new comments.

closed 2019-05-09 by kisvegabor