lvgl/lvgl #1022: How to use the font/image converters tools offline in a command line  on windows？

HinsunRan, 2019-04-15:
hi，i have some trouble about font or image converter tools.
My scene:
First, font or image is elective, and There are at least ten fonts and more image.
The Online convert tools or offline on LInux is inconvenient .
In other issues, the converter tools by  some developers recommend  is not suitable.
So, How to use the font/image converters tools offline in a command line  on window(7)？
Thank you very much.

embeddedt, 2019-04-15:
(cc @kisvegabor)

It looks like the Linux instructions would need to be adapted for Windows.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

kisvegabor, 2019-04-16:
In v6.0 the font converter probably will be replaced by a tool written in Node.js. It will be more convenient to use offline. See https://github.com/littlevgl/lv_font_conv

> It looks like the Linux instructions would need to be adapted for Windows.

Yes. I myself, not an experienced Windows developer so don't know much about it.

closed 2019-05-14 by stale[bot]