lvgl/lvgl #1028: lv_obj_set_parent with current parent moves the object in dev-6.0

upbeat27, 2019-04-18:
**Describe the bug**

In dev-6.0, if you use `lv_obj_set_parent` to set an object to the parent it already has (for example, to bring in front of an object that was created after it) the coordinates of the object change.

This did not happen in 5.x.

**To Reproduce**

Create an object at specific coordinates.
See where it is located.
Now, do `lv_obj_set_parent` with the parent being its current parent.
The object has moved.

**Expected behavior**

The coordinates of the object should not change, it shouldn't move at all (except z-index).

**Screenshots**

The dropdown was aligned with the text before `lv_obj_set_parent` is called.
After, it looks like this:
[image]

**Additional context**

In my case the parent is always a page, so that might be a prerequisite.

embeddedt, 2019-04-18:
@upbeat27 By default, the page's container will be responsible for laying objects out on the screen. That might be causing the issue you're seeing.

You could try turning it off with `lv_cont_set_layout(lv_page_get_scrl(page), LV_LAYOUT_OFF);`.

upbeat27, 2019-04-18:
@embeddedt I've got `lv_page_set_scrl_layout(main_page, LV_LAYOUT_OFF);`
The key here is that the same menus worked fine in 5.3.
I can see the coords of the object change after calling `lv_obj_set_parent`. The same code compiled against 5.3, the coords do not change, as expected.

embeddedt, 2019-04-18:
(cc @kisvegabor)

kisvegabor, 2019-04-19:
It was due to fix in `lv_page`. I fixed the fix :)

kisvegabor, 2019-05-08:
I added `lv_obj_move_foreground` and `lv_obj_move_foreground` to replace `lv_obj_set_parent`.

I hope your issue is solved by the latest fix so I close it. If still not working feel free to reopen this issue.

closed 2019-05-08 by kisvegabor