lvgl/lvgl #1033: How to select btn in messagebox?

Ax2401, 2019-04-19:
I create a mbox with two btns: apply , close. But if I press ENTER on keyboard, only the Apply button can be select.  How to select and "click" close btn use keyboard?

code:

    lv_obj_t * mbox1 = lv_mbox_create(lv_layer_top(), NULL);
    lv_mbox_set_text(mbox1, "Default message box");        
    static const char * btns[] = { "Apply", "Close", "" }; 
    lv_mbox_add_btns(mbox1, btns, mbox_apply_action);

static lv_res_t mbox_apply_action(lv_obj_t * mbox __unused, const char * txt __unused)
{
    printf("Mbox button: %s\n", txt);  //  !!  Always print : Mbox button: apply.

    return LV_RES_OK;
}

embeddedt, 2019-04-20:
Possibly the Tab key? I'm not 100% sure.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

closed 2019-04-20 by embeddedt

Ax2401, 2019-04-20:
> Possibly the Tab key? I'm not 100% sure.
> 
> We use GitHub issues for development related discussions.
> Please use the [forum](https://forum.littlevgl.com/) to ask questions.

not working, I also tried left/right key, up/down key, and home/end key, pageup/pagedown  . All not working.