lvgl/lvgl #1035: LV_SYMBOL_CLOSE/LV_SYMBOL_OK keyboard buttons should fire on `LV_EVENT_CLICKED` events, not `LV_EVENT_PRESSED`

canardos, 2019-04-20:
All of the keyboard buttons currently respond to the `LV_EVENT_PRESSED` and `LV_EVENT_LONG_PRESSED_REPEAT` events exclusively. This makes sense for the regular keyboard keys and is normal keyboard behaviour, but it's a problem for the apply/cancel buttons, which trigger the `LV_EVENT_CANCEL` / `LV_EVENT_APPLY` events and in most apps will close the keyboard.

These buttons should respond as normal buttons do, to click events.

The problem (aside from the non-standard UI pattern) is that the the `LV_EVENT_CANCEL` / `LV_EVENT_APPLY` events will typically hide/destroy the keyboard and replace it with something else. With the current approach, the area formerly occupied by the apply/cancel button is still being pressed and results in `LV_EVENT_PRESSED` events being sent to the new content, as well as a `LV_EVENT_CLICKED` event when the button/touch is released.

Separate from this (but included here for discussion since it's the same code location), the current CANCEL/APPLY control flow looks quite strange. From `lv_kb_def_event_cb`:

```
} else if(strcmp(txt, LV_SYMBOL_CLOSE) == 0) {
    if(kb->event_cb) {
        lv_event_send(kb, LV_EVENT_CANCEL, NULL);
    } else {
        lv_kb_set_ta(kb, NULL); /*De-assign the text area  to hide it cursor if needed*/
        lv_obj_del(kb);
    }
    return;
} else if(strcmp(txt, LV_SYMBOL_OK) == 0) {
    if(kb->event_cb)
        lv_event_send(kb, LV_EVENT_APPLY, NULL);
    else
        lv_kb_set_ta(kb, NULL); /*De-assign the text area to hide it cursor if needed*/
    return;
}
```

My guess is that the `if(kb->event_cb)` conditional is supposed to check if a custom event_cb has been assigned, if so, call it, otherwise handle the object cleanup (`lv_kb_set_ta`, `lv_obj_del`) here.

But even if no custom handler has been assigned, kb->event_cb will point to `lv_kb_def_event_cb` (i.e. this function), and if it didn't, then this function wouldn't be executing. So I can't see when the `else` clause would ever be called. Perhaps the conditional was supposed to be `if(kb->event_cb != lv_kb_def_event_cb)`?  The `else` clauses also differ, with `LV_SYMBOL_CLOSE` deleting the keyboard, but `LV_SYMBOL_OK` leaving it.

Perhaps an alternative solution would be to send the `LV_EVENT_CANCEL`/`LV_EVENT_CANCEL` events regardless and handle them as first class events rather than conditionals on pressed/clicked events.

embeddedt, 2019-04-20:
(cc @kisvegabor)

kisvegabor, 2019-04-22:
Good point.

`LV_BTNM_CTRL_CLICK_TRIG` has exactly this purpose. See this commit: https://github.com/littlevgl/lvgl/commit/2d5183b05d199e7138a42397091195e2973646ab

canardos, 2019-04-26:
Thanks.

I also noticed that the `if(kb->event_cb)` problem was correct in this [commit](https://github.com/littlevgl/lvgl/commit/5353dae0274b912ae2bbe7acd19e6ac16c16863a).

Regarding the updated code/events/event system, I'm trying to take a look, but am unsure as to where the keyboard object should be deleted under the current system (V5 used the return value to determine if the object was still valid).

I had tested deleting the keyboard object in the custom kb event handler on `LV_EVENT_APPLY`/`LV_EVENT_CANCEL` events. This was working previously (although perhaps it shouldn't have been), but with the current code the kb object is referenced after the return of the custom handler, so that won't work.

To avoid having to dig in too deep, could you please let me know where the deletion should take place?

kisvegabor, 2019-04-27:
I fixed the `LV_EVENT_APPLY`/`LV_EVENT_CANCEL`. Now it should work.

canardos, 2019-04-27:
I'm still having the same issue. The kb object is accessed after the return from the `LV_EVENT_APPLY/LV_EVENT_CANCEL` handler. Since the kb object has been deleted, it causes a segfault. I'm assuming per your above comment that deleting in the `LV_EVENT_APPLY/LV_EVENT_CANCEL` handler is appropriate?

The events are also still being called on a press rather than a click.

The segfault issue appears to be caused by the use of the static flag `event_act_obj_deleted` in `lv_obj.c` in code which is re-entrant, although I haven't had time to examine exactly how the re-entrancy is handled.

Below is the call stack leading to the fault in addition to some of the earlier returned paths:
```
lv_kb_signal(LV_SIGNAL_PRESSED)
-> lv_btnm_signal(LV_SIGNAL_PRESSED)
   -> lv_obj_signal(LV_SIGNAL_PRESSED)
   -> lv_event_send(LV_EVENT_SELECTED)
      -> custom_event_handler(LV_EVENT_SELECTED)
         -> lv_kb_def_event_cb(LV_EVENT_SELECTED)
            -> lv_event_send(LV_EVENT_APPLY)
               -> custom_event_handler(LV_EVENT_APPLY)
                  -> lv_obj_del(kb)
                     -> lv_kb_signal(LV_SIGNAL_CLEANUP)
                        -> lv_btnm_signal(LV_SIGNAL_CLEANUP)
                           -> lv_obj_signal(LV_SIGNAL_CLEANUP)
                           -> lv_mem_free
                     lv_mem_free
                     return LV_RES_INV
                  return
               deleted = true
               return LV_RES_INV
            return //lv_kb_def_event_cb
         return // custom_event_handler
      deleted = false
      // and now it's just a matter of time...
```

kisvegabor, 2019-04-27:
It seems to work here. Please send a minimal code example to reproduce the issue.
Also make sure that you are using the latest version from `dev-6.0`.

canardos, 2019-04-28:
Please see below. `hal_init` is per the SDL Eclipse sample with a single display and I've omitted the includes.

```
static void hal_init();
static void build_ui();

int main()
{
    lv_init();
    hal_init();
    build_ui();
    uint32_t prior_ms = SDL_GetTicks();
    while (true) {
        lv_task_handler();
        uint32_t now_ms = SDL_GetTicks();
        lv_tick_inc(now_ms - prior_ms);
        prior_ms = now_ms;
        SDL_Delay(5);
    }
    return EXIT_SUCCESS;
}

static lv_obj_t* ta;
static lv_obj_t* kb;
static lv_obj_t* label;

static void kb_cb(struct _lv_obj_t * obj, lv_event_t event)
{
    assert(obj == kb);
    if (event != LV_EVENT_APPLY && event != LV_EVENT_CANCEL) {
        lv_kb_def_event_cb(kb, event);
        return;
    }
    if (event == LV_EVENT_APPLY) {
        bool valid = true;
        if (!valid)
            return;
        lv_label_set_text(label, lv_ta_get_text(ta));
    }
    lv_obj_del(kb);
    lv_obj_del(ta);
}

static void build_ui()
{
    label = lv_label_create(lv_scr_act(), NULL);
    ta = lv_ta_create(lv_scr_act(), NULL);
    kb = lv_kb_create(lv_scr_act(), NULL);
    lv_obj_align(label, ta, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 0);

    lv_kb_set_ta(kb, ta);
    lv_obj_set_event_cb(kb, kb_cb);
}
```

kisvegabor, 2019-04-28:
I was able to reproduce the issue with the code above and found the root of the problem. It should be fixed in `dev-6.0`.

> The events are also still being called on a press rather than a click.
It's also fixed.

kisvegabor, 2019-04-28:
I improved the "delete in event" handling further. The previous solution was the workaround of a workaround, but I figured out how to implement it in a correct and robust way.

I hope it will work for you too. :)

canardos, 2019-04-29:
Great work, everything appears to be working correctly!

I'll close this issue since everything highlighted has been resolved.

closed 2019-04-29 by canardos

kisvegabor, 2019-04-30:
Awesome! Thank you for the feedback!