lvgl/lvgl #1038: Extra label created if log level is set to INFO

huming2207, 2019-04-22:
## Describe the bug

Hi there,

I've found a weird bug that if I set `LV_LOG_LEVEL` to `LV_LOG_LEVEL_INFO`, my panel module will generate a weird label on the top left corner. It will disappear if:
 - I turn off the logging by setting `USE_LV_LOG` to 0, or,
 - I set the log level to `LV_LOG_LEVEL_TRACE`.

Here's what it supposed to look like:

[image]

Here's when logging level set to INFO, and here comes the weird bug:

[image]

Regards,
Jackson

## To Reproduce

Target platform: 
  - ESP32, with PSRAM disabled (500k RAM only)
  - ST7789V, 240*240 IPS panel module
      - something similar to this: https://www.aliexpress.com/item/New-IPS-1-3-inch-3-3V-12PIN-SPI-HD-Full-Color-TFT-Display-Screen-ST7789/32859772356.html

Host environment:
  - ESP-IDF SDK: v4.0-dev-402-ga20d02b7f
  - Toolchain (Xtensa GCC from Espressif): gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a)
  - Manjaro Linux 18, rolled to latest version

My code: https://github.com/huming2207/esp-lvgl-st7789/commit/b15b95ae6352e47b2608125637ce4ebc4f83cb4c

Please try downloading it and revert the commit I've mentioned above (`b15b95ae6`). Thanks.

embeddedt, 2019-04-22:
Very strange. Can you reproduce this on the PC simulator?

(cc @kisvegabor, most relevant files are below)

`lv_conf.h`: https://github.com/huming2207/esp-lvgl-st7789/blob/41cea6826f4d079b862aa0501697e39b96cfc760/main/lib/lv_conf.h

`main.c`: https://github.com/huming2207/esp-lvgl-st7789/blob/41cea6826f4d079b862aa0501697e39b96cfc760/main/esp_lvgl_main.c

huming2207, 2019-04-22:
Hi @embeddedt 

I haven't got time to try it yet. I will try it tomorrow.

Regards,
Jackson

kisvegabor, 2019-04-23:
It's really strange. 

Unfortunately, I couldn't reproduce the issue in PC simulator. Can you narrow the issue to exactly one function call? I.e. which function call creates the label.

huming2207, 2019-04-23:
Hi @kisvegabor 

I've tried this:

In the file of [esp_lvgl_main.c](https://github.com/huming2207/esp-lvgl-st7789/blob/41cea6826f4d079b862aa0501697e39b96cfc760/main/esp_lvgl_main.c#L72), after line 72:

 - I removed all "hint" labels, the weird extra label disappeared
 - I added one "hint" label back (can be any one of those three "hint" labels), still fine
 - Then I add another "hint" back,  the weird label shows up.
 - Seems that it's not relevant with those three "code" labels (those with dejavu_mono_40 fonts).

It may be some kinds of memory corruption. But the problem is that if I set the log level to `TRACE`, the weird label disappeared. So I don't have any trace logs for that. I think maybe it's memory corruption somewhere.

Jackson

embeddedt, 2019-04-23:
@huming2207 Yes, it sounds like memory corruption to me. Maybe you could try increasing your stack size.

huming2207, 2019-04-24:
@embeddedt yes it is a bit hard to find this issue as I don't have a JTAG debugger. I will keep an eye on if these similar problems occur later on. 

I will leave this issue open. You can close it if you would like to, and I will re-open it later if that happens.

kisvegabor, 2019-04-24:
To keep the list of the issues clear I'd rather close it and reopen if required.

closed 2019-04-24 by kisvegabor