lvgl/lvgl #1040: Moving object not correctly redrawing

ripe909, 2019-04-22:
In my current UI I have a vertical line that moves over time.  It is aligned to a parent object.

I update the position of the line over time by realigning it to the parent with different offsets.  
The old line location should disappear and the new location should be redrawn.

Most of the time it works as expected, but quite frequently it seems that the old line is not redrawn. leaving a ghost line, while the new location is drawn.

Is there a better way to "move" an object, and could it be a bug that the old line location isn't redrawn to remove it?

Here is the code snippet to create the objects:

`// parent window
//	lv_obj_t * graph_window;
graph_window = lv_obj_create(pattern_graph, NULL);
	lv_obj_set_size(graph_window, 290, 157);
	lv_obj_align(graph_window, NULL, LV_ALIGN_IN_TOP_RIGHT, 0, 0);

...

// chase line
	static lv_point_t vline_points[] =
	{
	{ 0, 0 },
	{ 0, 154 } };

//		lv_obj_t * chase_line;
	chase_line = lv_line_create(graph_window, NULL);
	lv_line_set_points(chase_line, vline_points, 2); /*Set the points*/
	lv_obj_align(chase_line, graph_window, LV_ALIGN_IN_TOP_LEFT, 0, 1);`

Here is the code that updates the line location (using a periodic timer):

`			lv_obj_align(chase_line, graph_window, LV_ALIGN_IN_TOP_LEFT, new_line_location, 1);
`

embeddedt, 2019-04-22:
@ripe909 I tested your code on both `5.3` and `6.0` and it worked properly for me (the line was always cleared and redrawn). Are you absolutely sure your display driver is working properly?

Technically you could just move the line with `lv_obj_set_x` (it would be slightly faster) but that's what `lv_obj_align` is doing behind-the-scenes anyway.

ripe909, 2019-04-22:
Here's a video demonstrating it:

[moving_line.zip](https://github.com/littlevgl/lvgl/files/3103618/moving_line.zip)

ripe909, 2019-04-22:
The lvgl handler is running in a FreeRTOS task which is blocked with a semaphore. I think what may be happening is my timer task preempts the lvgl handler task and updates the line location in the middle of the redraw.

embeddedt, 2019-04-22:
> The lvgl handler is running in a FreeRTOS task which is blocked with a semaphore. I think what may be happening is my timer task preempts the lvgl handler task and updates the line location in the middle of the redraw.

That will definitely cause problems. In that case, guard the call to `lv_task_handler` with a mutex, and guard the call to `lv_obj_align` with the same mutex. That should solve the problem.

ripe909, 2019-04-22:
That took care of it, and thanks for the tip on a more efficient way to move the object!

closed 2019-04-22 by ripe909