lvgl/lvgl #1044: Please provide examples of you sparsed CJK fonts!

puzrin, 2019-04-23:
Hi all! If you use sparsed CJK fonts, could you post your zipped sources here?

We are implementing new font format, and this info will be helpful to verify built-in optimizer and edge cases.

Also, see #995, and feel free to ask any question.

fallstool, 2019-05-08:
Hi:
I found a demo from website. [http://forum.armfly.com/forum.php?mod=viewthread&tid=86371&highlight=littlevgl](url)
you can download it here:[http://forum.armfly.com/forum.php?mod=attachment&aid=Mzc2NjR8OTQ5ZjJiNTd8MTU1NzI4NDQzMHwwfDg2Mzcx](url)

puzrin, 2019-05-09:
Thanks! From that project:

- not so many glyphs (~30),
- bigger size (20px)
- kerning not needed
- all sparse

kisvegabor, 2019-05-10:
> kerning not needed
> all sparse

I can confirm it. However, the number of glyphs and font size depends on the application.
It's easy to have a few thousand glyphs in CJK.

kisvegabor, 2019-06-05:
CJK should be well supported in the new converter. If something is not working please open a dedicated issue for it.

closed 2019-06-05 by kisvegabor