lvgl/lvgl #1048: On init: "Warn: Couldn't allocate memory"

mfp20, 2019-04-29:
On ESP32 this snipplet 

```
	// lvgl graphic library init
	lv_disp_drv_t disp;
	lv_disp_drv_init(&disp);
	disp.disp_flush = lv_iliflush;
	disp.disp_fill = lv_ilifill;
	lv_disp_drv_register(&disp);
	esp_register_freertos_tick_hook(lv_tick_task);
```
brings to
```
I (1513) spi_master: Allocate RX buffer for DMA
D (1613) DeviceDisplay:: Enable backlight.

Warn: Couldn't allocate memory  (/MyESP32App/components/ESP32Cplus/os/lvgl/lv_misc/lv_mem.c #161)
Error: Out of memory!   (/MyESP32App/components/ESP32Cplus/os/lvgl/lv_hal/lv_hal_disp.c #81)
```
Notes: 
- I tried lvgl v5.3, bleeding edge, and esp32_ili9341; within my esp-idf framework.
- I increased by hundreds of KB the stack size of the running task and all the other tasks, then I realized that sizeof(lv_disp_t) is just 28 bytes and that's all lv_mem_alloc is trying to allocate without success (ie: no need to increase stack size). No need to say that lowering LV_MEM_SIZE didn't help as well.
- I can't switch to custom memory alloc/free as I get other errors both with plain malloc/free and the FreeRTOS custom ones.

I'm lost, any help would be appreciated.

kisvegabor, 2019-04-30:
Do you call `lv_init()` before `lv_disp_drv_register()`?

mfp20, 2019-04-30:
@kisvegabor (facepalm) thanks. All good now. I focused on the porting issue so I didn't check any example app yet. But I've been lurking a bit the docs and it is not clearly stated anywhere to call lv_init(). All the code snippets I can find around omit that. Probably docs should be updated with this detail :)

closed 2019-04-30 by mfp20

embeddedt, 2019-05-01:
@mfp20 

> it is not clearly stated anywhere to call lv_init(). 

It's stated in `README.md` under "Porting to an embedded hardware": https://github.com/littlevgl/lvgl/blob/master/README.md

It's also mentioned under "Project set-up": https://docs.littlevgl.com/#Porting