lvgl/lvgl #1049: Animation callback int16_t / int32_t

CostasCal, 2019-04-30:
**Describe the bug**

Lots of functions used as animation callbacks are defined like this:

 static void lv_label_set_offset_x(lv_obj_t * label, lv_coord_t x)

where lv_coord_t is a int16_t

But the animation callback type is defined like this:

 typedef void (*lv_anim_fp_t)(void *, int32_t);

And there is a cast during the assignment to fp, so the compiler will not complain about the difference.
I guess this will work on a 32 bit platform, but it breaks on my 16 bit platform.

**To Reproduce**

If you have a 16 bit platform, try setting up some rolling text, then set a breakpoint in lv_label_set_offset_x - the first time it is called it should have x = anim.start  which is always zero, but I never see zero.

I would fix this, but I'm not sure which way to go. My instinct is that it all should be lv_coord_t, but it seems like the animation code is deliberately using int32_t for start, end etc, so I am not sure.

kisvegabor, 2019-04-30:
This issue really remained hidden on 32 and 64-bit machines where we usually test and use LittlevGL. I suppose it's because the `int16_t` parameters are passed in 32 bit registers so there is no overflow other problems.

It might be related to the [General integer type](https://github.com/littlevgl/lvgl/issues/981) issue.
@embeddedt What do you think?

kisvegabor, 2019-04-30:
Or maybe we could introduce a `variable_size` (1, 2, 4 or 8 bytes) field in `lv_anim_t`.

@amirgon 
As we are talking about animations in #1036 as well it might be interesting for you too.

embeddedt, 2019-04-30:
@kisvegabor This one is tricky. In this case it would be better to use a common integer type. One option (like @CostasCal suggested) is to switch to using `lv_coord_t` in all `lv_obj_set` and `lv_obj_get` functions, as well as the `lv_anim_t` structure. That would solve the majority of the issues.

kisvegabor, 2019-05-01:
The problem with `lv_coord_t` is, not only coordinates can be animated. For example, you can animate with `lv_obj_set_opa_scale` to fade in/out an object and its children.

embeddedt, 2019-05-01:
@kisvegabor I see. In that case we might have no choice but to introduce `lv_int_t`.

kisvegabor, 2019-05-01:
>  I see. In that case we might have no choice but to introduce lv_int_t.

I've just realized there are some problems with it too. For example:
```c
void lv_obj_set_opa_scale(lv_obj_t * obj, lv_opa_t opa_scale)
```

Here `lv_opa_t` is `uit8_t`. Using `lv_int_t opa_scale` can work but less informative.

Or with an animation you can change boolean values too. E.g. make hidden something 500 ms later.
```c
void lv_obj_set_hidden(lv_obj_t * obj, lv_int_t en)
```
would look ugly.

Right now I can imagine only the ide posted [here](https://github.com/littlevgl/lvgl/issues/1049#issuecomment-487935391). However, it's also not perfect because required extra settings from the user.

embeddedt, 2019-05-01:
@kisvegabor I don't like the `variable_size` idea. It seems cumbersome from a user's point of view.

amirgon, 2019-05-01:
Why don't you add support for more than one callback prototype for animations? So the user would be able to select what kind (type) of value he would be changing.

kisvegabor, 2019-05-02:
> Why don't you add support for more than one callback prototype for animations? So the user would be able to select what kind (type) of value he would be changing.

Finally, it's similar to the `variable_size` idea, isn't it? However, have the advantage of any prototype can be added. The list even can be extended by the user. 
On the other, it's more difficult to decide which function to choose. In the variable size approach you need to write only something like: `a.var_size = sizeof(lv_coord_t)`

@embeddedt 
> I don't like the variable_size idea. It seems cumbersome from a user's point of view.

I'm also not happy with it but still don't have a better idea. It would be great if this value could be extracted automatically but I'm afraid it's not possible.

embeddedt, 2019-05-02:
@kisvegabor How would you implement `a.var_size` internally? Call the function in four different ways depending on it's value?

```c
if(a.var_size == 1)
    ((void (*)(void *, int8_t *))a.fp)(a.var, value);
else if(a.var_size == 2)
    ((void (*)(void *, int16_t *))a.fp)(a.var, value);
```

If so, it works, but I think it's extremely ugly.

kisvegabor, 2019-05-03:
@embeddedt 
Yes, that was my idea :sweat_smile: 

However, meanwhile, I found a (hopefully) good solution.
Change the prototype of the animator function from: 
```c
void f (void *, int32_t);
```
to
```c
void f (void *, int);
```

Using `int` will ensure that the systems native register size is used. So passing only one byte is fine too.

On LittelvGL's side, the only required thing is to ensure that only <= 16 bit variables are animated. It seems doable as `lv_coord_t` seems the largest type, but it's `int16_t` internally.

What do yu think?

embeddedt, 2019-05-03:
@kisvegabor In that case, I think it would be better to specify `int16_t` for the animator function. The behavior should be consistent across all platforms.

kisvegabor, 2019-05-03:
Using `int16_t` makes sense too.

@CostasCal Can you test how it works if you replace `int32_t` with `int16_t` [here](https://github.com/littlevgl/lvgl/blob/master/lv_misc/lv_anim.h#L40).

CostasCal, 2019-05-06:
Sorry for the delay getting back, just replacing the int32_t there didn't work, I had to make a few other changes to get it to compile. Then it compiled but didn't work, to get it working I had to change this line:

-    else step = (a->act_time * LV_ANIM_RESOLUTION) / a->time;
+    else step = ((int32_t)a->act_time * LV_ANIM_RESOLUTION) / a->time;

On a 32 bit machine the constant LV_ANIM_RESOLUTION would have made sure the top was 32 bits, but on a 16 bit machine of course the top is only 16 bits and was overflowing. I am guessing that there will be other places in the animation code that this is happening too.

Have attached all the changes I made (I also changed start and end to be 16 bits and a few other things).

[t.txt](https://github.com/littlevgl/lvgl/files/3148856/t.txt)

kisvegabor, 2019-05-07:
@CostasCal 
Thank you for the patch.

I applied the changes in `dev-6.0`. I was afraid to add it to the `master` because it might break something for some people. 

Finally, I added a `typedef int16_t lv_anim_value_t`.

I've also removed `lv_obj_animate()` as it was discussed in #1036 

See https://github.com/littlevgl/lvgl/commit/be84581bbc8194cf9c6b633a938db56457fed1c5

CostasCal, 2019-05-07:
Great!

I think it would be good if someone could scrub the animation code for similar issues where the constants are promoting a expression to 32 bits on 32 bit machines. I suggest changing the constants in that file to 16 bit: eg 1024 -> UINT16_C(1024 ) and check that the tests still work on the simulator.  I don't think I have time to do this right now as I don't have the simulator installed.

kisvegabor, 2019-05-08:
>  I suggest changing the constants in that file to 16 bit: eg 1024 -> UINT16_C(1024 ) 

I tested and worked well. We will see how it behaves in `v6.0` and we can fix it if still not perfect.

As the issue is solved I close it. Thank you for your contribution! :clap:

closed 2019-05-08 by kisvegabor

CostasCal, 2019-05-08:
Thanks for testing it!