lvgl/lvgl #1051: Text color is incorrect when setting LV_COLOR_DEPTH is 8

PierreKao, 2019-05-02:
I use pc_simulator to develop.
I found the text color would be gray(like below image) when I configured the LV_COLOR_DEPTH to 8 and set text color as LV_COLOR_WHITE.
Is it normal ? or I miss something need to do?

I also set object background color as white or set some white image. I got normal behavior.
Could someone help me to clarify this issue?

Thank you.

[image]

embeddedt, 2019-05-02:
@PierreKao Which PC simulator version are you using? The CodeBlocks one or the Visual Studio one?

If you're using the CodeBlocks one I haven't added `LV_COLOR_DEPTH == 8` color support yet.

closed 2019-05-02 by embeddedt

PierreKao, 2019-05-02:
Hi,

I am not sure which version I used. (for eclipse)
I download from this github https://github.com/littlevgl/pc_simulator_sdl_eclipse.git

If I configure as LV_COLOR_DEPTH == 8 on dev board, would this issue happen?

Thank you

embeddedt, 2019-05-02:
@PierreKao Please continue this discussion on the forum.