lvgl/lvgl #1055: Dropdown list should close any other expanded list when selected

neilabdev, 2019-05-08:
Thanks for the great work!  Just thought I'd share the following:

When selecting a dropdown list, it should automatically close any other open dropdown list. For example, if there is a series of dropdown list which represent  a date, when selecting the year, the month should automatically close if open, etc. At no time, should two dropdown list be expanded, etc.

I think I have an interim solution for this, which in short, is to keep track of those expanded, and manually trigger the collapse of the last selection if exists, etc. If it turns out to be an alternative clean solution, I'm happy to do a PR, but at the moment just trying to satisfy a client request, but thought I'd share for the future. Thx!

embeddedt, 2019-05-09:
----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions or suggest enhancements.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2019-05-09 by embeddedt

neilabdev, 2019-05-09:
Yes, I read this. It however would appear that that particular characteristic is not normal functionality on any UI I've experienced. It is different from say,  creating a new control, or a cosmetic enhancement, etc; or another Item I was to post, which is providing the ability for a control to have the concept of "enabled"/"disabled", which while functionality for every UI I have used, would nevertheless constitute a feature enhancement in my mind, and is not itself, and error in behavior, but a lack of functionality. However, I can post it there if you like. Thx.