lvgl/lvgl #1057: New font converter

kisvegabor, 2019-05-09:
A lot of things were being said, so let's summarize what we know so far.

**In general about the converter**
1. The new  font converter is written in Node.Js and can be run offline (CLI) and online in a website using a simple web server (e.g. Apache) 
2. We have a well-designed [binary format](https://github.com/littlevgl/lv_font_conv/blob/master/doc/font_spec.md) which can be considered as an intermediate format or can be used directly to describe the converted fonts.
3. Glyphs are described in bounding boxes
4. Fonts have an "ultimate line height"  property (or can be easily calculated) which gives the maximum height required by the font.
5. A converted font can contain more independent fonts. Each "sub-font" can be continuous or sparse.
6. Kerning is supported via a simple list of pairs and an array of classes
7. Kerning data should optional. If the font's kerning format is not supported a warning should be displayed, but the font still should be converted without kerning data. 
8. Fonts can be compressed with bpp=3, XORing and RLE

**Summary for LittlevGL API**
1. There should be a general `glyph_dsc` and `font_dsc` format containing only the most essential data.
2. The fonts (`lv_font_t`) can be extended with custom, implementation specific data.
3. Data is accessible via `get_glyph_dsc` and `get_glyph_bitmap` functions stored in `lv_font_t`.
4. It is possible to store the fonts' data as either text or binary format (it's just custom data and hidden by the callback)
5. Now, text format will be implemented (if we don't find blocking obstacles)
6. FreeType and other high-level font engines will be supported on glyph level
7. Although we didn't get feedback from CJK characters there shouldn't be problem with them

**Tasks to do**
1. @kisvegabor will create a text-based description format for fonts
2. @puzrin will add that format as an output format of the converter
3. ? font decompressor (font callbacks)

**Question**
Adda as comment
To clean up the font converter related issues continue the discussion about font converter here.

kisvegabor, 2019-05-09:
@puzrin
Two things are still not clear about kerning. 

1) Why Format 0 is required? Can't you always build classes and use only Format 3?

2) The other open question is scaling as highlighted [here](https://github.com/littlevgl/lvgl/issues/995#issuecomment-490687135).

> kerning value as Uint8
> Kerning scale as FP8.8

The cursor shift can be calculated like this, right?
```c
//The basic equation 
cur_shift = (adv_w  / 16) + (kern_value / kern_scale)    //16 becasue adv_w is in 8.4 format
  
//Elimitate the divisons
cur_shift * 16 * kern_scale = (adv_w * kern_scale) + (kern_value * 16)

//So finally
cur_shift = ((adv_w * kern_scale) + (kern_value * 16)) / (16 * kern_scale)
```
If it's correct then the extra division might be a problem because it takes more clock cycles. E.g. Cortex M0+ doesn't have hardware divisor. The extra division seems a bigger problem than the lower precision. 
IMO even 2 or 3 bit fractional (both in kerning and advanced width) can give a decent quality. With rounding it's 0.125px precision.

puzrin, 2019-05-09:
> Why Format 0 is required? Can't you always build classes and use only Format 3?

1. It's more effective for small sets (ascii 0x20-0x7F). Just install it and try different combinations with debug log. It shows sizes of both formats.
2. There can be strange situation, when table not supported by source. In this case Format 3 will be less effective than Format 0. Because it Format 3 relies on fact that glyphs can be grouped.

In short: trust me, nothing can be dropped from spec :).

> The cursor shift can be calculated like this, right?

No, more simple:

```
cur_shift_FP = adv_width + ((kern_value * kern_scale) >> 4);

cur_shift_int = (cur_shift_FP + 2 << 3) >> 4
```

No division.

PS. Please, check length of fractional parts in spec. I noticed you used different values several times.

kisvegabor, 2019-05-10:
> cur_shift_int = (cur_shift_FP + 2 << 3) >> 4

Why `+ 2 << 3`? Is it a letter space?
The rest is clear.

puzrin, 2019-05-10:
> Why + 2 << 3? Is it a letter space?

For correct rounding (that's like `float + 0.5`).

kisvegabor, 2019-05-10:
I see. But I suppose it should be `1 << 3` because `2 << 3 = 16`. In addition, parenthesis is required because `+` has a higher precedence than `<<`.

So `cur_shift_int = (cur_shift_FP + (1 << 3)) >> 4`. Do you agree?

puzrin, 2019-05-10:
Ups... my mistake. You are right. That should be `(1 << 3)`

kisvegabor, 2019-05-10:
No problem. 

I started to implement the text format according to the spec and got a question about cmap formats. Now all 3 three formats are able to describe sparse fonts:
- Format 0: max 255 gaps
- Format 6 max 65535 gaps
- Spatial: Arbitrary number of gaps

But there is no effective format to store continuous ranges. My suggestion is to have 2 formats:
1. for continuous rages: only define
  - range start
  - range size
  - glyph_id_offset
2. for sparse:  something like spatial but define only:
  - range start
  - range size
  - glyph_id_offset
  - unicode list: map `glyph_id_offset + 1`, `glyph_id_offset + 2` etc to unicode

Actually, continuous can be only a header, and if it's sparse then `unicode_list` should be presented as its payload.

This approach has the following advantages:
1. If you describe the whole ASCII range with Format 0 (the most compact) it needs 127 bytes. In my suggestion, it takes 0 bytes.
2. For sparse fonts "Spatial" uses 2 + 2 bytes / character. My suggestion uses on 2 bytes / character.

By the way, why the "spatial" name? I've seen and used "sparse" so far.

puzrin, 2019-05-10:
statial => sparse. I don't rememper why i typed it. Probably mistake.

https://github.com/littlevgl/lv_font_conv/blob/master/doc/font_spec.md#subtable-format-6-data see note:

> This is reserved and not used in current converter.

I was not ready to drop it immediately, but it's not used in latest convertor. Everything is represented as combination of multiple `Format 0` & `Format sparse`. That has optimal size and reasonably fast lookup.

Format 0 data is taken from OpenType spec. I think, dropping data for different subtables layout possible, but i'd like to not diverge from OpenType spec without critical needs. It'd difficult to predict possible side-effect in future.

For example, you have no 100% guarantee for all cases, that glyph_id-s will be consecutive.

puzrin, 2019-05-10:
`0x401,0x410-0x44F,0x451` - that's range for russian letters. 3 headers required.

Also, if you decide to generate list of "only used letters", fragmentation may increase significantly. It's difficult to estimate pros and cons for ALL possible cases. So, i'd prefer to not diverge from standard, and rely on experience of OpenType developpers. Standard already proven to be universal.

kisvegabor, 2019-05-10:
> 0x401,0x410-0x44F,0x451 - that's range for russian letters. 3 headers required.

So Russian has 65 characters. It's 65 extra bytes in Format 0. Three headers are still smaller. I know it's only one case.

We can even mix continuous and sparse ranges. E.g.
- continuous
   - 0x20-0x7F
   - 0x100-0x123
   - 0x345-0x39F
- sparse  
  - 0x80-0x4FFF
  - 0x21000-224AA

Note that in most of the cases one sparse range (0x0000...0xFFFF) will be enough. Only extreme cases would require two or more.

This way glyphs still can be easily found:
1. First search in continuous ranges
2. If not font there, check the sparse ranges and if the correct range is found search the glyph there.

> Also, if you decide to generate list of "only used letters", fragmentation may increase significantly. 

Probably they fit into one big sparse range. Although it might be less optimal then Format 0 because it uses  2 bytes instead of one. However, if you have a lot of characters and a lot of languages are supported that 5000 characters x 1 extra byte won't be an issue.

And converting only the used characters is not a typical scenario. Imagine that your GUI gets some text from a server and it wants to display it. In this case, even with i18n, all characters are required from the servers "language".  We should rather optimize the more common cases where mainly continuous ranges are used.

> So, i'd prefer to not diverge from standard, and rely on experience of OpenType developpers. Standard already proven to be universal.

It could be a long argument but in my opinion, some customizations are not evil. OpenType is not designed for embedded systems therefore probably not fully optimal for our case.

puzrin, 2019-05-10:
> And converting only the used characters is not a typical scenario.

That's correct for lvgl5 only. IMO you mix "user needs" and "lack of instruments". As soon as lv_i18n will get extension to dump charlist, users will be happy to use those, especially when CJK exists. Because this automates glyphs selection without any efforts.

IMO you underestimate importance of data design flexibility. It should predict not only cases you know, but also cases you can not imagine now. Since we both have no full knowlegde of OpenType details, the only thing we can do - make copy close to original. Then we have chance to reuse experience of OpenType.

My change to Format 0 (glyph_id offset) is "extension". Because it not conflicts with original, and allows original use. Your change is completely different thing, invented from zero. By default, i don't like to risk for small local benefit. May be, you change will be better, may be not. This requires more careful check. I did not said it's bad. I said it's risky.

> OpenType is not designed for embedded systems therefore probably not fully optimal for our case.

It's approaches are still better than everything been done for embeddeds before :). And i used very simple method of design - reused max possible things, not inventing my own without need. As you see, result is better than everything you can find in similar projects.

embeddedt, 2019-05-10:
> It's approaches are still better than everything been done for embeddeds before :). And i used very simple method of design - reused max possible things, not inventing my own without need. 

Not necessarily. OpenType was designed for full-fledged systems where saving a few KBs of space is not critical. Reusing all components from that format can result in unnecessary bloat on an embedded system.

puzrin, 2019-05-10:
> OpenType was designed for full-fledged systems where saving a few KBs of space is not critical.

That's your personal opinion. My one is different. OpenType continues TtueType. And TrueType was designed many time ago, when hardware had a lot of memory & CPU constrains.

> Reusing all components from that format can result in unnecessary bloat on an embedded system.

I can say opposite. Desire to make things simple may produce horrible results for situations, which have no simple solution. I have absolutely no desire to invent my own format and spend time for polish next 5 years :). Especially, in situation when you can compare with existing "embedded" alternatives, designed by embedded developpers.

kisvegabor, 2019-05-11:
I think we have nothing to convince each other. I usually figure out things on my own and use the standards to check the required features and get ideas while you insist on standards. My approach probably will result in a better-optimized solution while yours might require fewer iterations. Somehow we need to make a consensus. 

Note that we don't need to cover every possible, exotic case. 
According to my experience (which should be taking into account because I talk about embedded GUIs with people for several years):
1. A big part of UIs are **English only** with some extra characters like `°` or `±` and symbols from FontAwesome. Here we need a continuous ASCII range and a sparse range.
2. An other big (but smaller) part is **CJK** UIs with FontAwesome symbols. As CJK and FontAwesome characters are < 0xFFFF we need only one spare range.
3. A small portion of the UIs are for **multilanguage**. Using `lv_i18n` here we need only one big sparse range and maybe a full ASCII.
4. A really small portion is UIs with **only one non-English language** (e.g. Spain or Russian only).  It also requires an ASCII (probably) and some other continuous ranges and maybe a sparse range. 

The 4 cases above cover almost all use cases of LittlevGL. In conclusion, we can see that usually there are:
- one or more continuous range(s)
- and exactly one sparse range (as even CJK and Font awesome are < 0xFFFF). Although we need to support character beyond 0xFFFF too. 

We need to find a solution which solves these problems.

I meant something like this: 
https://github.com/littlevgl/lvgl/pull/1050/files#diff-c9a0584e0a5338884d57c1246ae52962R104

It's quite simple, effective and knows everything that we need.

puzrin, 2019-05-11:
If you suggest _optional_ extension, i see no problem:

1. offset = 0 => skip data and map ID-s continuously
2. offset != 0 => delta-coded ID-s

Previous postings sounded like (1) only, without (2).

> As CJK and FontAwesome characters are < 0xFFFF we need only one spare range.

Just for you info, FontAwesome has no "correct" unicode mapping. Just put everything into PUA. The most of new added unicode "images" have codes > FFFF. http://www.unicode.org/charts/ - see "Emoji & Pictographs".

puzrin, 2019-05-11:
Updated spec, see last commits https://github.com/littlevgl/lv_font_conv/commits/master/doc/font_spec.md

Convertor will be updated later, since it's output still valid.

kisvegabor, 2019-05-12:
Two questions:
1. Let's take an example. We have a font with the following content
   - continouous ASCII
   - Russian with  `0x401,0x410-0x44F,0x451` (as you mentioned eariler)
   -  3 pieces, < 0xFFFF FontAwesome characters

In my apporach it would require 3 cmap tables (it directly comes from the ranges defined by the user): 
  - 1 continoues for ASCII  
  - 1 continues for 0x410-0x44F
  - 1 sparse for the rest (with 5 glyphs)

If we have 2 ways to describe sparse fonts (Format 0 and sparse) how will you decide which one is more optimal and when to use which?

2) Why `delta-encoded Glyph1 ID` is required in "Sparse" format. As you can store the glyph dscs in arbitrary order you ensure direct (incremental) mapping:
- First parse glyph: `Glyph ID offset`
- Second sparse gylph: `Glyph ID offset + 1`

puzrin, 2019-05-12:
> how will you decide which one is more optimal and when to use which?

In current implementation, criteria is "minimal total size" (with uncontrolled subtables count). It may be not ideal, but we could not invent something better in limited time.

[2 table types + size] can be transformed to math problem "find shortest path in graph"

https://github.com/littlevgl/lv_font_conv/blob/master/lib/font/cmap_build_subtables.js

> Why `delta-encoded Glyph1 ID` is required in "Sparse" format.

It's not necessary. Just postponed decision and did this the same way for all subtables. This feature is easy to "disable" by setting 0 in header.

kisvegabor, 2019-05-13:
I see. So it seems the formats I imagined can be realized with the converter. Now the question is shall LittlevGL support all the possible settings of formats or is it enough to support:
- Format 0 without payload
- Sparse with only codepoints and without Glyph IDs

As I mentioned this format can be directly derivated from the input ranges defined by the user.

puzrin, 2019-05-13:
I would not like to write separate algorythms for lvgl. When i promised too create lvgl writer, i meaned "different template" only, not everything.

kisvegabor, 2019-05-13:
> I would not like to write separate algorythms for lvgl. 

I think this is the root of most of our disagreements. I consider the font converter as a tool whose main purpose is to serve LittlevGL needs and not more. In contrast, as I see you consider the converter as a general tool which actually has a feature to export a LittlevGL compatible output. Is that correct?

I have three problems with the extra features (not required for LittlevGL) of Format 0 and Sparse:
- Implementing those features takes time, testing and can be the source of more issues later
- They make font descriptors more complicated. Contributors will understand a more simple interface easier.
- I just don't want features in the library which I don't think useful because I (and the maintainer) need to live with that for years.

embeddedt, 2019-05-13:
@puzrin
> and increase divergence from well tested standard to untested things.

The well tested standard is not 100% applicable to the specific needs of this project, like @kisvegabor said.

puzrin, 2019-05-13:
> > I would not like to write separate algorythms for lvgl.
> 
> I think this is the root of most of our disagreements.

No. That's result of different design approaches. From my point of view, you break data design principles, used be me.

> I consider the font converter as a tool whose main purpose is to serve LittlevGL needs and not more. In contrast, as I see you consider the converter as a general tool which actually has a feature to export a LittlevGL compatible output. Is that correct?

IMO, data design is universal. If it's pinned to specific lib implementation - something goes wrong with architecture.

One of important principles - to have reserve of flexibility in uncertain places. I don't like to risk with getting unmodifiable brick again, like we have in v5.

> I have three problems with the extra features (not required for LittlevGL) of Format 0 and Sparse:

First, from my point of view, you use not suitable instruments for this task. I said at start, that text format is not suitable to describe non-trivial data. But instead of switching to binary format, you wish to fit data for your limited tool.

More common way for me would be to use bin and protect from regressions via tests. This reduces need of debugging almost to zero. That's not theory - you may check JS code. Tests there are not for fun.

Second, we can inspect all cases.

> 
> * Implementing those features takes time, testing and can be the source of more issues later

Now there is only one feature - indexed lookup of Format 0. I don't think that's too much in all spends.

> 
> * They make font descriptors more complicated. Contributors will understand a more simple interface easier.

1. That's because you wish to use text format. But that's your choice, not my one. I tried to explain, that if something was good for you before, there are no guagantee to propagate that approace to all cases with success.
2. Contributor will see no big difference (i'm not sure they will ever look this part at all). The only difference now is indexed lookup of Format 0. Size of appropriate code is small, compared to all size of font reader.

> * I just don't want features in the library which I don't think useful because I (and the maintainer) need to live with that for years.

Your opinion about "usefulness" is limited by your experience and your imagination. I'd like to point your attention, that such approach produced current code in v5. I think you see from my comments in other issues, this could be done more flexible with string-based APIs and with comparable cost.

---

In short: if there is only one difference, i'd suggest to implement indexed access of Format 0 because i ask you :). I think, at current moment you can rate result of my approach to data design (size of generated fonts), and know that i don't wish bad things to lvgl.

puzrin, 2019-05-13:
> @puzrin
> 
> > and increase divergence from well tested standard to untested things.
> 
> The well tested standard is not 100% applicable to the specific needs of this project, like @kisvegabor said.

You can not do "well" acritecture design, breaking generic design principles anytime you don't like something local, id designing from bottom to top. Current state of font spec was achieved not because i'm lucky. I think, current convertor and data size tell better than many words, that i understand something in this area.

kisvegabor, 2019-05-14:
I think it's not really constructive this way. Everybody has right to a certain extent but we are getting nowhere if we can't make decisions.

So it's not about @embeddedt and I has always right but as maintainers, we know the best what is reasonable for the library because we are here with the community for years. And again, we will live with the decisions so it should be what we agree on.

For example, yes, the whole Format 0 can be implemented but we think it's not required. And we can't make progress if all of our decisions are questioned or overruled. Suggestions and ideas are very welcome and always considered but as in every healthy project, the maintainers should decide what direction to take.

puzrin, 2019-05-14:
I'd say right to make decisions is powered by personal time, people spend for project. If mainainer's dicision includes their time for maintaining lvgl-specific font spec and creating lvgl writer for convertor (or rewrite it entirely if they don't like it) - i will understand everything.

Also, IMO you should have similar difficulties with `kern` table, as with `cmap` table. It may be better to review all at once, instead of stopping on `cmap` as on single and most important thing.

kisvegabor, 2019-05-15:
> We can consider my current work as investigation with neutral output, which can be used anyhow other wish. IMO such result is consistent and self-sufficient.

That's the problem. Our goal is to have a better font converter for LittlevGL while you are creating a "neutral" font converter. Theoretically, it's possible that you create the initial work, investigation etc and somebody else will modify it for LittlevGL but it seems no other contributors are interested in the font converter right now. So if you don't "prepare" it for LittlevGL I don't know who will. Working on a general converter which once *might* support LittlevGL (in a way we imagined it) is not reasonable from our point of view.

> Also, IMO you should have similar difficulties with kern table, as with cmap table. It may be better to review all at once, instead of stopping on cmap as on single and most important thing.

I didn't check them in such detail yet. Let's have a common mindset first to see how to move forward.

puzrin, 2019-05-15:
I have no clear direct plan to create universal convertor and promote it everywhere, but use universality as important checkpoint of good design. Also, it's convenient for asking developers of other formats to compare, check mistakes and so on. The "worst" case would be to stop on current point, but while result of my work is valuable and universal, that's not critical in long term.

IMO words "for this project only" may mask very strange things, but since you prove with your own time doing anything you wish, i have no right to criticize. Problem is, that i don't like to work in such mode and we need border, to keep our principles separate :). Having different principles does not mean we can not do anything together at all.

Please, understand me right, working on lvgl-locked solution is not the best way of spending my time. Consider situation as opportunity to get extra benefit, if you leave "vendor-locked" way. Only you can decide if this worth your efforts or not.

kisvegabor, 2019-05-16:
Okay. I think this discussion was required to clearly understand each other's intent and motivation. 
I still see a bit potential in the new font converter and see its benefits so we shouldn't give up! :)

Let's turn back to agree on features. I've checked kerning spec again and it seems fine as it is. I can build a text format for it and I agree that both formats are reasonable.

So the only missing point is the "gap" support of Format 0. Can you imagine a situation when it can be used effectively? I know that we can't imagine all possible cases but if none of us can tell a real use case chances are high that it's not really required. 

My suggestion is leaving the Format 0 gap support in the spec but don't implement it until we get a valid use case. What do you think?

puzrin, 2019-05-16:
> So the only missing point is the "gap" support of Format 0. Can you imagine a situation when it can be used effectively?

1. At first glance:
   - autogenerated list of symbols from i18n. Capital letters will have many gaps.
   - some languages like "russian", where symbols have gaps.
2. There is general tendency to increase subtables count. As everything not from OpenType - not investigated and a bit suspicious. I try to keep reserve for reverse is such places (because we have to time for more deep investigations). Alternative could be replace sparse format with subtable Format 4 (due lack of time - postponed for plan B).

> My suggestion is leaving the Format 0 gap support in the spec but don't implement it until we get a valid use case. What do you think?

Good spec should provide something like recommendations or reference implementation for ALL features. It would be not convenient to have feature, which may requires full rework of complicated splitting algorythm. Also, without using feature and collecting statistics, it will be impossible to decide is feature "dead" or not.

Could you explain difficulties of Format 0 index implementation? At first glance - zero check + table lookup. That can not take a lot of code. Additional array name - also not too much, compared to all other places. May be i don't know something.

kisvegabor, 2019-05-16:
> Could you explain difficulties of Format 0 index implementation? At first glance - zero check + table lookup. That can not take a lot of code. Additional array name - also not too much, compared to all other places (and that will be autogenerated anyway). May be i don't know something.

My fear is that indexed Format 0 will be generated even for continuous ranges. (E.g. only an ASCII). Or will you say when the number of consecutive characters is more then a table header size then a new non-indexed Format 0 table will be added?

I forget one other thing which we haven't agreed on yet. In Sparse format why `delta-encoded Glyph1 ID` is required? It can be dropped if the glyphs are stored in the appropriate order because in this case the `array index + Glyph ID offset` can map the glyph to the Unicode character. Similarly to Format 0 just on two bytes.

puzrin, 2019-05-16:
> My fear is that indexed Format 0 will be generated even for continuous ranges. (E.g. only ASCII). Or will you say when the number of consecutive characters is more then a table header size then a new non-indexed Format 0 table will be added?

I guess, algorythm could be like this:

1. First pass - split all to combinations of indexed Format 0 + sparse ranges (works now).
2. Second pass - search consecutive Format 0 ranges and try to optimize via index drop (solve again task of filling range with optimal combination of 2 formats).

That's only assumption, to demonstrate solution exists and it will be "good enougth".

> I forget one other thing which we haven't agreed on yet. In Sparse format why delta-encoded Glyph1 ID is required?

Technically, it's not required (as entries count not required for format0). But header records are created for unified processing, and i tried to minimize difference. In real world, you can rely it will be always zero, if that's principle for you.

> It can be dropped if the glyphs are stored in the appropriate order because in this case the `array index + Glyph ID offset` can map the glyph to the Unicode character. Similarly to Format 0 just on two bytes. If you agree with that the spec can be simplified to have only one format with `codePoint Size = 1 or 2 byte`.

IMHO that's a secondary, independent part of your suggestion, not related to existance of "Glyph ID offset" in header. It sounds:

"Use fact, that glyph ID-s are consecutive with high probability, to reduce table size.".

My comments are similar to format0 case:

- Since it will not work well for 100% of cases (monotonic glyph ids increase is not guaranteed), we should use it with care, and leave space for maneuvres.
- Since it should work well in many cases, it worth think how to use this.

At quick glance, i'd suggest:

- Update existing sparse format from "tuples" to 2 arrays ([delta codepoints] + [glyph ids])
- Introduce "tiny sparse" format, without [glyph ids].
- Use "Glyph ID offset" property to store extra data for tiny sparse layout.

That's not final, only my first thoughts.

kisvegabor, 2019-05-17:
>> My fear is that indexed Format 0 will be generated even for continuous ranges. (E.g. only ASCII). Or will you say when the number of consecutive characters is more then a table header size then a new non-indexed Format 0 table will be added?

> I guess, algorythm could be like this ...

If Format 0 is optimized to use indexes only if required then fine.

> - Since it will not work for 100% of cases (monotonic glyph ids increase is not guaranteed), we should use it with care, and leave space for manoeuvres.
    - Example: multiple codes can be mapped to the same glyph (Roboto has only 2 such glyphs and we just duplicate those now)

As you mentioned in the worst case we need to store an extra glyph in these very very rare cases. IMO it's not worth the extra complexity to optimize for these very rare cases.

In the text format I separated `glyph_dsc` and `gylph_bitmap` array. If it was separated in the spec too only an extra `glyph_dsc` would be required (pointing to the same bitmap) which size is negligible. The overall size wouldn't change because the header's `Glyph id1 data`, `Glyph id2 data` could go to `gylph_data` as `Glyph Bitmap index`

puzrin, 2019-05-17:
> As you mentioned in the worst case we need to store an extra glyph in these very very rare cases. IMO it's not worth the extra complexity to optimize for these very rare cases.

IMHO you reverse reasons and findings. Term "optimization" usually means, you already have 100% coverability, working in predictable way. Usual approach is progressive:

1) Predictable formats created
2) Optimizations applied

You try to do (2) without (1) and pray that exclusion will be not too bad. But writing spec has goal to provide predictable resolution for ALL cases. IMO at this level, choice is:

a) investigate details of sparse-tiny-only use and prove mathematically correct coverability
b) provide enough flexibility with supported formats

Since i don't know how to do (a) in reasonable time, my suggestion is to select (b). That will cost something, of cause, but seems not to much (if we use your and my time as metrics). I tried to suggest layouts, which allow "unified" implementation with small difference in the end (code size will not increase too).

kisvegabor, 2019-05-17:
> investigate details of sparse-tiny-only use and prove mathematically correct coverability

I can't give a mathematically exact description but let's think about it logically:
- The task is to map: `codepoint` -> `glyph_dsc` -> `glyph_bitmap`
- As every `glyph_dsc` corresponds to a `codepoint`, `glyph_dsc`s can be ordered accordingly.  I see no problem with guaranteeing this. Simple 1-to-1 relation. Here a translation is required from `codepoint` to `glyph_dsc`. 

Let's say you have the following code points:  
 `0x23`, `0x34..0x66`, `0x78`, `0x99`, `0x200..0x240`, `0x249`, `0x42A`, `0x12AB` .
But in this `cmap` you want to include only the following `codepoint`s (arbitrary selected):
`0x23`, `0x99`,  `0x42A`.

| Codepoint | Glyph dsc ID |
| ------ | ------ |
| 0x23   | 0 |
| 0x99   | 1 |
| 0x42A | 2 |

It's a simple array of `code points` and the index tells the `glyph ID`. Although, it **requires the `glyf` table to be local in the `cmap` table**. Picking other `codepoint`s (still in ascending order) for another `cmap` also can be easily mapped to its local `glyph` table through array indexes 

In case of text format, it means only moving `lv_font_glyph_dsc_built_in_t * glyph_dsc` to `lv_font_cmap_built_in_t` . See [here](https://github.com/littlevgl/lvgl/blob/font/src/lv_misc/lv_font.h#L123). I hope it illustrates what I mean.

- If two codepoints use the same `bitmap` both `glyph_dsc`s can point to it. N-to-1 relation. It's the easier part.

Have I missed something?

puzrin, 2019-05-17:
Imagine bad case - 1000 "synonym" chars mapped to single glyph. You will need 1000 subtables, with cost 16*1000. Additional problem - such splits are very inconvenient for inventing algorythms (when combination of subtables searched, number of criterias should be minimal).

If option to have IDs index exists, we are not affected and situation is more predictable. Then we have guarantee to get acceptable solution for worst case, and use more optimal (index-less) subtable for regular cases.

kisvegabor, 2019-05-18:
Ok, then please update the spec to be sure we are talking about the same thing. :)

puzrin, 2019-05-18:
- [sparse layout update](https://github.com/littlevgl/lv_font_conv/commit/78dac3f4ba9886054c5879c4387fd8c61d2b10d9#diff-b90d968b98e0449148877cd4a7dd9286)
- ["tiny" versions of subtables](https://github.com/littlevgl/lv_font_conv/commit/0976a9a844dce02d7a3328d0eff2393853f59804)

OK?

kisvegabor, 2019-05-18:
Looks good, thank you! I'll try to add it to the text format and then I will see is better.

I started to add the kern tables to text format. But something was not clear with classes. The spec says: `left-hand classes mapping, index = glyph_id, value => class id.`

So let's say I have the 0x37 character. 
1. I get its `glyph ID` from `cmap` tables according to "Format 0" or "Sparse". E.g. I get 0x12.
2. I get the class like: `uitn8_t class = left_class_map[0x12]`

Is that correct?

puzrin, 2019-05-18:
> So let's say I have the 0x37 character.
> 
>  1. I get its `glyph ID` from `cmap` tables according to "Format 0" or "Sparse". E.g. I get 0x12.
> 
>  2. I get the class like: `uitn8_t class = left_class_map[0x12]`
> 
> 
> Is that correct?

Yes.

puzrin, 2019-05-18:
If i understand right, the only notable difference in the end should be using pointers instead of relative data offsets. Because there are no technical possibility to express such offsets via code. Everything else should "mirror" binary layout as is.

kisvegabor, 2019-05-18:
> If i understand right, the only notable difference in the end should be using absolute (4-byte) pointers instead of relative data offsets.

Sorry, I don't really understand. Can you explain it more specifically?

Can you confirm this?

###  Format 0 tiny
`glyph_id = glyph_id_start + (codepoint - range_start)`

### Format 0 full
`glyph_id = glyph_id_start + glyph_id_list[codepoint - range_start]`

### Sparse tiny
`glyph_id = glyph_id_start + search(unicode_list, codepoint - range_start)`

### Sparse full
`glyph_id = glyph_id_start + glyph_id_list[search(unicode_list, codepoint - range_start)]`

puzrin, 2019-05-18:
> Sorry, I don't really understand. Can you explain it more specifically?

Example: https://github.com/littlevgl/lv_font_conv/blob/master/doc/font_spec.md#table-loca , glyph data start defined via "offset" (2 bytes for table < 65K). In your case that will be pointer (for all such places).

Everything else from bin format has analogue in C syntax (via struct/union), without size loss.

> Can you confirm this?

Seems correct.

kisvegabor, 2019-05-18:
> Example: https://github.com/littlevgl/lv_font_conv/blob/master/doc/font_spec.md#table-loca , glyph data start defined via "offset" (2 bytes for table < 65K). In your case that will be pointer (for all such places

I meant to solve this by separating the `bitmaps` and `glyph_dsc`s. `gylph_dsc`s has a fixed size so I can build an array from them. The `bitmap` array is just a long stream of bytes. The `glyps_dsc` has a `bitmap_pointer` field to tell where its bitmap starts. It's just more natural to handle. Is it managable for you in lvgl6 writer?

puzrin, 2019-05-18:
I don't understand. You need point to bitmap anyway. If you move relative reference from one place to another, it will not become possible in C.

In theory, it's possible to calculate size of objects and create array of relative references (offsets) manually, but that's certainly not kind of things i wish to do. That will be a fragile and strange mixture of text and bin things.

kisvegabor, 2019-05-18:
Sure, I need one pointer:
1. In bin format: you point to the beginning of the new "glyph_dsc+bitmap" via the "loca" table
2. In text format: I'm planning to point to the bitmap from the `glyph_dsc`

However, using an offset `glyph_dsc+bitmap` 
1. means an extra indirection every time we deal with glyphs.
2. the alignment of `glyph_dsc` data is not guaranteed

So it seems to me having a `bitmap` and a `glyph_dsc` table is more reasonable and has the same memory cost.

puzrin, 2019-05-18:
> So it seems to me having a `bitmap` and a `glyph_dsc` table is more reasonable and has the same memory cost.

While your text can be generated via templates without additional compicated algorythms, it's ok. I don't think you will get real benefits with splitting spec's glyph data, but since that's not principal for convertor's writer, i see no reasons to complain.

Additional size "loss" will be `2 * glyphs_count` (lack of relative references) + may be some loss due different bitmap properties packing. IMO - not too much, acceptable price for text format.

kisvegabor, 2019-05-19:
Just be sure I understood well: supporting separate `bitmap` and `glyph_dsc` arrays in lvgl6 writer is ok?

puzrin, 2019-05-19:
> Just be sure I understood well: supporting separate `bitmap` and `glyph_dsc` arrays in lvgl6 writer is ok?

Yes.

If you do only linear transforms of binary spec => only output templates affected => no problems + easy maintenance.

kisvegabor, 2019-05-20:
Ok.

I added the types required for the text format. See [this commit](https://github.com/littlevgl/lvgl/pull/1050/files#diff-73e3f69582204d7a41298e9ab8f3bf7eR37).

Only the declarations are added so far and not tested yet.

kisvegabor, 2019-05-22:
@puzrin 
Could you have look at it?

puzrin, 2019-05-22:
I think it would be more productive to review final samples of fonts. It's a bit difficult to imagine final result from header only.

Could you create a dummy font with couple of glyphs and all tables filled? Properties can be random, and `glyf` arrays can contain zeros, that would be enougth.

puzrin, 2019-05-22:
I mean, we solved all principal questions about spec. As soon as you show font template, i will ask some minor questions and start adding writer to convertor. Or do you expect something else?

kisvegabor, 2019-05-23:
> Could you create a dummy font with couple of glyphs and all tables filled? Properties can be random, and glyf arrays can contain zeros, that would be enougth.

Ok, I'm working on it. Probably it will be ready tomorrow.

kisvegabor, 2019-05-24:
@puzrin 
Here is the template: https://github.com/littlevgl/lvgl/pull/1050/commits/5ef94e395d4af83783c8d04a7ca5f2d31d7adcb2

What do you think?

puzrin, 2019-05-24:
At first glance, 2 issues:

1. Loss of many font header properties (font metrics and may be some flags). You risk to need those in future or have different dirty effects (for example, labels shift after adding new glyphs).
2. Bitmaps representation is for old format - will not work for compressed bitstreams.

Everything else seems okr (i will check once more).

puzrin, 2019-05-25:
From "moving forward" point of view, in first version of convertor:

- glyph data can be done as single line arrays, with glyph code in comment
- missed properties can be added later

However, it worth minimise rewrites count. Let me know your deadlines, how urgent do you need lvgl format support. Should i start ASAP, or we can discuss my comments about templates first.

kisvegabor, 2019-05-25:
> Loss of many font header properties (font metrics and may be some flags). You risk to need those in future or have different dirty effects (for example, labels shift after adding new glyphs).

> Bitmaps representation is for old format - will not work for compressed bitstreams.

I commented on the commit.

> From "moving forward" point of view, in first version of convertor:
> glyph data can be done as single line arrays, with glyph code in comment
> missed properties can be added later

Sounds good! I'll need some time to update lvgl to work with the new font file. Probably some issue will turn out during the integration.
So the most import would be to have a font file I can start to work with. Formatting and some miscellaneous parameters can be added in the next iteration.

puzrin, 2019-05-28:
LVGL writer ready to play.

```
./lv_font_conv.js --bpp 1 --size 16 --font ./node_modules/roboto-fontface/fonts/roboto/Roboto-Regular.woff -r 0x20-0x400 --format lvgl -o ./lvgl_sample.txt
```

Everything located in https://github.com/littlevgl/lv_font_conv/tree/dev/lib/writers/lvgl and should be easy to change.

If you decide to apply changes - i'd suggest use separate branch as sandbox. After stabilised - merge to `dev`.

kisvegabor, 2019-05-29:
Great! 

I started to integrate it. There are some minor typos in the generated format (e.g. missing `;`) but there is something I don't understand.

I've run:
```
lv_font_conv  --no-compress --no-prefilter --bpp 1 --size 16 --font ./Roboto-Regular.ttf -r 0x20-0x7F --format lvgl -o ./lvgl_sample.c
```

But for `'!'` bitmap I got:
```
    /* U+21 "!" */
    0xd0, 0xff, 0xfc,
```

As we discussed earlier, without compression every line of the bitmap should be byte aligned. So it should be something like:
```
    /* U+21 "!" */
    0xC0, 0xC0,  0xC0, 0xC0, 0xC0, 0xC0, 0x00, 0xC0,
```

[27 more replies]

closed 2019-06-06 by kisvegabor