lvgl/lvgl #1059: Extra symbol

zavovi, 2019-05-10:
Hi,
please could you help me with extra symbols? I downloaded Font Awesome and generated fonts from 61440 to 63537. Now, I put instead of the lv_font_symbol_20 and I want use symbols form my extra fonts, but I don't know, how. I wanted change the defines as SYMBOL_AUDIO etc... But there should be something wrong in my definitions. I cannot find my symbols. 
Please, could you help me, what numbers should I put into this _SYMBOL_VALUE3(EF,A0,EE) and why ?

Thank you very much

seyyah, 2019-05-10:
Maybe useful `php font_conv_for_lists.php` [1]

---
We use GitHub issues for development related discussions.
Please use the forum to ask questions or suggest enhancements.

It's possible to sign in to the [forum](https://forum.littlevgl.com/) with your GitHub account... you don't need to make a separate account.

## :clipboard: Reference
1. https://github.com/littlevgl/lv_utils#list-parameter-specific-utility

zavovi, 2019-05-10:
Hi, thank you, but I still cannot understand this:
Unicode: f2c5 (is it in hexadecimal?) and now this:
#define LV_SYMBOL_THERMOMETER_EMPTY     "\xef\x8b\x8b" 
Why 0xEF 0x8B 0x8B ?

And another question, when I downloaded version 5, then it is ok using unicode from version 5, right?

Thank you

seyyah, 2019-05-10:
Please read this code part,

```c
  // https://fontawesome.com/v4.7.0/icons/
  // label: unicode
  $unicodes = array(
    "THERMOMETER_EMPTY" => "f2cb",
    "TINT             " => "f043",
    "BALANCE_SCALE    " => "f24e",
    "BARS             " => "f0c9",
    "MICROCHIP        " => "f2db"
  );
```
> And another question, when I downloaded version 5, then it is ok using unicode from version 5, right?

I use v4.7.0. I don't experience fontawesome v5.

## :clipboard: Reference
1. https://github.com/littlevgl/lv_utils/blob/master/font_conv_for_lists.php#L7-L15

zavovi, 2019-05-10:
Oh thanks. I understand now. Solved my issue :-)

closed 2019-05-10 by zavovi

embeddedt, 2019-05-10:
@zavovi Please note: for future questions like this, use the [forum](http://forum.littlevgl.com/).