lvgl/lvgl #1062: Problem with lv_obj_create(*parent, *copy) when copy parameter points to a screen object

barmaley70, 2019-05-14:
The function subsequently calls lv_obj_get_x(copy) and lv_obj_get_y(copy) that accessing parent->coords.x1 and parent->coords.y1 without checking the parent for NULL. In case if copy object is a screen and does not have a parent it leads to NULL pointer access.

embeddedt, 2019-05-14:
Looks like a valid bug from a quick read of `lv_obj_create`.

(cc @kisvegabor)

kisvegabor, 2019-05-15:
It's really a bug.

Fixed here: https://github.com/littlevgl/lvgl/commit/c8d1413eced815ae903b25e99c0500a459b9d85e
Please verify if it's work now.

I applied the same fix for `dev-6.0` too.

kisvegabor, 2019-05-22:
@barmaley70 if it's still not working please comment here and reopen the issue.

closed 2019-05-22 by kisvegabor