lvgl/lvgl #1073: obj->design_func(obj, &obj_ext_mask, LV_DESIGN_DRAW_MAIN); causes hang

nanokatz, 2019-05-20:
As in #431 I have an SAMD21 based arduino with an Adafruit screen I am porting the library to.
However in lv_refr_obj
 obj->design_func(obj, &obj_ext_mask, LV_DESIGN_DRAW_MAIN); 
causes the controller to hang
As suggested in #431 I looked at the obj values:
.piolibdeps/littlevgl_ID6228/lv_core/lv_refr.c@591->design_func: 12041
.piolibdeps/littlevgl_ID6228/lv_core/lv_refr.c@593->obj: 536880188
.piolibdeps/littlevgl_ID6228/lv_core/lv_refr.c@595->obj_ext_mask:  x1:0 y1:0 x2:479 y2:7

They look just fine to me. Increasing the memory did not help.

Another suggestion there was:
>Will the crash disappear if you immediately return from lv_refr_area_part_vdb (lv_refr.c)?
Doing this stops the controller from hanging.

nanokatz, 2019-05-20:
On further digging I've traced the error further to _sw_color_fill_
When the function attempts to acces _mem[col]_ it simply stops. mem has a valid value as does col
 
`.piolibdeps/littlevgl_ID6228/lv_draw/lv_draw_vbasic.c@624->mem: 536871897
.piolibdeps/littlevgl_ID6228/lv_draw/lv_draw_vbasic.c@626->col: 0
`

nanokatz, 2019-05-21:
Another update. As in #431 it turned out to be an issue of compiler optimization. Disabling optimization in lc_draw_vbasic.c stops the system from hanging.

embeddedt, 2019-05-21:
@nanokatz Maybe this would be helpful? https://github.com/littlevgl/lv_arduino/issues/2#issuecomment-466004502

nanokatz, 2019-05-21:
> @nanokatz Maybe this would be helpful? [littlevgl/lv_arduino#2 (comment)](https://github.com/littlevgl/lv_arduino/issues/2#issuecomment-466004502)

I saw this issue as well. However I believe my setup code is correct:
```

  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.disp_flush = disp_flush;
  disp_drv.disp_fill = disp_fill;              /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
  disp_drv.disp_map = disp_map;                /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
  lv_disp_drv_register(&disp_drv);
```

embeddedt, 2019-05-21:
@nanokatz I see. Maybe @kisvegabor has an idea.

kisvegabor, 2019-05-22:
@nanokatz 
1. Please attach your lv_conf.h
2. SAMD21 has only 32 kB RAM. Maybe you have a stack overflow. Try to lower the memory usage reducing `LV_MEM_SIZE` and `LV_VDB_SIZE`.

nanokatz, 2019-05-22:
I doubt it is a memory issue as it occured with very low memory values. My lv_conf is here https://justpaste.it/2hygl

However I will also note this issue does not occur with v6. Optimization makes no differnce then

kisvegabor, 2019-05-23:
Can you reproduce it in the simulator?

embeddedt, 2019-05-23:
@nanokatz `mem` is an unaligned pointer. That is probably the reason why it is crashing.

Did you set `LV_ATTRIBUTE_MEM_ALIGN` to `__attribute__((aligned(4)))`? Maybe that will solve your problem; it will force the compiler to align important LittlevGL structures on a 4-byte boundary.

nanokatz, 2019-05-23:
I am actually having trouble recreating the bug now. It compiles and runs with optimizations enabled. Any idea on how to generate a misaligned structure?
I've switch to the dev-6.0 branch for my current project where it has not occurred so far.

embeddedt, 2019-05-23:
@nanokatz When optimization is enabled, the compiler will sometimes disregard alignment in an effort to reduce the amount of memory that is consumed. Using `LV_ATTRIBUTE_MEM_ALIGN` should fix the problem.

Notice that the address `536871897` (or `0x200003D9`) is not divisible by 4. This is probably causing an alignment fault inside the CPU.

nanokatz, 2019-05-23:
Sounds then like this is a problem that already has a solution. I'll see if I can recreate the problem and see if it goes away with the MEM_ALIGN setting;.

kisvegabor, 2019-05-24:
> Notice that the address 536871897 (or 0x200003D9) is not divisible by 4. This is probably causing an alignment fault inside the CPU.

I agree it can cause issues like this. We already faced this issue with Arduino and -Os optimization.

I suggest closing this issue now and reopen if the problem persists with `LV_MEM_ALIGN` too.

closed 2019-05-27 by kisvegabor