lvgl/lvgl #1075: lv_cont_get_fit_height returns width

nanokatz, 2019-05-21:
There is a small but major but in _lv_cont_get_fit_height_.
Currently it uses the width.....

```
lv_coord_t lv_cont_get_fit_height(lv_obj_t * cont)
{
    lv_style_t * style = lv_cont_get_style(cont);

    return lv_obj_get_width(cont) - 2 * style->body.padding.hor;
}
```

embeddedt, 2019-05-21:
@nanokatz Nice catch! I've fixed it here: https://github.com/littlevgl/lvgl/commit/037e828c8aeb9f37f106d643ad1fa9ad61a1bf70. Does it work now?

nanokatz, 2019-05-21:
I made the same changes locally and works just fine!

embeddedt, 2019-05-21:
@nanokatz Great! In the future, I would recommend just pulling the latest `master` branch from GitHub. That way you get any other fixes for other problems.

kisvegabor, 2019-05-22:
FYI, In `dev-6.0` this function is renamed to `lv_obj_get_height_fit` and there this bug was already fixed.

closed 2019-05-22 by nanokatz