lvgl/lvgl #1076: V6 Symbols Not being Drawn

nanokatz, 2019-05-22:
Hello Again;
Having seen the  number of improvments and fixes in V6 I decided to give it a shot. After the nessecary changes it's working fine...execpt one issue. Symbols are not being drawn

With V5.3:
[image]
With V6:
[image]

Config is here: https://justpaste.it/2758d

embeddedt, 2019-05-22:
@nanokatz Do they work if `LV_TXT_ENC` is set to `LV_TXT_ENC_UTF8`?

(cc @kisvegabor)

nanokatz, 2019-05-22:
> @nanokatz Do they work if `LV_TXT_ENC` is set to `LV_TXT_ENC_UTF8`?
> 
Yup that did it! Thanks for the super quick response!

embeddedt, 2019-05-22:
It's a bug, as they should be working in ASCII mode as well.

kisvegabor, 2019-05-23:
To make symbols general, in the new font converter they will be used with their real Unicode value from FontAwesome. 
However, the user will have the opportunity to remap the characters when converting a font. E.g. `0xEF23 => 0xF0`. 

In the online converter, we can have a list of configurations to make it easy to generate typical fonts. IMO, ASCII only encoding is not so common and it's acceptable to convert a new font in this case.

How does it sound? Or do you think we still should have a built-in ASCII + SYMBOLS (mapped to < 0xFF)?

nanokatz, 2019-05-23:
I think this is not a huge problem. I had ASCII enable previously just for testing; I use UTF8 for everything and in the end for this project as well. However maybe add a note in the conf template that using symbols with ASCII requires manual intervention.

kisvegabor, 2019-05-24:
> However maybe add a note in the conf template that using symbols with ASCII requires manual intervention.

We will see how things work with the new converter but I will add a note if it's still unclear.

kisvegabor, 2019-06-05:
The new font system is merged to `dev-6.0`. To support symbols for ASCII encoding of the symbols in the built-in-fonts the number of built-in fonts needs to be doubled. I think it's not worth the extra complexity of the config file because ASCII encoding is used very rarely in special cases.

In light of that, I close this issue. If you have any objection or idea, please comment here and reopen the issue.

closed 2019-06-05 by kisvegabor