lvgl/lvgl #1081: [6.0] lv_style_anim_get_user_data_ptr: infinite recursion

rreilink, 2019-05-24:
Compiling lvgl gives several warnings:

```c
In file included from lvgl/src/lv_objx/lv_gauge.c:9:
In file included from lvgl/src/lv_objx/lv_gauge.h:29:
In file included from lvgl/src/lv_objx/../lv_core/lv_obj.h:24:
lvgl/src/lv_objx/../lv_core/lv_style.h:257:1: warning: all paths through this function will call itself [-Winfinite-recursion]
{
```

This warning is correct since indeed `lv_style_anim_get_user_data_ptr` calls itself:
```c
static inline lv_anim_user_data_t * lv_style_anim_get_user_data_ptr(lv_anim_t * a)
{
    return lv_style_anim_get_user_data_ptr(a);
}
```

embeddedt, 2019-05-24:
(cc @kisvegabor)

Very strange. I assume that this function just hadn't been implemented yet.

rreilink, 2019-05-24:
Should it read `return lv_anim_get_user_data_ptr(a);` ?

embeddedt, 2019-05-24:
@rreilink I think so.

kisvegabor, 2019-05-25:
> Should it read return lv_anim_get_user_data_ptr(a); ?

Yes, I fixed it here 6093053

closed 2019-05-25 by kisvegabor