lvgl/lvgl #1083: Calling lv_canvas_fill_polygon results in a stack overflow

fivdi, 2019-05-27:
I'm using Espressif Systems [esp-iot-solution](https://github.com/espressif/esp-iot-solution) which comes with LittlevGL v5.3 to display information on a 2.4" 320x240 pixel LCD display with an ST7789 driver.

Everything from LittlevGL that I've used so far has worked well except for the `lv_canvas_fill_polygon` function. Running the following program which calls `lv_canvas_fill_polygon` results in a stack overflow.
```C
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "iot_lvgl.h"

/* Buffer for canvas */
static lv_color_t cbuf[sizeof(lv_color_t) * 100 * 80];

static void create_canvas(void) {
  lv_color_t bg_color = LV_COLOR_MAKE(0xec, 0xf0, 0xf1);
  lv_color_t fill_color = LV_COLOR_MAKE(0xbc, 0xa0, 0xa1);

  /* Set canvas pixels to light blue */
  uint32_t i;
  for(i = 0; i < sizeof(cbuf) / sizeof(cbuf[0]); i++) {
      cbuf[i] = bg_color;
  }

  /* Create the canvas object */
  lv_obj_t *canvas = lv_canvas_create(lv_scr_act(), NULL);

  /* Assign the buffer to the canvas */
  lv_canvas_set_buffer(canvas, cbuf, 100, 80, LV_IMG_CF_TRUE_COLOR);

  /* Draw a polygon */
  lv_point_t points4[] = {
    {.x=10, .y=10},
    {.x=40, .y=10},
    {.x=30, .y=40},
    {.x=20, .y=40},
  };

  lv_canvas_draw_polygon(canvas, points4, 4, LV_COLOR_RED);
  lv_canvas_fill_polygon(canvas, points4, 4, LV_COLOR_RED, fill_color);
}

void app_main() {
  lvgl_init();

  create_canvas();

  while (true) {
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}
```
If the call to `lv_canvas_fill_polygon` is removed, the stack overflow no longer occurs and the polygon is displayed correctly.

What I did notice is that `lv_canvas_fill_polygon` calls `lv_canvas_boundary_fill4` [here](https://github.com/littlevgl/lvgl/blob/v5.3/lv_objx/lv_canvas.c#L487) and that `lv_canvas_boundary_fill4` calls itself recursively [here](https://github.com/littlevgl/lvgl/blob/v5.3/lv_objx/lv_canvas.c#L509-L512).

These recursive calls appear to be the cause of the stack overflow.

If the following `printf` is added at the start of function `lv_canvas_fill_polygon`
```C
printf("x=%d, y=%d\n", x, y);
```
the output of running the test program is
```
brian@linuxbox:~/src/esp-idf/esp-iot-solution/examples/canvas$ make monitor
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6008
load:0x40078000,len:9188
load:0x40080000,len:6140
0x40080000: _iram_start at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/freertos/xtensa_vectors.S:1779

entry 0x40080330
0x40080330: _KernelExceptionVector at ??:?

I (28) boot: ESP-IDF v3.1.3 2nd stage bootloader
I (28) boot: compile time 12:00:29
I (28) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 40MHz
I (37) boot: SPI Mode       : DIO
I (41) boot: SPI Flash Size : 4MB
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (82) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x4aee8 (306920) map
I (199) esp_image: segment 1: paddr=0x0005af10 vaddr=0x3ffb0000 size=0x02914 ( 10516) load
I (203) esp_image: segment 2: paddr=0x0005d82c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/freertos/xtensa_vectors.S:1779

I (206) esp_image: segment 3: paddr=0x0005dc34 vaddr=0x40080400 size=0x023dc (  9180) load
I (218) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0x2631c (156444) map
0x400d0018: _flash_cache_start at ??:?

I (278) esp_image: segment 5: paddr=0x0008633c vaddr=0x400827dc size=0x071f4 ( 29172) load
0x400827dc: heap_caps_realloc at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/heap/heap_caps.c:278 (discriminator 1)

I (297) boot: Loaded app from partition at offset 0x10000
I (297) boot: Disabling RNG early entropy source...
I (298) cpu_start: Pro cpu up.
I (301) cpu_start: Starting app cpu, entry point is 0x40080ec4
0x40080ec4: call_start_cpu1 at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/esp32/cpu_start.c:232

I (0) cpu_start: App cpu up.
I (311) heap_init: Initializing. RAM available for dynamic allocation:
I (318) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (324) heap_init: At 3FFCFF30 len 000100D0 (64 KiB): DRAM
I (330) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (337) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (343) heap_init: At 400899D0 len 00016630 (89 KiB): IRAM
I (349) cpu_start: Pro cpu start user code
I (143) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (548) gpio: GPIO[22]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
x=25, y=25
x=26, y=25
x=27, y=25
x=28, y=25
x=29, y=25
x=30, y=25
x=31, y=25
x=32, y=25
x=33, y=25
x=34, y=25
x=35, y=25
x=34, y=26
x=35, y=26
x=34, y=27
x=33, y=26
x=34, y=26
x=33, y=27
x=34, y=27
x=33, y=28
x=34, y=28
x=33, y=29
x=34, y=29
x=33, y=30
x=32, y=29
x=33, y=29
x=32, y=30
x=33, y=30
x=32, y=31
x=33, y=31
x=32, y=32
x=33, y=32
x=32, y=33
x=31, y=32
x=32, y=32
x=31, y=33
x=32, y=33
x=31, y=34
x=32, y=34
x=31, y=35
x=32, y=35
x=31, y=36
x=30, y=35
x=31, y=35
x=30, y=36
x=31, y=36
x=30, y=37
x=31, y=37
x=30, y=38
x=31, y=38
x=30, y=39
x=29, y=38
x=30, y=38
x=29, y=39
x=30, y=39
x=29, y=40
x=28, y=39
x=29, y=39
x=28, y=40
x=27, y=39
x=28, y=39
x=27, y=40
x=26, y=39
x=27, y=39
x=26, y=40
x=25, y=39
x=26, y=39
x=25, y=40
x=24, y=39
x=25, y=39
x=24, y=40
x=23, y=39
x=24, y=39
x=23, y=40
x=22, y=39
x=23, y=39
x=22, y=40
x=21, y=39
x=22, y=39
x=21, y=40
x=20, y=39
x=21, y=38
x=22, y=38
x=23, y=38
x=24, y=38
x=25, y=38
x=26, y=38
x=27, y=38
x=28, y=38
x=29, y=38
x=28, y=39
x=27, y=38
x=28, y=37
x=29, y=37
x=30, y=37
x=29, y=38
x=28, y=37
x=29, y=36
x=30, y=36
x=29, y=37
x=28, y=36
x=29, y=36
x=28, y=37
x=27, y=36
x=28, y=36
x=27, y=37
x***ERROR*** A stack overflow in task main has been detected.
abort() was called at PC 0x40087ea0 on core 0
0x40087ea0: vApplicationStackOverflowHook at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/esp32/panic.c:676

Backtrace: 0x40087cb0:0x3ffd0c90 0x40087e87:0x3ffd0cb0 0x40087ea0:0x3ffd0cd0 0x400859bb:0x3ffd0cf0 0x40086fbc:0x3ffd0d10 0x40086f72:0x3ffd40b8
0x40087cb0: invoke_abort at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/esp32/panic.c:676

0x40087e87: abort at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/esp32/panic.c:676

0x40087ea0: vApplicationStackOverflowHook at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/esp32/panic.c:676

0x400859bb: vTaskSwitchContext at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/freertos/tasks.c:4579

0x40086fbc: _frxt_dispatch at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/freertos/portasm.S:406

0x40086f72: _frxt_int_exit at /home/brian/src/esp-idf/esp-iot-solution/submodule/esp-idf/components/freertos/portasm.S:206

Rebooting...
```
Increasing the size of the stack doesn't help. The stack overflow will still occur, it just takes a little longer for it to happen.

seyyah, 2019-05-27:
The boundary fill algorithm has limited capability :(

Critical line is

```c
if(c.full != boundary_color.full && c.full != fill_color.full)
```

If the objects overlap, the stop condition may fail.

Would you try filling out a single polygon having big area?

fivdi, 2019-05-27:
>The boundary fill algorithm has limited capability :(

That's unfortunate. Is it suitable for usage on systems with a limited amount of RAM?

>Critical line is
>
>if(c.full != boundary_color.full && c.full != fill_color.full)
>
>If the objects overlap, the stop condition may fail.

I'm not sure I understand here. If which objects overlap? The goal of the test program is to display a single object (polygon).

>Would you try filling out a single polygon having big area?

The polygon was increased in size to fill most of the canvas:
```c
  /* Draw a polygon */
  lv_point_t points4[] = {
    {.x=5, .y=5},
    {.x=95, .y=5},
    {.x=85, .y=75},
    {.x=15, .y=75},
  };
```
Unfortunately, the stack overflow still occurs.

seyyah, 2019-05-27:
> The goal of the test program is to display a single object (polygon).

I'm sorry :( I've just realized.

> Unfortunately, the stack overflow still occurs.
> That's unfortunate. Is it suitable for usage on systems with a limited amount of RAM?

The fillin algorithm is based on recursively "boundary fill". But when I searched a little, I came across non-recursive realization [1].

PR: How about adding this feature?

## :clipboard: Reference
1. https://tr.scribd.com/doc/246681478/Boundary-Fill-Algo-Without-Recursion

kisvegabor, 2019-05-27:
@seyyah 
For `v6.0` I'm planning to update the canvas to directly support `lv_draw_...` functions (via a simple API). Although there is no `lv_draw_polygon` function yet I think I can implement it by splitting the polygon to triangles and use `lv_draw_triangle`. This way we wouldn't need drawing functions directly for canvas. Do you agree with that?

seyyah, 2019-05-27:
Yes I agree. Thanks @kisvegabor

fivdi, 2019-05-27:
@seyyah @kisvegabor Thank you.

I'll avoid calling `lv_canvas_fill_polygon` for the moment as this seems like the best thing to do.

kisvegabor, 2019-05-30:
@fivdi Alright. It seems this issue can be closed then.

closed 2019-05-30 by kisvegabor