lvgl/lvgl #1088: Objects with rounded edges are not clipped appropriately

embeddedt, 2019-06-02:
When you set rounded borders on an object, the resulting draw area is not what you would expect.

Consider the following example (admittedly ridiculous, but it does a good job of illustrating the point):

```c
    lv_obj_t * ta = lv_ta_create(lv_scr_act(), NULL);
    lv_style_t ta_style;
    lv_style_copy(&ta_style, lv_ta_get_style(ta, LV_TA_STYLE_BG));
    ta_style.body.radius = 100;
    lv_ta_set_style(ta, LV_TA_STYLE_BG, &ta_style);

    lv_obj_set_size(ta, 100, 100);

    lv_obj_align(ta, NULL, LV_ALIGN_CENTER, 0, 0);
```
[image]

It looks to me like the clipping algorithm does not consider the rounded corners of the circle during clipping.

kisvegabor, 2019-06-02:
It's a known limitation.
It would be very computation heavy to mask the corners. Now only a rectangular area is used to decide a pixel is on the parent or not. With corner checking, it would be much more difficult.

There was a discussion about it once and we concluded that It would be faster to buffer the corners before drawing and restore them after all children are drawn. It could be enabled for each object one-by-one if required. 

The buffer size in an other question.
A circle with `radius = 30 px` has` area = 30 * 30 * 3.14 = 2,824` 
A square with `a = 2 * 30 px` has `area = 60 * 60 = 3600`

The difference is the area which needs to be buffered `buf_size= 3600 - 2824 = 774 byte`.
So 774 byte is required to buffer the 4 corners of a rectangle with 30 px radius. 
It's `8,600 byte` with `radius = 1000 px`. Not so much IMO.

What do think about this approach? Do you think it's worth to implement it? (Probably in v6.1)

embeddedt, 2019-06-02:
> What do think about this approach? Do you think it's worth to implement it? (Probably in v6.1)

I do like the approach, but I'm a bit concerned about the memory requirements.

> The difference is the area which needs to be buffered buf_size= 3600 - 2824 = 774 byte.
So 774 byte is required to buffer the 4 corners of a rectangle with 30 px radius.
It's 8,600 byte with radius = 1000 px. Not so much IMO.

This assumes 1 byte per pixel colors, right? If it was 32-bit color, it would require `744 * 4 = 3096` bytes for a 30px radius, and `34,400` bytes for 1000px.

kisvegabor, 2019-06-02:
True, it still needs to be multiplied by 2 or 4 typically.
I've just realized that I wrote 1000 px radius for 8,600 bytes, but it's for 100 px.

I've created a chart to see how it scales. It's in px unit so it needs to be multiplied according to the color depth. It hits the 1K px buffer size at 35 px radius.

[image]

embeddedt, 2019-06-02:
@kisvegabor Given your chart, it looks like memory usage is minimal as long as your radius is below 15. After that it starts rising quickly. Is that acceptable for an embedded GUI library?

kisvegabor, 2019-06-02:
In my opinion, it can be acceptable if the user can set the maximal radius to buffer.

kisvegabor, 2019-06-04:
There is another, more flexible approach in my mind.
There could be an `lv_mask` object type. It should have function like `lv_mask_get_value(x,y)`. This function can work in a lot of ways (even the user can create custom `get_value` functions):
- have bitmap to know which pixel can be drawn
- store the point of a line. Pixels above the line are drawn, below doesn't. Id the line is rotated nice animations can be done.
- buffer the points a quarter circle to quickly test if the pixels are inside or outside of a radius.

As `lv_mask` a normal object, it could be positioned, sized, and can be the parent of other objects. In its `design` function instead of drawing the mask it should set something like `lv_draw_set_mask();` When its children are rendered they need to check if a mask is set and if pixels can be drawn or not. In `LV_DESIGN_DRAW_POST` the mask can be removed by `lv_draw_remove_mask()`. 

It's a very flexible, memory effective and relatively fast. What do you think about it?

embeddedt, 2019-06-04:
@kisvegabor I like the `lv_mask` idea, but specifically for the quarter circles scenario, it seems like extra work for the user.

To get my example above to work, one would theoretically have to create an `lv_mask` object and then create the text area as a child of it, right?

kisvegabor, 2019-06-04:
>  I like the lv_mask idea, but specifically for the quarter circles scenario, it seems like extra work for the user.

Yes, it's extra work, but the "rounded rectangle" mask can be built-in.

> To get my example above to work, one would theoretically have to create an lv_mask object and then create the text area as a child of it, right?

Exactly.

embeddedt, 2019-06-04:
> the "rounded rectangle" mask can be built-in.

I'm confused. Before you were saying that this would be part of the `lv_mask` object. What do you mean by "built-in" here?

kisvegabor, 2019-06-04:
> I'm confused. Before you were saying that this would be part of the lv_mask object. What do you mean by "built-in" here?

I mean `lv_mask` can have some built-in making option (like a rounded rectangle) but it would allow the users to define their custom mask as well.
For example it could have an API like `lv_mask_set_test_cb(msk, my_mask_cb)` but there would be `lv_mask_test_rounded_rect_cb` to set.

embeddedt, 2019-06-04:
@kisvegabor I see.

BrianPugh, 2019-06-04:
If storing a buffer mask, why not do 1bit per pixel instead of 1byte? Additional compression could be had by just storing the edges and indicating if the right/left is out of bounds. I'm assuming the byte approach is just done for faster/simpler indexing?

kisvegabor, 2019-06-05:
@BrianPugh 
The API proposed here:
> For example it could have an API like lv_mask_set_test_cb(mask, my_mask_cb) but there would be an lv_mask_test_rounded_rect_cb to set.

is more flexible than bitmaps.  `my_mask_cb` can operate on a bitmap inside but it can do anything else too. For example 
```
if (x < x_limit) return 0;
else return 1;
```
It's faster then a bitmap and `x_limit` can be easily changed dynamically to make an animation. 

IMO it would be better to return `lv_opa_t` to enable alpha mapping. Fo example:
```
lv_opa_t my_mask_cb(x, y) {
  return my_opa_map[x]; 
}
```

creates a horizontal fade out.

Bitmaps are still, e.g.
```
lv_opa_t my_mask_cb(x, y) {
  return bitmap_get_valus(x,y) ? LV_OPA_COVER  LV_OPA_TRANSP; 
}
```

What do you think?

BrianPugh, 2019-06-05:
Seems reasonable to me

kisvegabor, 2019-06-06:
If you agree this kind masking could be a very interesting feature for `v6.1`.

kisvegabor, 2019-10-28:
This feature is added in `dev-7.0` and can be enabled by `style.body.corner_mask = 1`.
See #1243

closed 2019-10-28 by kisvegabor

embeddedt, 2019-10-28:
Can't wait to play with it!