lvgl/lvgl #1089: [Help wanted] New online font converter

kisvegabor, 2019-06-02:
The new font system is [successfully integrated](https://github.com/littlevgl/lv_font_conv/issues/10#issuecomment-497982732) into LittelvGL. Still there are a few minor issues but the majority is already working.

Summary of the new converter:
- Can be found in [lv_font_conv](https://github.com/littlevgl/lv_font_conv) repo
- Written in Node.js
- Multiple font files can be used to include them in one font

The next task is to update the online font converter. 
I created the new layout here: https://littlevgl.com/font_conv_new
The source is here: https://github.com/littlevgl/lv_font_conv/tree/web/web

Help wanted with JS part of the converter. @puzrin created an example JS file [here](https://github.com/littlevgl/lv_font_conv/blob/web/web/index.js).

To try it
```sh
git clone https://github.com/littlevgl/lv_font_conv
cd lv_font_conv
git checkout web        # use the web branch
npm install
npm start               # build and open in browser
```

By clicking the "Convert" button some data will be printed into the console. It should be updated to create the real C output format (`format: 'lvgl'`) with the entered parameters. 

The "Add font" button does nothing now but it should add a new "TTF/WOFF file + Range + Symbols" section to include more fonts.

kisvegabor, 2019-06-04:
Finally, I managed to make it work with some limitations:
- Only 2 fonts can be included
- Range/character remapping is not supported with `0x200-0x3007=>0x100`

As I'm a very beginner in JavaScript it's a super primitive but working implementation. 

If you are interested in improving it or it's not working well, please comment here and reopen this issue.

embeddedt, 2019-06-04:
@kisvegabor I tried cloning it but I'm having trouble compiling the `scss` file in the `web` directory. I get an error saying it can't find the stylesheet `~bootstrap/scss/bootstrap`. I know I probably need to download something else but I don't know where to find this file.

puzrin, 2019-06-04:
@embeddedt i could try to help, if you show exact list of terminal commands.

Probably fogot to execure `npm install` to setup dependencies.

embeddedt, 2019-06-04:
@puzrin I didn't use `npm`. It only downloaded the font convertor itself - and not any of the HTML/JS files that @kisvegabor created. Those are what I am having problems with.

kisvegabor, 2019-06-04:
I've followed my own instructions and realized `git checkout web` is not working because there is a `web` folder too. `git checkout web --` should work.

As far as I know, the `npm` part is required because it does some black magic to put things together. 

`npm start` starts a local server at `1234` port and opens a browser with `localhost:1234` to see the result.

To get online deployable `html` + `js` versions `npm run build` needs to be called. The files will be created in the `dist` folder.

Note that, if you open the html file on your machine it won't be styled, because I assume that it will be included in `div` on littelvgl.com and gets bootstrap from there.

embeddedt, 2019-06-04:
@kisvegabor Now I understand. It seems to be working.

puzrin, 2019-06-04:
@embeddedt

> @puzrin I didn't use `npm`. It only downloaded the font convertor itself - and not any of the HTML/JS files that @kisvegabor created. Those are what I am having problems with.

You need some basic knowledge about ALL node.js projects. If you download sources manually, you need to run `npm i` then, to install dependencies.

If project is published, and you install it from registry via npm, then dependencies are installed automatically.

It's always good idea to check [script](https://github.com/littlevgl/lv_font_conv/blob/master/package.json#L19) section of `package.json` to see available `npm run XXX` commands.

---

Regarding to html files...

It's common practice to NOT store builds in repo, because those are needed only for upload/publish. Build files case frequent unsync or lot of garbage in commits history. For development it's more convenient to use built-in dev server - it rebuild and reload pages on the fly when you change sources.

embeddedt, 2019-06-04:
@kisvegabor Is jQuery currently used for the website?

puzrin, 2019-06-04:
https://github.com/littlevgl/lv_font_conv/commit/7125e6337f12bd17fc74cdbadd2a60b84668610f

Pushed proposals to separate branch:

- use partials, to show dev server page styled (and build raw content for production)
- dropped all injected styles, use bootstrap ones instead
  - that requires bootstrap 4, sorry :(, your site uses bootstrap 3.

I don't insist on anything. Use any part you wish if you like it.

kisvegabor, 2019-06-05:
@embeddedt 
> Is jQuery currently used for the website?

Yes, it is used.

> Pushed proposals to separate branch:

Looks good4 This way people can use the converter offline even without CLI.

puzrin, 2019-06-05:
> This way people can use the converter offline even without CLI.

I would not recommend to promote such practices. Browsers have different security restrictions for local files. One day you can find that something does not work and can not be fixed.

If you don't like pain with manual site updates, i'd suggest to publish convertor page in `gh-pages` branch of `lvgl` repo, without your site wrapper, and create script for update.

kisvegabor, 2019-06-05:
> I would not recommend to promote such practices. Browsers have different security restrictions for local files. One day you can find that something does not work and can not be fixed.

I don't have enough experience to judge this, but I belive. :)

> If you don't like pain with manual site updates, i'd suggest to publish convertor page in gh-pages branch of lvgl repo, without your site wrapper, and create script for update.

I'd like to keep the converter on littelvgl.com to direct more traffic there. It's worth more than manually updating the files sometimes.

embeddedt, 2019-06-05:
> I would not recommend to promote such practices. Browsers have different security restrictions for local files. One day you can find that something does not work and can not be fixed.

I agree. It also depends on the exact browser that you're using (i.e. I think Firefox works fine, but Chrome blocks local JS from executing).

puzrin, 2019-06-05:
http requests, cookies, localstorage, and some other things are blocked everywhere. I don't remember all details. Only the final conclusion that this mode is very specific and should not be used, except reading generated docs.

kisvegabor, 2019-06-05:
Okay, understood, let's burn this idea :)

puzrin, 2019-06-05:
@kisvegabor do you wish me to add partials to show CSS in dev server? I consider `/web` folder as your personal sandbox, and don't plan to change there anything without your permission.

Example is here https://github.com/littlevgl/lv_font_conv/commit/7125e6337f12bd17fc74cdbadd2a60b84668610f (but without content change, to keep bootstrap3 compatibility)

embeddedt, 2019-06-05:
Can partials also add jQuery? I would like to use it in my implementation for more than 2 font files as it is much easier than writing plain JavaScript.

puzrin, 2019-06-05:
See commit. It adds separate wrapper file, to include any missed libs from CDN. There are no technical limitations. The only conditions is to add the same on website too.

allyzc, 2019-06-25:
Share a LvglFontTool.
[https://github.com/allyzc/LvglFontTool](url)

embeddedt, 2019-06-25:
@allyzc 

> (GitHub repo description) I recently made a LittlevGL font generator that supports multi-country text, supports anti-aliasing, feels much better than the official one, and is now shared with partners who need it.

Our font convertor also supports antialiasing. Can you please explain exactly what you have improved? 

Additionally, there is no source code provided in your repository.

amirgon, 2019-06-25:
> Additionally, there is no source code provided in your repository.

@allyzc Most people **will not install and run executables from untrusted source**.  

I suggest you add sources, remove the executables, and instead provide instructions how to build your sources.  

Providing sources instead of executables will have many benefits:

- People will download and use your project
- People will be able to point at bugs, suggest pull requests etc.
- If you fork from the official lvgl font generator, it will be easier to see the differences from the official version, as well as contribute improvements back to the official version.
- Your code could run on other OSes, Linux and Mac for example. Many people use them.

What do you think?

embeddedt, 2019-07-17:
@kisvegabor Getting this to work with any number of fonts is still on my list, but it's behind the file browser and the Micropython simulator. Should the issue be closed for now?

kisvegabor, 2019-07-18:
If you will deal with it later we can leave it open as a reminder.

embeddedt, 2019-07-25:
I've added the feature in https://github.com/littlevgl/lv_font_conv/pull/19.

It's not heavily tested yet but CI passed and it seemed to work when I tested it with FontAwesome.

kisvegabor, 2019-07-26:
I've tested it and it was working for me too.

However, it seems there is some issue with the HTML code. The layout is messed up.
See https://littlevgl.com/font_conv_test

puzrin, 2019-07-26:
I'd suggest to use wrapper for dev page, as been suggested in https://github.com/littlevgl/lv_font_conv/commit/7125e6337f12bd17fc74cdbadd2a60b84668610f. IMO current layout issue happened because dev server output is "not useable" (no full css, no full js).

embeddedt, 2019-07-26:
@kisvegabor I think I know why. I'll fix it today.

kisvegabor, 2019-07-26:
@puzrin 
> I'd suggest to use wrapper for dev page, as been suggested in littlevgl/lv_font_conv@7125e63. IMO current layout issue happened because dev server output is "not useable" (no full css, no full js).

Sorry, but I don't remember, why wasn't it merged finally?

@embeddedt 
>  I think I know why. I'll fix it today.

Awesome, thanks!

embeddedt, 2019-07-26:
>  why wasn't it merged finally?

IIRC it was because it was created using Bootstrap 4, but our site uses Bootstrap 3.

puzrin, 2019-07-26:
Yes, and it was done on diverged branch (where were a lot of temporary commits with broken travis tests).

I try to not intrude into `/web` folder, because don't know all details about website (and probably, don't like to know :). So, i uploaded proposal and suggested to use any part you wish if you find it useful. That change is very small and easy to copy.

kisvegabor, 2019-07-26:
> That change is very small and easy to copy.

It's not that trivial to me what is required from the commit and what isn't. :sweat_smile: 
If it's really not a big work could you add it in a new branch? Even with Bootstrap 4. From that point, it would be really easy to customize it to work with Bootstrap3.

puzrin, 2019-07-26:
Done, boss :) !

https://github.com/littlevgl/lv_font_conv/tree/web-wrap

Note:

- `index.html` renamed to `content.html` without change (all bugs preserved :)! I only added wrapper.
- Now in `/dist` you should use `content.html` and ignore `index.html`.

Diff for new `index.html` is ugly, just look all file body instead, it's small.

kisvegabor, 2019-07-27:
Work perfectly, thank you!

puzrin, 2019-07-27:
merged to master

embeddedt, 2019-07-27:
@kisvegabor The issue should be fixed now. Sorry that it's a day late.

kisvegabor, 2019-07-29:
@embeddedt It works well now, thank you!

Only one minor thing can be improved. Can you add a `<hr>` between the "font boxes"?

[image]

embeddedt, 2019-07-29:
@kisvegabor Added.

kisvegabor, 2019-07-30:
@embeddedt 
Uploaded, tested, working, thank you very much! :slightly_smiling_face:

closed 2019-07-30 by kisvegabor