lvgl/lvgl #1093: The Forum has been moved

kisvegabor, 2019-06-05:
Hi,

I'm happy to announce that we have replaced the old Forum with a more modern one.
The new forum is available at https://forumtest.littlevgl.com/  On July 1 it will be changed to `forum.littlevgl.com` and the old forum will be `forumold.littlevlg.com`. The old forum still will be available as a "read-only archive".

Please register at https://forumtest.littlevgl.com/ and ask your questions there. 

If you find some issues on the new Forum please, comment here.

C47D, 2019-06-05:
Hi,

The topics on the old forum will be ported to the new one?

embeddedt, 2019-06-05:
@C47D No, you will have to access them by doing a Google search or going to `forumold.littlevgl.com`.

At the moment it isn't possible to transfer topics. Possibly in the future we can look into that.

amirgon, 2019-06-06:
@kisvegabor I don't know how it behaves for others, but when I'm clicking the link on the new banner you added to the old forum, I'm redirected to https://littlevgl.com/ and not to the new forum.

kisvegabor, 2019-06-06:
@amirgon I updated it.

embeddedt, 2019-06-06:
@kisvegabor The new forum is not notifying me of topics as soon as they are created. For example, [this topic was created](https://forumtest.littlevgl.com/t/anthill-regulator/38) but I was never emailed about it.

kisvegabor, 2019-06-06:
By default, you aren't "Watching" any of the categories. I updated it for new users to get notified about new topics in the following categories (not every post, only new topics):
- Announcements
- Getting started
- How-to
- My projects

embeddedt, 2019-06-09:
@kisvegabor Thanks for adjusting that. I found that mailing list mode seems to send an email on every new post (like GitHub does).

kisvegabor, 2019-06-11:
@embeddedt Do you think we should enable it for every user by default? I think sending notification about the first post is enough. This way user can decide if he is interested in and "Watch" that topic.

embeddedt, 2019-06-11:
> I think sending notification about the first post is enough. 

I agree.

puzrin, 2019-06-14:
@kisvegabor Why do you changed your preference in favour of discourse? What's changed since last time we discussed possible forums?

kisvegabor, 2019-06-15:
@puzrin 
What you showed Discourse for the first time I also saw that it's better than PhpBB (at least in my opinion) but didn't want to:
- pay for VPS
- learn server-side things

which are required for Discourse.

However, with the sponsorship from Digital Ocean and with the help of @seyyah both issues were solved.

amirgon, 2019-06-16:
> However, with the sponsorship from Digital Ocean and with the help of @seyyah both issues were solved.

@kisvegabor Do you rely on continuous sponsorship from Digital Ocean? What if next year they decide not to continue it? would you go back to PhpBB?

kisvegabor, 2019-06-16:
> Do you rely on continuous sponsorship from Digital Ocean? What if next year they decide not to continue it? would you go back to PhpBB?

I believe they won't stop the sponsorship in the next few years but no guarantee of course. If it happens I will pay that ~ $200/per year.

kisvegabor, 2019-06-18:
@amirgon As you have already used the new Forum what do you think about it? Do still think the PhpBB version was as good as the new one or do you see the benefits of the update?

amirgon, 2019-06-18:
> @amirgon As you have already used the new Forum what do you think about it? Do still think the PhpBB version was as good as the new one or do you see the benefits of the update?

Discourse gives a better user experience compared to PhpBB, no doubt about that.  

One thing I liked about PhpBB though, is the separation to sub-forums.  
Discourse has Categories, but still, PhpBB **forces** you before starting a new post to decide on which sub-forum it should go. In Discourse most people don't bother to label and categorize their messages so we get a single flow of posts.

kisvegabor, 2019-06-19:
> Discourse has Categories, but still, PhpBB forces you before starting a new post to decide on which sub-forum it should go.

I also miss it. To be honest I haven't checked so far if the posts are ad the category or not but it seems most then are added. Or @embeddedt silently added them. :smile:

embeddedt, 2019-06-19:
> Or @embeddedt silently added them. :smile:

I have not been... but I don't mind, so I'll just watch for when I see a new post pop up. :wink:

kisvegabor, 2019-06-20:
I'll check it too!

amirgon, 2019-06-28:
@kisvegabor

Someone just sent me a message on the old forum, ‏‏but it doesn't seem to be available in more. 

How can I access the old forum?

kisvegabor, 2019-06-28:
I've changed the URLs as it was stated in the first comment:
- new forum: https://forum.littlevgl.com
- old forum: http://forumold.littlevgl.com/

I tried to redirect `formtest` to `froum` but it's not working well so far due to SSL issues.

seyyah, 2019-06-28:
> I tried to redirect formtest to froum but it's not working well so far due to SSL issues.

You can try to forwarding. For example we do it this way on the GoDaddy [1].

[image]

## :clipboard: Reference
1. https://tr.godaddy.com/help/forward-my-domain-12123

amirgon, 2019-06-28:
@kisvegabor could you update the link to the new forum from the old forum? It's still forumtest.

embeddedt, 2019-06-28:
@amirgon It was already mentioned, but I updated it to make it clearer.

amirgon, 2019-06-28:
No, I still get this:

‘‘‘
This site can’t be reached The webpage at https://forumtest.littlevgl.com/ might be temporarily down or it may have moved permanently to a new web address.
ERR_FAILED
‘‘‘

embeddedt, 2019-06-28:
@amirgon You need to go to https://forum.littlevgl.com now. That is what the post says.

amirgon, 2019-06-28:
I know that @embeddedt

I'm saying that the link on the old forum to the new forum is wrong.

embeddedt, 2019-06-28:
@amirgon Oh... now I understand. Sorry about that.

Not sure if I can update that one. I will check.

kisvegabor, 2019-06-28:
It turned out that I can't solve it in an easy way... My host doesn't give SSL for subdomains by default so I can't redirect `https://forumtest....` only `http://forumtest...`. 

I need to buy an "SSL install" service for `forumtest` subdomain only to be able to redirect it. I know it's ridiculous...

Have you got an email notification about [this post](https://forum.littlevgl.com/t/the-forums-url-has-been-changed/136/4)? It should let people know what happened. 

Or do you think it would be really important to make the redirection work? It's ~$20 to add SSL to the subdomain. So not a big deal but I wouldn't buy it if not required.

amirgon, 2019-06-28:
@kisvegabor No I don't think you need redirect at all.  

I think it's enough to change the URL on the old forum, such that clicking on it will open the right forum URL instead of forumtest.  

It looks like the string changed correctly to `https://forum.littlevgl.com`, **but the `a href` still points to forumtest**:

[image]

```html
<a href="https://forumtest.littlevgl.com" 
  title="LittlevGL"><img class="header-banner" 
  style="border-radius:0px; width:1500px;" 
  src="http://forumold.littlevgl.com/images/banner.jpg" alt="Forum">
</a>
```

kisvegabor, 2019-06-28:
Aaah, I see.
Done! :slightly_smiling_face:

seyyah, 2019-06-29:
> My host doesn't give SSL for subdomains by default so I can't redirect `https://forumtest....` only `http://forumtest...`.

Do you want to switch Cloudflare free? I can help

## :clipboard: Reference
1. https://www.cloudflare.com/plans/#compare-features

kisvegabor, 2019-06-30:
> Do you want to switch Cloudflare free? I can help

If I understood well Cloudflare is a nameserver. Using it will I am able to redirect https even in the site doesn't have certificates?

seyyah, 2019-07-01:
> > Do you want to switch Cloudflare free? I can help
> 
> If I understood well Cloudflare is a nameserver. Using it will I am able to redirect https even in the site doesn't have certificates?

Yes, sure. You also configure for your subdomain (or specific page) [1].

## :clipboard: Reference
1. https://support.cloudflare.com/hc/en-us/articles/200172286-Configuring-URL-forwarding-or-redirects-with-Cloudflare-Page-Rules

kisvegabor, 2019-07-02:
Thank you @seyyah. I'm still considering it. I'm a little bit afraid of risk to break everything. :)

seyyah, 2019-07-02:
> I'm still considering it. I'm a little bit afraid of risk to break everything. :)

I'm not network expert. I've enough info/practice for DNS record etc. I can help for DNS settings.

Firstly we migrate from mediacenter.hu [1] to Cloudflare[2] with (maybe) cpanel.

Then we manage your domain record with littlevgl.com related below

| Type | Name | Value |
|--------|----------|---------|
| A | littlevgl.com | IP address ^1 |
| CNAME | www.littlevgl.com | |
| CNAME or A | docs.littlevgl.com | ^1|
| A | forum.littlevgl.com | |
| CNAME or A | blog.littlevgl.com | |

Notes

^1: Page(s) can be deployed Github pages.

## :clipboard: References
1. https://whois.domaintools.com/littlevgl.com
2. https://support.cloudflare.com/hc/en-us/articles/205195708-Changing-your-domain-nameservers-to-Cloudflare

kisvegabor, 2019-07-04:
@seyyah 
Thank you for offering your help. I was thinking about it and concluded that the current issue (forumtest->forum) will be deprecated soon. Therefore I wouldn't like to risk further issue now without real benefit. 
Anyway, I keep this option in mind! It might come handy in the future!

amirgon, 2019-07-04:
@kisvegabor The old forum, which still contains important information, is not indexed by google.   
 
For example, [searching google for "littlevgl monitor_cb"](https://www.google.com/search?q=littlevgl+monitor_cb&rlz=1C1GCEB_enIL841IL841&oq=littlevgl+monitor_cb&aqs=chrome..69i57j33.2653j0j7&sourceid=chrome&ie=UTF-8) returns a **broken link** to ["[Solved]Help with lv_examples Benchmark - LittlevGL forum"](https://forum.littlevgl.com/viewtopic.php?t=62). Most users will not know how to reach this post on http://forumold.littlevgl.com/.

I suggest adding some links **to the old forum** on LittlevGL web site to improve its SEO, otherwise no one will be able to reach it by a web search.

amirgon, 2019-07-04:
@kisvegabor another problem - after you changed the new forum url to forum.littlevgl.com, **images on already existing posts are no longer available**.

For example, see https://forum.littlevgl.com/t/micropython-downs-when-using-lv-obj/132

Is it possible to automatically fix image URLs on these older posts?

seyyah, 2019-07-04:
> @kisvegabor another problem - after you changed the new forum url to forum.littlevgl.com, **images on already existing posts are no longer available**.
>
> For example, see [...]

We try to fix with this trick [1].

## :clipboard: Reference
1. https://meta.discourse.org/t/replace-a-string-in-all-posts/48729/4

seyyah, 2019-07-04:
> @kisvegabor The old forum, which still contains important information, is not indexed by google.
>
> For example, [searching google for "littlevgl [...]

I think @kisvegabor just add the DNS record/CNAME for the old forum. Maybe transfering guide [1] can also be reviewed.

## :clipboard: Reference
1. https://www.phpbb.com/support/docs/en/3.1/kb/article/transferring-your-board-to-a-new-host-or-domain/

kisvegabor, 2019-07-04:
@amirgon 
> I suggest adding some links to the old forum on LittlevGL web site to improve its SEO, otherwise no one will be able to reach it by a web search.

I've added the old forum to google search console. Let's see if Google will index it in a few days.

> Is it possible to automatically fix image URLs on these older posts?

I've also noticed it and found that by editing the post and adding a 'space' somewhere the image can be fixed. However, I don't know about an automatic method.

@seyyah 
> We try to fix with this trick [1].

Which trick do you mean exactly?

> I think @kisvegabor just add the DNS record/CNAME for the old forum. Maybe transfering guide [1] can also be reviewed.

I've added a CNAME record too. Thank you for the tip.

If google search console and CNAME doesn't help I'll add links at littlevgl.com. Probably google needs some time to realizes what has happened with the `forum*` sites :)

kisvegabor, 2019-07-10:
I think this issue can be closed as almost two weeks have passed since switching to new urls.

@amirgon 
It seems google indexed it `forumold`. I've tested it with "Charts with micropython bindings littlevgl".
 

Thank you for the feedback and help!

closed 2019-07-10 by kisvegabor