lvgl/lvgl #1095: lv_obj_set_pos() may encounter a NULL pointer

lianggongkai, 2019-06-06:
void lv_obj_set_pos(lv_obj_t * obj, lv_coord_t x, lv_coord_t y)
{
    /*Convert x and y to absolute coordinates*/
    lv_obj_t * par = obj->par;
    ...........
    /*Send a signal to the parent too*/
    par->signal_func(par, LV_SIGNAL_CHILD_CHG, obj);
}
here , we should make sure par is not NULL pointer. If par is null , par->signal_func() will encounter a segment fault, or usageFault.

embeddedt, 2019-06-06:
@kisvegabor Is it necessary to check for this? I thought screens can't be moved.

kisvegabor, 2019-06-06:
>  Is it necessary to check for this? I thought screens can't be moved.

Yes, this problem can occur if you are trying to position a screen. Actually, none of the coordinate manipulator functions work for screens (e.g. `lv_obj_set_size`). I think it's enough to emphasize it in the documentation.

embeddedt, 2019-06-06:
Added note in littlevgl/docs#19.

closed 2019-06-06 by embeddedt